prusa3d/Prusa-Firmware #2636: [BUG] Autoloading screen wrong

ulab, 2020-04-27:
**MK3S**
**3.8.1**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

**Describe the bug**
The autoloading screen shows the following text:

> Autoloading filament is active, just press the knob and insert filament...

When I "press the knob" it jumps back to the main menu. Instead I have to insert filament and wait for the autoload to happen.
  
**To Reproduce**
Start autoloading from the main menu.
  
**Expected behavior**
The screen should display the correct text on how to autoload filament.

awenelo, 2020-04-27:
This is expected behaviour. The instruction `just press the knob and insert filament` intends for you to exit back to the main menu and the then for you to trigger autoload by inserting filament. There is no difference in autoload behaviour no matter where you are in the menu structure, which I believe it is trying to indicate by sending you back to the main menu.

ulab, 2020-04-27:
I still find it confusing. 

Perhaps something like "Autoloading is active, just insert filament." is enough?  There's no need to explain people to press the button, since that's how you go back with other info screens too.

JohnnyDeer, 2020-12-07:
Hi @ulab this is feature as @awenelo wrote. I close this issue.

closed 2020-12-07 by JohnnyDeer