prusa3d/Prusa-Firmware #2641: [BUG] upgrading to release v3.9.0.RC3

henrikkoch, 2020-04-28:
Could not upgrade without getting error and message about failed. Even though the info showed RC3 in version

**Printer type** - MK3S, 
**Printer firmware version**- 3.9.0-RC3  (upgraded from ...RC1

**Describe the bug**
 Could not upgrade without getting info about failed upgrade
  
**To Reproduce**
 Tried 3 times
  
**Expected behavior**
  to pass

** solution
After turning power off and on and again upgrading, it passed

leptun, 2020-04-29:
If you can make it fail again, please post the avrdude logs from PrusaSlicer.

Photogad, 2020-04-29:
I did not have this issue, sadly

Makhaira, 2020-04-29:
I had this, watching the advanced log live I believe there is not enough wait time on the reboot.

Upgrading from 3.8.1 to 3.9.0 RC3. 
Flash proceeds successfully to 100%
Printer reboots
Flash then says failed, unable to connect to printer (printer is still on the lcd splash text at this point) 
Printer finishes booting a few seconds later, displays correct RC3 version number in Support menu. 

Test print in progress now. 

Feels like there's not enough of a wait time before trying to reconnect to the printer.

SRIFRAF, 2020-04-29:
Flash, restore, Flash to 3.9.0 RC3 went fine but problem remains..
I upgraded from 3.8.1 RC1 2851, that was the last FW build that solved my Z-Leveling Enforcement when starting a mesh for a print but I was still having initial Z-height inconsistences to the point of scratching the bed... This upgrade to 3.9.0 RC3 brings back the Z-leveling Enforcement problem during the Mesh level before every print again.  Both XYZ Cal and machine is perfect, seems like this relationship with the probe and bed temp is an unending painful issue?

kic883, 2020-04-30:
I don't have any failure on my MK3S firmware upgrade (from 3.8.1)

SRIFRAF, 2020-05-03:
Can someone confirm in RC1, RC2, and RC3 (I have tried all, and they all have the Z-leveling Enforcement fail on my new Mk2.5S, so I am stuck with 3.8.1 RC1 2851 and it messes up my initial Z-height) Is the bed heater turned OFF by default when the Mesh leveling starts and runs, seems my bed heater light flickers during mesh leveling, I was under the impression that this was supposed to be OFF in these releases. Perhaps there is something I must change in the firmware to "deactivate" the bed heater during Mesh?

3d-gussner, 2020-05-19:
@henrikkoch What is the status on this issue? If it is [solved] can you please consider to close it.

3d-gussner, 2020-06-08:
@JohnnyDeer Stale issue, consider to close it.

closed 2020-06-08 by JohnnyDeer