prusa3d/Prusa-Firmware #2642: [BUG] Hotter bed temp = less accurate thermistor reading -- compensate via table in firmware?

Photogad, 2020-04-29:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0RC3

**Describe the bug**
The hotter that the heated bed it set to, the less accurate the thermistor reading is (temp on LCD). It's very accurate up to about 40c, but the hotter you go from there, the less accurate it is. I measured the actual bed temp with both a thermocouple and an IR gun. I trust the thermocouple more; though both produced similar results verifying that the bed was much cooler at higher temperatures that the printer thinks.

I also measured the top of the heated bed, as well as the bottom of the heated bed (right next to the thermistor) and got similar results.
  
**To Reproduce**
Tape a thermocouple to the center of the heated bed and compare your actual readings to what the printer thinks.
  
**Expected behavior**
If I set the printer to 90c for heated bed, it should be 90c, not 78c.

**I've long suspected that Prusa is aware of this issue, and that's why they use a bed temp of 90C in their PETG slicer profiles, when most websites say 75c bed temp is optimal for PETG. So if you account for the fact that the bed is only 78c when you set it to 90c, this makes sense. I just wish Prusa could confirm this.**

I don't think this heated bed behavior is anything to be ashamed of. From what I've researched, this is just how thermistors work --- thermistors aren't totally accurate at all temperatures, and usually the higher or lower you go, the less accurate they are. But I'm just posting this issue on behalf on the community, to see if there's any way that the temperature variances can be compensated for via firmware, maybe with offsets stored in a table.

I know the Marlin firmware has a temperature offset value that I could set during a firmeware re-compile, but it's just a blanket offset throughout the whole temperature range. So for example if I set it to -12c, that means I would get accurate temps when my bed is at 90c, but when it's at 40c it will actually be 62c.

[image]
[image]
[image]
[image]
[image]

leptun, 2020-04-29:
Could be related https://github.com/prusa3d/Prusa-Firmware/issues/2601
¯\\\_(ツ)\_/¯

Photogad, 2020-04-29:
> 
> 
> Could be related #2601
> ¯\_(ツ)_/¯

Not sure, you are the expert, I just find the issues and post them.

That linked issue seems to be related to PINDA temp. I'm not so sure this issue here is a related. This just seems to be the nature of the bed thermistor, a simple resistance measuring device rather than an actual thermometer, the thermistor attempts to guess the temperature but has a harder time the higher the temperature rises.

I'm interested in if other people could check this on their beds with a thermocouple (even a meat thermometer should work) and see if they get same results I did. Maybe my bed is just defective but I doubt it.

leptun, 2020-04-29:
[image]
12V bed is doing just fine. Not sure about the normal 24V MK3 bed.

vintagepc, 2020-04-29:
This is definitely related to 2601 . There is an offset built in to the bed temperature read-back in the firmware (which 2601 reports is incorrectly also applied to the PINDA). 

It behaves exactly as you have observed, slowly adding a 5 degree offset from 40C to 50C, and then adding another 5 from 50->100.

Photogad, 2020-04-29:
> [image]
> 12V bed is doing just fine. Not sure about the normal 24V MK3 bed.

Yeah I'm using the 24v MK3 bed. IR gun is fine, but when I used mine, it was reading 4-5c hotter than the thermocouple (thermocouple more accurate). So like at 90c set temp, the thermocouple said bed was 78c, IR gun said bed was like 82-83c.   Yours looks much closer to being where it should be though.

Photogad, 2020-04-29:
> This is definitely related to #2601 . There is an offset built in to the bed temperature read-back in the firmware (which 2601 reports is incorrectly also applied to the PINDA).
> 
> It behaves exactly as you have observed, slowly adding a 5 degree offset from 40C to 50C, and then adding another 5 from 50->100.

so in layman's terms... it's "working as intended" ?

vintagepc, 2020-04-29:
> so in layman's terms... it's "working as intended" ?

More like "working as implemented". Whether that is the correct behaviour based on your evidence... remains to be decided. :wink:

vintagepc, 2020-04-29:
If you're experimental I suppose you could disable the correction (analog2bed() in temperatures.c).
Just be aware that this will also shift your PINDA temp calibrations if you do that.

Photogad, 2020-04-29:
> More like "working as implemented". Whether that is the correct behaviour based on your evidence... remains to be decided.

Well im just glad theres some sort of a cause of it found. I did email prusa support about this a while back  and they told me i need to do a firmware reset on the printer to reset the saved calibrations, but it didn't seem to me like that would fix anything at all related to the bed temperature

Photogad, 2020-04-29:
> If you're experimental I suppose you could disable the correction (analog2bed() in temperatures.c).
> Just be aware that this will also shift your PINDA temp calibrations if you do that.

Im still trying to learn how to compile my own firmware at all, lol.  Maybe ill figure it out.

I still dont understand about the pinda calibration though. I never calibrated that thing as far as im aware.

Fwiw, im using the gray PINDA (v2.2) not the latest v2.3 black pinda

3d-gussner, 2020-05-01:
@Photogad Have a look at https://github.com/prusa3d/Prusa-Firmware#using-linux-subsystem-under-windows-10-64-bit and https://github.com/prusa3d/Prusa-Firmware#using-git-bash-under-windows-10-64-bit this might help you to get a known+ stable development environment.

3d-gussner, 2020-09-21:
@Photogad As this issue got stale can you please update this or close it?

vintagepc, 2020-09-21:
Not sure what there really is to update here - there's observable code that inflates the bed and PINDA temperatures above 40C (which is the original issue). A workaround was suggested but that doesn't resolve the original problem, we need a definitive decision on what to do about the bed temp correction. It's definitely wrong for the PINDA (#2601) but I suspect it cannot/will not be changed for the bed for historical reasons.

TomaszSzt, 2021-07-28:
I am running a similar problem here. When printing ABS bed should be at 105°C for prints to stick well. My current setup (factory assembled printer, just ~200hours of run, so practically new) reports 105°C while thermocouple shows only 97°C.  The thermocouple is placed at the center of a smooth steel sheet, glued with an AL foil and thremo-transfer paste, printing head moved away.  So it is not strange, that the print does not stick.

Setting the bed temperature to 120°C gives a 107°C on thremocouple.

I am aware it is an uncorrectable problem not only due to some software or math issues but mainly due to the different heat transfer from heater to the built-in temperature sensor when compared to the heat transfer to the steel sheet and to the outside air. Correcting it in software may create issues rather than solving a problem, because it would depend on enclosure used, external temperature and etc. What I would like to see it is an ability to either edit temperature compensation curve via menu (ie. raw reading = X , real temp = Y). If done this way all internal compensations should use the "raw reading" as a reference and only a printing process should use a real-temp, corrected readings. If done this way altering compensation curve does not have to cause any re-callibrations.

The hardware solution will be tricky, and an ideal solution would include attachable temperature sensor at the steel sheet or an IR sensor. Both costly, troublesome to make and use, and prone to breaking. Managing a good heat transfer and thermal isolation would also help. Insulating heater with some glass-wool is possible, but correcting a heat transfer to a steel sheet and an external cooling effect would require using a thermal paste what defends the purpose of detachable magnetic sheets.

I can accept a solution in which the manual would clearly state that temperature is approximate and user must correct it due to differences in working environment. Plus some guide which will show how to do it. 

Notice, in such case the hard upper temperature limit must include the room for at least +/-30°C degree offset. I am writing about it because my old makerbot, which also had a heat transfer problems, rejected any settings above 130°C, so efficiently it could not go higher than 100°C.

Best regards,
  Tomasz Sztejka

closed 2023-10-17 by github-actions[bot]