prusa3d/Prusa-Firmware #2644: Printer is unable to keep temp at 205°C but 230°C works fine

oxivanisher, 2020-04-29:
**Printer type** - MK2.5S
**Printer firmware version**-  3.8.1 and 3.9 RC3

**MMU Upgrade** - No

**Describe the bug**
I am really unsure, where to post this, since I am unable to really pinpoint the source of my problem.

The general problem description would be: _The printer is unable to keep 215°_. But please read the complete post first before answering.

Environment: Prusa i3 MK2.5S; PrusaSlicer 2.2.0 (already had that problem with earlier versions).
I also did a printer reset to forget the PID and temperature tuning I did, nothing solved the problem. Also using different PLA filaments from different vendors with the same result.

Here are some Octoprint screenshots of the temperature curves:
[image]
[image]
[image]

**A**: Manually choosing different temperatures with Octoprint, no filament in extruder (fw 3.8.1): The printer reaches and stays at target temp without any problem.
**B**: Printing something small with Prusament PLA profile (fw 3.8.1): Shortly after the start, the printer drops its heater Temp to ~205°C. Manually pushing it back up with Octoprint or the printer controls does not help. Print manually cancelled.
**C**: After update to fw 3.9 RC3: Exactly the same behavior as in B. Print manually cancelled.
**D**: Manually setting the same temps as during the print C. Filament in extruder and letting extrude at max rate trough the printer controls: Temperature keeps stable.
**E**: Printing the same object with the Generic PETG profile: Works like a charm with 230°C.
**F**: Printing the same object with the Generic PLA profile (same filament as in B): Same effect as with B

First I thought it is a problem with the printer hardware, but since it seems to work under the same conditions (except movement of the axis) trough octoprint and with the same object printed in PETG, I am no longer sure. It seems as if the printer has problems holding 215°C but not 230°C. Which points to the firmware?

P.S: I also have a i3 MK3S and it does not have the same issues with gcode produced by the same slicer installations.
  
**To Reproduce**
  It happens on every PLA print I do, but I am unable to say when it began since i only realized it when a print failed several weeks ago.
  
**Expected behavior**
  Keep the temperature at 215°C.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
[g-codes.zip](https://github.com/prusa3d/Prusa-Firmware/files/4552115/g-codes.zip)

leptun, 2020-04-29:
The sad reality is that you're experiencing this bug as well: https://github.com/prusa3d/Prusa-Firmware/issues/1746
It's caused by the fan shroud design, not the firmware.

vintagepc, 2020-04-29:
Silicone socks on the heat block can help mitigate this issue (as well as reduce the damage if you make a blob-of-doom.

oxivanisher, 2020-04-30:
@leptun Yes you are 100% right.
@vintagepc Yes, this helped. But I would like to not use a sock if possible...

closed 2020-04-30 by oxivanisher