prusa3d/Prusa-Firmware #2645: [BUG]Scrolling in XYZ calibration too fast

bladel, 2020-04-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.0-RC3

**MMU Upgrade** - No
**MMU upgrade firmware version No

**Describe the bug**
Screen message over 3 screens scrolls too fast in XYZ calibration.
  
**To Reproduce**
Start XYZ calibration until message appears
  
**Expected behavior**
 Slower scrolling
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
https://drive.google.com/open?id=1TTFDEPdQaKQYRU8DXZzShttutrbS84uC

leptun, 2020-04-29:
Wow. Did you do a factory reset or did you just run XYZ calibration from the menu?

leptun, 2020-04-29:
Nevermind. Ignore my last question. Can you consistently generate the issue or is it occasional?

bladel, 2020-04-30:
No works fine now after multiple reboots.

3d-gussner, 2020-05-02:
@bladel As you cannot reproduce the issue and we didn't see that in any tests we did, would it be okay for you to close this issue?

If that kind of issue popups again we can reopen it or refer to your issue.

BTW thanks for testing the RC3 and reporting back, all feedback is welcome (hope we get once an issue that says `Everything is working`) 😉

bladel, 2020-05-05:
Yes please close it.

closed 2020-05-13 by JohnnyDeer