prusa3d/Prusa-Firmware #2647: [ENHANCEMENT] - Add warning when changing Sheet type

jweaver100, 2020-04-30:
Printer Type: MK3 FW: 3.8..1

Its currently far too easy to change the sheet/bed type without realising.

To many times I have accidentally enabled my Textured sheet when I have the Smooth installed..

This then causes print problems until the mistake is realised. For some people, who have a large Z height difference between sheets, this could even damage the bed.

**Describe the solution you'd like**

When changing the bed, a simple warning or question could be added, so that the user authorizes the change.

3d-gussner, 2020-05-01:
@jweaver100 Thanks for the enhancement idea.
Can you describe the workflow you have in mind?
Here some things to remember:
- The printer doesn't can't know which steel sheet is placed on the bed.
   - If you have an idea how the printer could determine which steel sheet is placed, please let us know.
- You could add to the End G-code in your slicer to send `M117 Note used sheet`
- And in the Start G-code at the beginning `M117 Check note last sheet`

jweaver100, 2020-05-01:
I am not suggesting an an automatic feature.. that would be impossible... in my post included a “solution i would like”.

All we need is a way to stop you from selecting the bed type without warning,... at the moment its too easy to switch between Smooth and Textured..

At best this results in failed prints... at worst damage to the bed,

So all i am asking for is a warning or confirmation when changing the bed type.

3d-gussner, 2020-05-01:
`When changing the bed, a simple warning or question could be added, so that the user authorizes the change.`

Changing the steel sheet in firmware is quite long procedure as you have to to `Settings -> HW Setup -> Steel Sheets`, so I am pretty sure you mean physically changing the sheet without choosing the correct one.

I can't see a way at this moment HOW the printer could recognize which steel sheet is on the bed.

I suggested to use End G-code to remember you to take a note which steel sheet you have used.
And Start G-code to remember you to have a look at these notes.

jweaver100, 2020-05-01:
Am I in the right group? Is this for the Mk3 firmware or Mini?

I am reporting this for the Mk3!

For some time there has been a setting on the TOP menu to change sheets and it does it without warning..

The 'top' menu is:

- Info
- Pre Heat
- Print from SD
- SHEET
- Auto Unload
- etc

When you click on 'sheet' it cycles through your saved sheets.. Its a nice feature no doubt and something I was happy to see introduced (and look forwards to the same on the Mini soon).

But if you accidently click on 'sheet' when you ment to do 'print', it toggles between sheets (in my case Smooth and Textured) without warning and sometimes you don't realise..

I have my Smooth sheet on (99% of the time) I have accidently clicked on 'sheet' without reaslising and changed to 'textured' bringing the Z down by 0.3mm.. Dangerously close to damaging the bed, and far to low for a success print.

For me its just a waste of time as I end up cleaning the bed, and retrying until I realise my mistake.. But it could be much worse if the Z height set puts the hotend low enough to scratch the bed.

So what I am requesting here is that when you select "sheet" it doesn't just toggle from Smooth to Texture and visa verse.. It asks for a confirmation.

awenelo, 2020-05-02:
So, if I understand correctly, you want to have it so that when the "sheet" option is clicked in the main menu, instead of immediately changing sheets, you want to go to a confirmation screen asking whether or not the user intended to execute that command. The one problem that I can see with that is a user having 8 different steel sheet profiles set(the maximum). In this case, if they wanted to change from number 1 to number 8, they would have to click "yes" seven times.

The one feasible solution that I can think of would be to send the user to a menu listing all active sheets that when clicked would then ask if the action was intended, then switch. This would make it generally easier for users with many sheets, along with add the requested feature.

leptun, 2020-05-02:
The whole point of the toggle of sheets in the main menu was to quickly toggle between all sheets. Adding additional messages only makes the process slower and makes it useless.

3d-gussner, 2020-05-02:
@jweaver100 Sorry that i din't understand your issue at the beginning. I factory reset my printer so often for testing that i missed that the `Sheet    [abcdefgh]` is only added as soon you have more than one sheet with Z offset defined.

I agree with @leptun that the toggle between all sheets should be fast. Confirming every time you change the sheet isn't user friendly.

As the `Sheet    [abcdefgh]` menu is close to the `Print from SD`  i can imagine that there is a chance to hit it by accident when you try to choose `Print from SD`.

I am not sure if placing `Sheet    [abcdefgh]` to another location would be useful.

jweaver100, 2020-05-02:
I get that it needs to be quick.. But I would prefer a 2nd button press if it ment saving a print or even my bed.. This could do HORRIFIC damage to a bed.

Its not that I am describing something unlikely to happen.. it happens to me all the time.. Mainly beacuse the encoder wheel clicks don't always line up with the menus.. So sometimes i click on "print" and accidentally hit "sheet" instead.. Most of the time I notice, which is fine.. But many times I havn't and have then spent time trying to work out why my printing has gone bad...

And as I keep saying, you only want the Z height for one of your sheets to be too low and this could do damage to the bed.

So personally, I would prefer an extra click each time I change bed, than risk damage..

I am not talking about a long drawn out process to change the sheet.. Perhaps just a 2nd click? .

Or perhaps move the option to the bottom of the menu?? 

I know that people have many sheets and the option is designed to cycle through them.. But perhaps after clicking through and then stopping, you get a final warning whihc youy have to click?

Or perhaps just a warning when you change?? 

I don't know.. It just seems too easy to change beds at the moment..

awenelo, 2020-05-02:
Could we add an option in settings to enable/disable a warning? That way, if the user wanted to have the warning, they could while not affecting users who would like to have a single-click switch.

3d-gussner, 2020-05-02:
It's one option to add that, but needs additional code and eeprom space. 
Another could be to move the sheets menu away from the `Print from SD`

Any other ideas?

@ everyone what would you prefer?

jweaver100, 2020-05-02:
If there is no room in the FW it seems a waste to add code.. Moving it down the list would certainly help.. Right now its slap bang in the middle of options that is used all the time.

I would say between Settings and Calibration would be best?

gidthup, 2020-09-08:
I only use the SD card option to print (not the direct connection interface) and my motto is safety first. I switch often between sheets and changing from Coated to PEI is a risk as my Z-adjustment has to go from -0.470 to -0.250.  Last week the printer got me as I could not quickly find the Z-adjustment option in the menu after I ealised a started the print without changing the z-adjustment.  That left a nasty mark at the PEI bed -:(.

I have put the based settings for z-adjustment on a sticker on the printer.  A simple reminder/confirmation directly after giving the 'Print from SD' command, stating 'Z-adjust is  0.nnn.  Continue ? would already be a good reminder.   Another option is to automatically jump to the Z-adjustment option after giving start print.  This is a reminder and also a way to directly adjust the setting before the nozzle gets near the bed.  You press OK to continue with/without adjustment, or it automatically times out and goes to the Info screen 5/10 seconds after the printing has started.

jweaver100, 2020-10-29:
Is there any further thoughts about this? I just noticed that my Sheet was set to "Textured", so i obviously clicked it by accident at some point.

And now I have gone back to several days of back to back prints (where the bottom surface is critical) I have 2 days of ruined prints.

I can't believe I am the only person to suffer with this? Its so easy to change by accident when you mean to hit the "Print from SD" option.. Please can you consider changing it.. Either moving the option to the end of the list.. or just adding confirmation..

adamsatyr, 2023-09-23:
Sorry for necromancy, but I have similar feature request:

Display a message at the start of the print with currently selected sheet type (and maybe other critical info) and confirm before start of the print.

Then I can quickly compare the states between printer settings and reality.

3d-gussner, 2025-02-12:
See https://github.com/prusa3d/Prusa-Firmware/pull/4832

closed 2025-04-21 by github-actions[bot]