prusa3d/Prusa-Firmware #2648: Problem compiling firmware v3.2.3 - LCD shows boxes/blocks

jcarlo471, 2020-05-01:
Hey guys,

I'm trying to use the old firmware v3.2.3 that originally worked well for me. How do I go about this as I am receiving LCD problems (see photo below or [link to photo](https://imgur.com/k9dPV6m)).

Details:
- Using Firmware v3.9.0, I receive Motorend Stop YZ problems during my XYZ calibration. i.e whilst the Y axis is still being checked, LCD shows immediately as Checking Z-axis afterwards the Z axis lifts and continues upward thus leading to the error (for reference it should be going down per observations from v3.2.3 calibration.
- The error above is from recompiling firmware v3.9.0 (per instructions from readme.me).

If the firmware on v3.2.3 is loaded, the above does not happen and XYZ calibration works smoothly
[MK2 v3.2.3](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.3). I've tried compiling this firmware with edits to the z_max definition, however when it's flashed my LCD screen shows up with each display field blocked out (see photo below or [link to photo](https://imgur.com/k9dPV6m))

I've followed the instructions in the readme for both v3.2.3 and [MK3 Windows build->Arduino](https://github.com/prusa3d/Prusa-Firmware). Where the v3.2.3 lacks detail I've referred to the latest readme v3.9.0, for additional information.

Notes:
- Using Einsy13a board
- Ardiuno v1.8.5
- Removal of Liquid Crystal
- PrusaResearchEinsy RAMbo board compiles with no errors
-Rambo board compiles with:
"WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Build options changed, rebuilding all
Warning: platform.txt from core 'Marlin AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Sketch uses 254232 bytes (98%) of program storage space. Maximum is 258048 bytes.
Global variables use 6030 bytes of dynamic memory."

Any help would be most appreciated.

Regards,
[image]

floridaservices, 2020-05-01:
I have worked with 3.2.3 quite a bit in the past and i'm currently running 3.9.0 RC3.  Per Prusa's recommendation I use Arduino 1.8.5.  I have found that the latest Arduino version throws compile errors with the Prusa firmwares- but I cannot remember the exact errors at the moment.  If you do try the Arduino route be sure to: set the secondary language to 0 in config.h, and most importantly, and maybe this is where you are running into issues, **compile using board Rambo (not Prusa Einsy Rambo).**  The technical reasons behind the custom Prusa  board definition did not exist back in the 3.2.3 days so the Original Rambo should compile fine and is what was used back in 3.2.3 times.

jcarlo471, 2020-05-01:
> I have worked with 3.2.3 quite a bit in the past and i'm currently running 3.9.0 RC3. Per Prusa's recommendation I use Arduino 1.8.5. I have found that the latest Arduino version throws compile [...]

Thanks for the reply.
I've tried each variation of compiling i.e. with Prusa board definition and the regular RAMbo.
Below is the v3.2.3 RAMbo  output log - TLDR it doesn't compile.
```
avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\jcala\Google Drive\3D\Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\jcala\Google Drive\3D\Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex"
avrdude-slic3r: writing flash (254232 bytes):
avrdude-slic3r: stk500v2_command(): command failed

Writing | ################################################## | 100% 43.14s

avrdude-slic3r: 254232 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\jcala\Google Drive\3D\Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex:
avrdude-slic3r: load data flash data from input file C:\Users\jcala\Google Drive\3D\Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex:
avrdude-slic3r: input file C:\Users\jcala\Google Drive\3D\Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex contains 254232 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 31.78s

avrdude-slic3r: verifying ...
avrdude-slic3r: verification error, first mismatch at byte 0x3dbc6
                0x48 != 0x49
avrdude-slic3r: verification error; content mismatch

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

```

floridaservices, 2020-05-01:
I noticed you have your files in a google drive folder.  This could be problematic depending on how your google drive sync is set up.  I would move the files to the root into a basic folder.  You are getting verification errors and file mismatches - this seems to be a host configuration issue.

jcarlo471, 2020-05-01:
I've relocated files to the desktop. The results are the same when using the RAMbo board manager. 
If these are a host configuration issue, what other Arduino hardware or library modifications are required other than those instructed in the Prusa readme's, i.e.:
-Remove Liquid Crystal library from arduino (Original Prusa i3 MK2 Step 2)
-Copy Arduino addon hardware ->marlin to root Arduino folder (Original Prusa i3 MK2 Step 3)
-Copy Arduino addon libraries->L6470 & TMC26XStepper to root Arduino folder (Original Prusa i3 MK2 Step 3)
```
avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\jcala\Desktop\Prusa i3 Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\jcala\Desktop\Prusa i3 Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex"
avrdude-slic3r: writing flash (254234 bytes):
avrdude-slic3r: stk500v2_command(): command failed

Writing | ################################################## | 100% 43.77s

avrdude-slic3r: 254234 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\jcala\Desktop\Prusa i3 Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex:
avrdude-slic3r: load data flash data from input file C:\Users\jcala\Desktop\Prusa i3 Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex:
avrdude-slic3r: input file C:\Users\jcala\Desktop\Prusa i3 Firmware\Prusa-Firmware-3.2.3\Firmware\Firmware.ino.rambo.hex contains 254234 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 32.30s

avrdude-slic3r: verifying ...
avrdude-slic3r: verification error, first mismatch at byte 0x3c072
                0xe6 != 0xe7
avrdude-slic3r: verification error; content mismatch

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

```

3d-gussner, 2020-05-01:
@jcarlo471 When you open an issue please do not remove the template information, these are crucial information we need.

```
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.
``` 

Can you please edit/update your initial issue report.

3d-gussner, 2020-05-01:
### Tips to compile the firmware yourself:

If you have a MK2.5/s or MK3/s and want to compile the firmware yourself please try to use `PF-build.sh`
https://github.com/prusa3d/Prusa-Firmware/tree/MK3#using-linux-subsystem-under-windows-10-64-bit
or 
https://github.com/prusa3d/Prusa-Firmware/tree/MK3#using-git-bash-under-windows-10-64-bit

If you have a MK2 you still can use this as the script creates a know stable compiling environment.
Open the Arduino IDE `PF-build-env-1.0.6\windows-64\arduino.exe` to compile your MK2 firmware.

3d-gussner, 2020-05-01:
Okay you have an MK2 not an MK2.5/s or MK3/s:
- the latest firmware for the MK2 is https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.3
- You need to use the https://github.com/prusa3d/Prusa-Firmware/tree/MK2 branch for the MK2

3d-gussner, 2020-05-01:
Also after reflashing MK2 firmware please do a [Factory Reset --> All Data](https://help.prusa3d.com/en/article/factory-reset_2216/) as you got some MK2.5/s or MK3/s eeprom values written to it.

jcarlo471, 2020-05-01:
> ### Tips to compile the firmware yourself:
> If you have a MK2.5/s or MK3/s and want to compile the firmware yourself please try to use `PF-build.sh` [...]

This is a definitely out of my depth, i'm not familiar with git bash and am following the instruction word for word. 
Could you shed some light?
I'm not sure why the link isnt being created.
[image]

3d-gussner, 2020-05-01:
Give me some minutes maybe i can modify the PF-build.sh for MK2 branch...

3d-gussner, 2020-05-01:
Did you try a Factory Reset --> All data and then flash the release hex file?

I modified the `PF-build.sh` that should work for the MK2 called `PF-MK2-build.sh` difference that it uses an "older" Arduino Boards definition which is needed for that branch.

Download it, rename it, unzip and give it a try.
But best would be to try to flash https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.2.3/prusa3d_fw_mk2-3.2.3+MK2.415.201906041057.zip first.
[PF-MK2-build.zip.txt](https://github.com/prusa3d/Prusa-Firmware/files/4565220/PF-MK2-build.zip.txt)

jcarlo471, 2020-05-02:
``

> Did you try a Factory Reset --> All data and then flash the release hex file?
>
> I modified the `PF-build.sh` that should work for the MK2 called `PF-MK2-build.sh` difference that it uses an "older" [...]

General steps were:
1. Loaded original working v3.2.3.hex file from Prusa (to remove LCD problem)
2. Factory Reset -> All
3. Unzip Prusa-Firmware-3.2.3.zip
4. Modify .h file.
5. git bash compile
6. Flash via PrusaSlic3r-2.2.0
7. Problem with LCD screen during flash, is there a way to reset this, as I may have loaded the bootloader.hex by mistake.
[image]
8. Flash is complete and the LCD screen is still problematic.
9. Re-flash original Prusa v3.2.3.hex file
10. Repeat steps 2-6 using the .hex you provided.
-For clarity what is the difference in the .hex you provided?
11. LCD problem is gone, seems functional.
12. Factory reset prior to initial calibration 
13. Calibration begins
With the 'PF-MK2-build.zip.txt', just confirming so there is no confusion, it is a simple file extension change from ".zip.txt" to ".sh" ?
Appreciate your all your efforts so far gussner.

3d-gussner, 2020-05-02:
@jcarlo471 Good morning.

So the official 3.2.3 hex file flashed to the printer is working fine, correct?
Your modified and self compiled firmware creates an issue on LCD?

What if you compile an **unmodified** MK2 3.2.3 firmware and flash it, does it work or do you still have the LCD issue?

What do you change in the firmware source code?

jcarlo471, 2020-05-02:
> @jcarlo471 Good morning.
>
> So the official 3.2.3 hex file flashed to the printer is working fine, correct?
> Your modified and self compiled firmware creates an issue on LCD?
>
> What if you compile an [...]

Good evening @3d-gussner ,
1. Correct official 3.2.3 hex works fine.
2. Correct the self compiled creates an issue.
3. I came to the same conclusion and tried compiling an unmodified file. Results remain the same, inclusive of factory reset before changing and after flashing.
4. I am merely changing 1 line in Configuration_Prusa.h/einsyrambo13a
Z_max = 220 to 202 (Volcano eruption upgrade, resulting in lower z-height i.e. crashing in the bed), LOL all this effort for the one line.

3d-gussner, 2020-05-02:
@jcarlo471 Okay I see ... long time ago that I compiled something for the MK2.
Can you try to follow these instructions https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware/c5ba472259baeb6f92273ea860e00d8a7a02856c

These worked very well in the past for the MK2 branch.

It is a pity that the MK2 branch just gets "critical" patches only and is on hold.
But on the other hand my MK2 is still one of my work horses to print.

I like the MK52 bed with the spring steel sheet but I never understood why should I get rid of the MK42 bed as it is sometimes better than the MK52 (no warped steel sheet with difficult material).

3d-gussner, 2020-05-02:
BTW: I have a filament runout sensor on my MK2 for long long time, see
https://github.com/3d-gussner/Prusa-Firmware/releases?after=v3.1.2

closed 2020-05-03 by jcarlo471

3d-gussner, 2020-05-03:
Could you compile your own firmware?

jcarlo471, 2020-05-04:
> Could you compile your own firmware?

Sadly no, I just gave up and ended up using v3.9.0 RC3 as it was the best workaround.
As I have a MK3S coming in, when it ships. I'll most likely retire the MK2 until I get the bear upgrade parts, to which I'll get the upgrades to the 2.5/S.
Really appreciate your time and efforts on this @3d-gussner.
Stay safe and take it easy mate.

pboschi, 2020-10-13:
By the way, I have the same identical issue. Plus I had to use the following board definition, or the built image was too big and causing verification issue during flash:
https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json

JCalaunan, 2020-10-24:
Thanks! @pboschi will implement it and see how i go.