prusa3d/Prusa-Firmware #2649: [BUG]3.9.0 RC3 filament sensor test fails on calibration wizard

xarbit, 2020-05-01:
**Printer type**
MK2S

**Printer firmware version**
3.9.0 RC3

**MMU Upgrade**
MMU2S

**MMU upgrade firmware version 
1.0.6

**Describe the bug**
After doing the wizard, the filament sensor test will fail. The only way to recalibrate is to do all steps manually. 
  
**To Reproduce**
  Go to Calibration->Wizard and start it. Filament sensor test will fail.
  
**Expected behavior**
To pass the filament sensor test.

3d-gussner, 2020-05-01:
@xarbit Thanks for the bug report.

Is your printer a MK2S ??? This firmware FW3.9.0-RC3 is only for MK2.5/s and MK3/s
The MK2S latest firmware is https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.2.3

3d-gussner, 2020-05-01:
@xarbit I assume you have an MK3S as you also have the MMU2S.  Is the MK2S is a typo, if so please edit your initial issue and enter the correct printer.

Can you please connect your printer via USB to Pronterface or Octoprint? Would be helpful to get some serial/terminal output.
- Check under` Support --- Fil. sensor v.:`  which IR sensor is detected
   - ` unknown state`
   - ` 0.3 or older`
   - ` 0.4 or newer` 
- run in Pronterface/Octoprint `D3 Ax0d48 C1`
  - 00  means it is 0.3 or older
  - 01 means it is 0.4 or newer
  - ff means it is unknown (like after Factory Reset -> All Data)
- Go to `Settings -> HW Setup -> Fsensor Detection`, this should guide you through to detect which sensor version you have.

Serial/terminal data would be great and also if you could report these back here.

xarbit, 2020-05-01:
@3d-gussner sorry, yes that was a typo. 
I have the newer 0.4.  
Just for understanding, you want me to redo the wizard then check those positions? .

3d-gussner, 2020-05-01:
@xarbit First thanks for updating the Printer!

Don't go to the wizard just try to follow the instruction I wrote above.

3d-gussner, 2020-05-01:
You got the red one https://help.prusa3d.com/en/guide/5-e-axis-assembly_28536 right?
Or was the 0.4 an info from the firmware.

xarbit, 2020-05-01:
@3d-gussner 
sure, here we go

> Check under Support --- Fil. sensor v.: which IR sensor is detected 

0.4 or newer

> run in Pronterface/Octoprint D3 Ax0d48 C1

```
Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x7f51abf839d0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: 3.9.0-RC3-3401
Recv: echo: Last Updated: Apr 28 2020 12:19:39 | Author: (none, default config)
Recv: Compiled: Apr 28 2020
Recv: echo: Free Memory: 1667  PlannerBufferBytes: 1792
Recv: echo:Hardcoded Default Settings Loaded
Recv: adc_init
Recv: CrashDetect ENABLED!
Recv: FSensor ENABLED (sensor board revision: 0.4 or newer)
Recv: Sending 0xFF
Recv: echo:SD card ok
Recv: Error:No Line Number with checksum, Last Line: 0
Send: N0 M110 N0*125
Recv: Resend: 1
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok
Send: N2 M20*19
Recv: FIRMWARE_NAME:Prusa-Firmware 3.9.0-RC3 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Recv: Begin file list
....
Recv: End file list
Recv: ok
Send: M105
Recv: ok T:31.6 /0.0 B:30.6 /0.0 T0:31.6 /0.0 @:0 B@:0 P:29.2 A:32.2
Recv: MMU => 'start'
Recv: MMU <= 'S1'
Send: M105
Recv: ok T:31.6 /0.0 B:30.7 /0.0 T0:31.6 /0.0 @:0 B@:0 P:27.4 A:32.1
Recv: MMU => '106ok'
Recv: MMU <= 'S2'
Recv: MMU => '372ok'
Recv: MMU version valid
Recv: MMU <= 'M1'
Recv: MMU - ENABLED

Send: D3 Ax0d48 C1
Recv: D3 - Read/Write EEPROM
Recv: 0d48  01
Recv: ok
```

> You got the red one https://help.prusa3d.com/en/guide/5-e-axis-assembly_28536 right?

Yes

3d-gussner, 2020-05-01:
Thanks a lot for the serial log!!!

Did you run ?

> * Go to `Settings -> HW Setup -> Fsensor Detection`, this should guide you through to detect which sensor version you have.

Please also check under `Support -->Sensor` which values are shown 
- when **no** filament is in the extruder
- when filament passed the sensor

Also interesting would be to check `Support --> Voltages` which values are shown  
- when **no** filament is in the extruder
- when filament passed the sensor

Can you run the `Calibration -> Selftest ` (with serial log on)

Does the printer behaves like it should you expect now?

xarbit, 2020-05-01:
> Did you run ?
Go to Settings -> HW Setup -> Fsensor Detection, this should guide you through to detect which sensor version you have.

Yes, I did that after I figured the wizard is not working.

> Please also check under Support -->Sensor which values are shown

I do have the mmu2 unit and the chimney so it really does not pass anymore. But I did verify that by pressing on the screw that holds the door, making the sensor trigger.  
That works fine. 

> Also interesting would be to check Support --> Voltages which values are shown

voltage default 4.4V and 0.4V when triggered. 

> Can you run the Calibration -> Selftest (with serial log on) 

sure will follow.

> Does the printer behaves like it should you expect now?

Yes, I did a cube print and all fine.

xarbit, 2020-05-01:
@3d-gussner here is the output from the self calibration test. It is weird that everything passed using that routine.

[output.txt](https://github.com/prusa3d/Prusa-Firmware/files/4564478/output.txt)

3d-gussner, 2020-05-01:
Need to check if the wizard is doing some additional things for the sensor.
Waiting for the sensor info and voltage info, before we continue.

xarbit, 2020-05-01:
> Need to check if the wizard is doing some additional things for the sensor.
> Waiting for the sensor info and voltage info, before we continue.

ok, that information you are waiting for I posted one comment above. 
voltage default 4.4V and 0.4V when triggered. The sensor works fine (see above)

3d-gussner, 2020-05-01:
Is that the 1st firmware you run on the printer? Did you do a factory reset with all data?

3d-gussner, 2020-05-01:
The 4.4V and 0.4V are in the range to detect the `0.4 or newer` IR sensor so that is fine.
Can you rerun and send the serial log of

> * Go to `Settings -> HW Setup -> Fsensor Detection`, this should guide you through to detect which sensor version you have.

xarbit, 2020-05-01:
no it is not the first first firmware.. I had the 3.8.1 on there before. 
no I did not factory reset, and copied that I shouldn't ..

xarbit, 2020-05-01:
> The 4.4V and 0.4V are in the range to detect the `0.4 or newer` IR sensor so that is fine.
> Can you rerun and send the serial log of
> 
> > * Go to `Settings -> HW Setup -> Fsensor Detection`, this should guide you through to detect which sensor version you have.

```
Recv: echo:busy: paused for user
Recv: Measured filament sensor high level: 4.39V
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: echo:busy: paused for user
Recv: Measured filament sensor low level: 0.43V
```

3d-gussner, 2020-05-01:
There is a known bug with IR sensor detection after Factory Reset that will be fixed in Final. But as it is a rare thing we didn't want to delay the RC3 release.

Thanks again for the serial log, these values are totally in the range they should be.

3d-gussner, 2020-05-01:
I may have an idea why you got an issue during Wizard run.

You have a MMU connected and the Wizard has some "special" code for MK2.5/s or MK3/s with MMU.

Could you please re-run and re-confirm that you have an issue with the Wizard only?

Could you also force a filament runout (cut above the chimney) and restore it during a tiny mmu print (single material is enough).

Again BIG THANKS for reporting that, we got an good idea where to look.
We will investigate this issue and get hopefully a fast bugfix until final.

3d-gussner, 2020-05-01:
DON'T cut it above the chimney!!!

3d-gussner, 2020-05-01:
Sorry  for that, this is something you can do with "normal" extruder as the gears still have something to grab on to pull it out of the nozzle.
Cutting at the chimney would trigger after the filament passed the gear.

xarbit, 2020-05-01:
great, i was also thinking there is probably something special with the MMU routine. 

I started a print in the meantime :-) sorry.. I will re-test that again after my print is done in a few hours and report back. 

For what it is worth, I been using the RC3 firmware for a few days already so I am pretty sure it only happens on the wizard.  When I first encountered this issue, I did try it a few times with no luck.

leptun, 2020-05-01:
@xarbit I tried to run the Selftest with the v0.3 IRsensor myself. On the first try, I got the following error message:
```
Selftest error !
Please check :
Filament sensor
False triggering
```
This means that the IRsensor detected filament when it shouldn't just by spinning the extruder. This usually happens if there is some debris in the Bondtech gears. I removed the idler and I was able to notice quite a thick layer of gunk in the gears themselves. Probably greasing them just made it more prone to collecting filament dust. I also noticed that the gear on the stepper motor was loose, so I tightened that as well.
After I cleaned everything, I ran the Selftest again. This time it passed and I also noticed that the extruder made much fewer rattling noises, which is good.

Please check if this is also your issue.

3d-gussner, 2020-05-14:
We have a temporary fix for that https://github.com/prusa3d/Prusa-Firmware/pull/2660
At this moment:
- MK3s users updating from FW3.8.1 to FW3.9.0 need to run `Setting -> HW Setup -> Fsensor Detection` to detect which IR sensor is installed.
- MK3s+MMU2s users need to run the `Selftest` to set the the EEPROM value for the IR sensor.
  Also cleaning the gears from debris is always a good idea.

We are working on an improved version for the IR sensor, this will be in FW3.9.1.

3d-gussner, 2020-07-13:
@xarbit I think we fixed this issue can you please confirm and close this issue?

3d-gussner, 2020-09-21:
As this issue should be fixed and got stale, we gonna close it.
[stale][fixed]

closed 2020-09-21 by DRracer