prusa3d/Prusa-Firmware #2650: [BUG] 3.8.1 Power panic not working properly when using UPS (causes Thermal Runaway)

GunSmoker, 2020-05-01:
Prusa MK3S (silver PSU), 3.8.1 firmware (Bondtech).

When printer is connected to UPS, and UPS switches to battery - printer behave weirdly: both heaters are turned off, while printer itself continue to operate normally (no power panic is triggered). If this continues for long enough - a thermal runaway error will be triggered. Once UPS switches back to mains - heaters are turned on.

The reason seems to be power panic disabling heaters. However, it is not clear why printer itself continues to operate without triggering on power panic. Just to clarify: output voltages of PSU are 24.26V for both cases (running from wall socket and running from UPS battery).

The workaround is to unplug power panic sense cable from main board. 

This seems to be a common issue. For example: 
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/heaters-disabled-power-panic-sense-fault/
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/thermal-runaway-when-running-from-ups/
  
Steps to reproduce:
1. Plug printer into UPS (I am using APC BR900G-RS).
2. Turn UPS and printer on.
3. Use Preheat -> PLA menu item.
4. Expected and observed: bed LED will turn on.
5. Use Preheat -> Cooldown.
6. Unplug UPS from wall socket. UPS will switch to battery.
7. Use Preheat -> PLA menu item.

Expected: bed LED will turn on.
Observed: bed LED is off, temp is not raising, after some time - thermal runaway will be triggered
  
8. Turn printer off.
9. Unplug black/white power panic sense cable from the board.
10. Turn printer back on (UPS is still running from battery!).
11. Use Preheat -> PLA menu item.
12. Expected and observed: bed LED will turn on.

3d-gussner, 2020-05-01:
@GunSmoker 
Does the Power panic work when the printer is attached to wall outlet and you pull (not switch off) the power?

GunSmoker, 2020-05-02:
@3d-gussner Power panic was "working" before. And by "working" I mean "detecting power loss". 

I have installed UPS because power panic was not usable: it results in layer shift. It also seems to be a common issue. For example:
#2311
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-general-discussion-announcements-and-releases/layer-shift-on-power-panic./
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/power-panic-homing-failed-with-massive-layer-shift.-lost-a-14-hour-print./
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-prusa-i3-kit-archive/layer-shift-after-power-panic-b150/

I will be able to do more tests, but now right now...

GunSmoker, 2020-05-02:
Few notes to pay attention:
1. The problem is NOT the moment when UPS switches to the battery. If you connect powered off printer to UPS that is already running from the battery - issue repeats.
2. This or similar issue existed in FW 3.3.0 or possibly even earlier, see: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-general-discussion-announcements-and-releases/mk3-firmware-3.3.0-power-panic-triggers-when-on-a-ups/

GunSmoker, 2020-05-02:
Apologies. This issue seems to be a duplicate of these issues:
#898
#1161
For some reason I have failed to find these before.

closed 2020-05-02 by GunSmoker

3d-gussner, 2020-05-03:
@GunSmoker Thanks for looking again for duplicates and closing the issue, that really helps!