prusa3d/Prusa-Firmware #2651: [BUG] Weird behavior if print is started during filament auto-load

n3o59hf, 2020-05-02:
**Printer type** - MK3S (kit, no hardware modifications except cosmetics)
**Printer firmware version**- 3.9.0-RC3

**MMU Upgrade** - none
  
**To Reproduce**

-   Printer on, no filament inserted, hot-end temperature ready for filament insert, filament autoload is on, SD-card with printable files are in.
-   Start inserting filament
-   Navigate to SD-card menu for any print and start print (press knob) while filament is being pulled in
  
  It results in short motor movement (including repeated ramming into end stop, as shown in video, have seen both X and Y movements, X in both directions) and after that sometimes printer restarts, sometimes it beeps (even if set to silence) and display "Inactivity Shutdown" and locks down.

**Expected behavior**

  Print is started, home screen is displayed. After filament is loaded - prompt about filament load and color is displayed, print is continued. Alternatively error is displayed that print can't be started during filament load or even SD-card menu is not accessible.

**Video**

https://www.youtube.com/watch?v=PC1uv23G0-E

3d-gussner, 2020-05-02:
@n3o59hf Thanks for the Bug report.
Can you try to move the extruder more to the right before starting ? Just want to see if it also fails.

n3o59hf, 2020-05-02:
It crashes sometimes:

https://www.youtube.com/watch?v=yTPLieokdE4

other times it ends filament load, does not show confirmation for filament load, and shows "0 hours 0 minutes" time like it would just have finished print immediately:

https://www.youtube.com/watch?v=P-2z0xQ5-qU

3d-gussner, 2020-05-03:
That is strange.

Can you please re-flash the firmware and post the Advanced: Output log.

If that doesn't help try a [Factory reset --> All Data](https://help.prusa3d.com/en/article/factory-reset_2216) , if you don't mind to loose your print statistics.

If you want to keep the stats please report back.

Panayiotis-git, 2020-05-03:
It always looked strange to me, that when the filament is loaded or auto-loaded, instead of a blocking screen informing the user that "Loading filament..." (like the M600), it returns to the status screen allowing the user to navigate to other menus. It is even allowed to unload the filament, which in fact works OK after completing the loading procedure.

leptun, 2020-05-03:
I also find it weird, but I guess they had their reasons to do it this way when it was implemented.
Btw, is the behavior different in the MK2 firmware?

n3o59hf, 2020-05-09:
@3d-gussner It does not look like firmware update issue (anyone else have tried to reproduce this behavior?). Printer prints ok when started while not auto-loading. I can try firmware re-flash but would prefer not to do factory reset.

Panayiotis-git, 2020-05-10:
I've reproduced it. For me, the printer resets.

MohanMaker, 2020-09-07:
I have the same problem here with 3.9.0 on MK3S. No motor movement like others have described, but the printer restarts/resets itself.

tehcaster, 2020-12-26:
I found a similar bug that might be actually the same thing as this one. Instead of inserting filament, move Z axis over a long range (e.g. full top to full bottom) with quick knob rotation, exit the move screen, and while the axis is still moving, start a print from SD. The printer then restarts. I have found this by accident on 3.9.1, then repeated it to see if it's reproducible. It was, and I also saw an error message about static data being overwritten, before the restart. Later I noticed that my print stats became obviously corrupted - the filament counter was 12km instead of expected ~2km, and print time was changed by 50 days over cca 40 days (not running 24/7 anyway). I suspected this was related to the crashes, so third time I took a picture of the stats before and after, and also recorded the whole thing on video - https://photos.app.goo.gl/16GT5vvX96W5wycJ6 and indeed both counters got a bump (800m of filament, 39 days of printing).
To sum up:
- crash is reproducible
- how much stats get bumped seems to be nondeterministic

As my stats are already busted, I can try anything that involves their reset now. If there's a way to reset to non-zero values, I wouldn't mind doing that aftewards, to at least be somewhat close to reality (something like 50 days printing, 3000m of filament).

I can also try if the filament autoload variant screws the stats too.

wavexx, 2021-04-28:
@tehcaster I can reproduce this even on 3.10 :(. Could you log a separate issue for it? The scenario is slightly different internally.

closed 2023-07-26 by 3d-gussner