prusa3d/Prusa-Firmware #2652: [QUESTION]Why do we use a full byte to store bool values?

awenelo, 2020-05-03:
In looking through the table in eeprom.h, I noticed that when storing bool values, we use a full byte, while we only need to store a bit of data. Is there some reason we do this, or is it just done to make storage of values easier?

leptun, 2020-05-03:
Some settings could be more packed, especially the ones that only require a on/off option. But doing the optimization now is not really an option. We would need to not only restructure the entire eeprom data, but also handle all the backwords compatibility nonsense, which we really can’t do on this board. Also, packing more bits becomes complicated when you take into account that you don’t always know when values are uninitialized (erased eeprom, all 0xFF) or all bits are just set for all 8 settings in that byte.

awenelo, 2020-05-04:
When reading a boolean value from EEPROM, the only one bit is read (from my testing, bit 7). When writing a boolean value, the only possibilities are 255 or 0. Because of that, we could determine whether or not the settings have been compacted or not. This could be done by, when first compacting the values, setting bit 0 to the opposite of bit 7, which can't be done when storing boolean values. Using these two methods,  we can add more settings while keeping backwards compatibility. We would keep the same address for pre-existing settings, but only using bit 7, not the entire byte, while inserting new settings in 1-6 and setting it 0 to the opposite of 7.

leptun, 2020-05-04:
I understand that it is wasteful. There are ways to make it much more efficient, but now it is too late to do so. The firmware must be able to run just fine even when updating from 3.0.0-MK2 to a latestFW-MK2.5S.

Also, the eeprom settings are stored in the exact same place across multiple printer versions, so there are gaps in the eeprom storage where MK3 settings would be on a MK2.5S printer. This would be one place to optimize if it was needed, but considering that less than 50% of EEPROM is used on all printers, this isn't of concern.

awenelo, 2020-05-04:
Ok, that makes sense.

closed 2020-05-04 by awenelo

DRracer, 2020-05-27:
@awenelo Btw. I tried to "save" RAM recently by merging bool variables into bytes (8 bits in each byte). Lost ~500B of FLASH, because even the most optimized approach of updating/reading one bit needs 3 instructions, like:
`LDS R24, byte_with_bool_flags
ANDI R24, 0x4 ; mask the bit
STS R24 ; store the byte back or jump accordingly`

And it only saved ~30B of RAM