prusa3d/Prusa-Firmware #2655: [ENHANCEMENT] Allow to change print speed while not printing

yschroeder, 2020-05-03:
When I change the print speed in the `Tune` menu during a print, this speed also applies for subsequent prints. If I do not want to print the next print with the increased speed as well, I have to wait until the new print starts for the `Tune` menu to become available and set the speed back to 100%.

I would like to be able to set this speed while NOT printing (when the printer is idle).

leptun, 2020-05-04:
But you already can do that. Just spin the knob while in the status screen.

yschroeder, 2020-05-04:
Interesting I did not know that... Anyway, the `Tune` menu could be avialable while not printing, right?

leptun, 2020-05-04:
Don’t really see the point in doing so. Most, if not all options in the Tune menu are also available in the Settings menu. The Settings menu just has extra options that are not available during the print

yschroeder, 2020-05-04:
Could we then still integrate the speed setting into the settings menu? Then we have a named option for this (for completeness).

closed 2023-10-25 by github-actions[bot]