prusa3d/Prusa-Firmware #2658: [BUG] Error compiling MK3(S) on OSX 10.15.4 Catalina using Arduino 1.8.12

salmander, 2020-05-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S

**Describe the bug**
Getting this error while trying to compile the firmware on mk3 branch.

```
Arduino: 1.8.12 (Mac OS X), Board: "PrusaResearch Einsy RAMBo"

In file included from sketch/Configuration.h:60:0,
                 from sketch/Configuration.cpp:1:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from /Users/salmanahmed/workspace/prusa/Prusa-Firmware/Firmware/Firmware.ino:33:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Servo.cpp:44:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Dcodes.cpp:3:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/SdFile.cpp:20:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/BlinkM.cpp:5:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/Sd2Card.cpp:20:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/MarlinSerial.cpp:23:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/ConfigurationStore.cpp:3:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/backlight.h:6,
                 from sketch/backlight.cpp:3:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/SdFatUtil.cpp:20:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/SdBaseFile.cpp:21:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/SdVolume.cpp:20:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/cmdqueue.h:4,
                 from sketch/cmdqueue.cpp:1:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin.h:21,
                 from sketch/cardreader.cpp:1:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sketch/Configuration.h:60:0,
                 from sketch/Marlin_main.cpp:47:
eeprom.h:583:55: error: 'reinterpret_cast<M500_conf*>(20)' is not a constant expression
 static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
'reinterpret_cast<M500_conf*>(20)' is not a constant expression

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```
  
**To Reproduce**
1. Clone the repo.
2. Copy `1_75mm_MK3S-EINSy10a-E3Dv6full.h` from the variants directory to the `Firmware` directory.
3. Overwrite `Configuration_prusa.h`
4. Open `Firmware.ino` in Arduino 1.8.12.
5. Compile
  
**Expected behavior**
To compile successfully.

leptun, 2020-05-09:
Latest supported arduino ide version is 1.8.5

salmander, 2020-05-09:
Hey, thanks for the swift response. I have already tried to compile on "Arduino IDE 1.8.5" but I am getting a different error. Below is the error on 1.8.5:

```

Arduino: 1.8.5 (Mac OS X), Board: "PrusaResearch Einsy RAMBo"

fork/exec /Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++: bad CPU type in executable
Error compiling for board PrusaResearch Einsy RAMBo.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

leptun, 2020-05-09:
Please also post the verbose output as explained in the error. Enable all verbose information.

salmander, 2020-05-09:
Thanks again for looking into this for me 

```
Arduino: 1.8.5 (Mac OS X), Board: "PrusaResearch Einsy RAMBo"

/Applications/Arduino 1.8.5.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 1.8.5.app/Contents/Java/hardware -hardware /Users/salmanahmed/Library/Arduino15/packages -tools /Applications/Arduino 1.8.5.app/Contents/Java/tools-builder -tools /Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr -tools /Users/salmanahmed/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 1.8.5.app/Contents/Java/libraries -libraries /Users/salmanahmed/Documents/Arduino/libraries -fqbn=PrusaResearch:avr:prusa_einsy_rambo -ide-version=10805 -build-path /var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799 -warnings=none -build-cache /var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_cache_46171 -prefs=build.warn_data_percentage=75 -verbose /Users/salmanahmed/workspace/prusa/Prusa_Fresh/Prusa-Firmware/Firmware/Firmware.ino
/Applications/Arduino 1.8.5.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 1.8.5.app/Contents/Java/hardware -hardware /Users/salmanahmed/Library/Arduino15/packages -tools /Applications/Arduino 1.8.5.app/Contents/Java/tools-builder -tools /Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr -tools /Users/salmanahmed/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 1.8.5.app/Contents/Java/libraries -libraries /Users/salmanahmed/Documents/Arduino/libraries -fqbn=PrusaResearch:avr:prusa_einsy_rambo -ide-version=10805 -build-path /var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799 -warnings=none -build-cache /var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_cache_46171 -prefs=build.warn_data_percentage=75 -verbose /Users/salmanahmed/workspace/prusa/Prusa_Fresh/Prusa-Firmware/Firmware/Firmware.ino
Using board 'prusa_einsy_rambo' from platform in folder: /Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3
Using core 'prusa_einsy_rambo' from platform in folder: /Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3
Detecting libraries used...
"/Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_PRUSA_EINSY_RAMBO -DARDUINO_ARCH_AVR   "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/cores/prusa_einsy_rambo" "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/variants/prusa_einsy_rambo" "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/sketch/Firmware.ino.cpp" -o "/dev/null"
"/Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_PRUSA_EINSY_RAMBO -DARDUINO_ARCH_AVR   "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/cores/prusa_einsy_rambo" "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/variants/prusa_einsy_rambo" "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/sketch/AutoDeplete.cpp" -o "/dev/null"
"/Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_PRUSA_EINSY_RAMBO -DARDUINO_ARCH_AVR   "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/cores/prusa_einsy_rambo" "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/variants/prusa_einsy_rambo" "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/sketch/BlinkM.cpp" -o "/dev/null"
"/Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_PRUSA_EINSY_RAMBO -DARDUINO_ARCH_AVR   "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/cores/prusa_einsy_rambo" "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/variants/prusa_einsy_rambo" "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/sketch/Configuration.cpp" -o "/dev/null"
"/Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_PRUSA_EINSY_RAMBO -DARDUINO_ARCH_AVR   "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/cores/prusa_einsy_rambo" "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/variants/prusa_einsy_rambo" "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/sketch/ConfigurationStore.cpp" -o "/dev/null"
[... cut ...] rambo" "-I/Users/salmanahmed/Library/Arduino15/packages/PrusaResearch/hardware/avr/1.0.3/variants/prusa_einsy_rambo" "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/sketch/Firmware.ino.cpp" -o "/var/folders/_z/m7kkmrkj7fb0fxrty0pm4zjc0000gn/T/arduino_build_252799/preproc/ctags_target_for_gcc_minus_e.cpp"
fork/exec /Applications/Arduino 1.8.5.app/Contents/Java/hardware/tools/avr/bin/avr-g++: bad CPU type in executable
Error compiling for board PrusaResearch Einsy RAMBo.
```

leptun, 2020-05-09:
What macOS version are you using?
I think this thread applies to your issue: https://forum.arduino.cc/index.php?topic=620175.0
Arduino 1.8.5 is 32bit only, but your macOS version might only support 64bit applications. I think this is why it fails.
Please also try compiling any of the arduino example sketches (eg. Blink) using the Arduino UNO board (anything but the Prusa board). If you get the same error, then this is your issue.

Support for newer arduino versions is not here yet. Last time I tried it, I got it to compile and work on the printer, but I also experienced unexpected firmware crashes during long prints, something I didn't before updating to gcc 5.4.0 (the one used in arduino 1.8.6 and newer).

salmander, 2020-05-09:
Yeah, I came across this as well. 

So basically until Prusa supports newer Arduino officially I can't do anything.

Thank again for your help

DRracer, 2020-05-11:
@salmander thank you for reporting this, we are planning to upgrade the AVR compiler in the future (and cleaning-up stuff which doesn't compile). However, as @leptun already wrote, avr-gcc 5.4 produces/uncovers new bugs and thus transferring the development to it is not straightforward.

Btw. the expression:
`static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20);`
is "correct" in its true meaning, however the C++ standard doesn't allow reinterpret_cast in constexpr.

cskozlowski, 2020-05-19:
Having the same issue here, using 2020 MacBook Air running Catalina 10.15.4.

dmnkhhn, 2020-05-20:
Same here. 
I highly doubt that fixing this expression is the only thing we need to do or they would have done it already. Let's see what the future brings and hope for the best. :-)

cskozlowski, 2020-05-20:
My 'workaround' on my mac laptop was to load the arduino 1.8.5 IDE (running windows 10 with vmware).  It compiles successfully.

salmander, 2020-05-20:
I am thinking of creating a docker image to compile the firmware.

vertigo235, 2020-05-20:
@salmander that's a great idea, I already did that, I use drone.ci to automate builds as well

DRracer, 2020-09-07:
For everyone using the latest OSX: building FW on OSX Catalina is not supported right now. The problem was primarily caused by Apple dropping support for 32bit applications. For building the FW you'd need a newer Arduino package which also comes with a newer AVR-GCC compiler.

Even though we are planning to upgrade the AVR-GCC compiler in the future (and cleaning-up stuff which doesn't compile),
the newer compiler also produces and/or uncovers new bugs, glitches and thus upgrading the build chain **does** have severe negative impact on FW's stability.

**Please note the current build chain (Arduino 1.8.5) has been well tested and was proven in long term of FW development.**

For now probably the best solution is to use a docker image or a small virtual machine and build the FW inside.

If you are really feeling lucky and want to experiment a lot, there is an **unsupported solution** you may try - @leptun 's branch
with the updated AVR-GCC: https://github.com/leptun/Prusa-Firmware/tree/MK3_GCC
Only PF-build.sh is updated for now.

bikebandit101, 2020-09-08:
Hi DRracer,
Moving from my prior thread, I have installed 1.8.5,  I was running 1.8.13 . I made all the modification per the windows instructions for Ardiuno IDE. Now, choosing AVR ISP , AVRISP mkII, Atmel-ICE (AVR) , I get a compiling error of :
`
fork/exec /Users/valkyrianna/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++: bad CPU type in executable
Error compiling for board Original Prusa i3 MK3 Multi Material 2.0 upgrade.
`
Does this mean I am out of luck updating the firmware until the new update is out that supports that latest AVR-GCC that can be complied on OSX on either Arduino IDE or using the cmake / ninja compiling method.

Sorry for being short yesterday, I spent well over 12 hours trying to figure out how to get this to work. I moved my MMU2S on my MK3S outside my cardbox heat chamber because it was getting too hot and causing in stability with MMU.

On the other side, I have learned more about homebrew and cmake and ninja which I never before knew existed. So there is that.

bikebandit101, 2020-09-08:
@cskozlowski 

How well did VMware system work for you? I am thinking of doing this because I need to get printer up and running. Which VMware did you buy, and did you buy an image of windows?

MikePfaff, 2020-11-30:
Ran into the same problem on MacOS 10.15 and Arduino **1.8.12** (1.8.13 didn't work for me) and fixed it by changing line 594 in the eeprom.h from
`static constexpr M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`
to
`static M500_conf * const EEPROM_M500_base = reinterpret_cast<M500_conf*>(20); //offset for storing settings using M500`

(This is the fix made by the MK3_GCC fork, see also in diff https://github.com/prusa3d/Prusa-Firmware/compare/MK3...leptun:MK3_GCC#diff-e09465d9901d088a6392810e5cfbda16fee36d4adbc8e34936117c78a32cdcb4 )

bikebandit101, 2021-01-31:
@DRracer when you have a moment, do you have an update to when the ninja compiling for OSX Cat & Above is going to be available?

bikebandit101, 2021-04-24:
@MikePfaff , thank you so much for your comment. I am running 11.2 Big Sur and 1.8.13 Ardiuno IDE and I got the code to compile. I cannot believe I didnt understand your comment four months ago. But, there you go.

gudnimg, 2021-08-02:
I believe we can close this issue as it is resolved by: https://github.com/prusa3d/Prusa-Firmware/pull/2910 ?

closed 2023-10-25 by github-actions[bot]