prusa3d/Prusa-Firmware #2659: [BUG] Printer loosing step with 3.9.0 -RC3

luckyguido, 2020-05-09:
**Printer type** - MK3
**Printer firmware version**-  3.9.0 RC3

**Describe the bug**
  Printer loosing step, after updating firmware it failes 2 over 2 print, one time loosing step x axis and the other on y
  
**To Reproduce**
  Print something, on my printer it was 100% of print
  
**Expected behavior**
 Print finish correctly, now im reverting to last RC1 and try print again

**G-code**
[PiCase_sopra_1x_0.2mm_PETG_MK3_1h35m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4603316/PiCase_sopra_1x_0.2mm_PETG_MK3_1h35m.gcode.zip)

**Photos**
[image]
[image]

awenelo, 2020-05-09:
Can you please:
* Run `belt test` and post the numbers of `Support –> Belt Status`
* Check that there is no debris in the pulley
* Power off the printer and move the axis back and forth, checking that the belt doesn't skip over the pulley and that the axis moves with minimal resistance
* Check that there is no damage to the linear rods

Also, have you made any changes to hardware since upgrading the firmware?

luckyguido, 2020-05-09:
- Belt test -> x: 253 y:285
- It all seems to be clear and moving smooth

I reverted to RC1, same gcode it printed correctly, i need more of this part so i will do further testing with same gcode

awenelo, 2020-05-10:
I notice that you have a raspberry pi camera attached to the printer. Are you using octoprint?

luckyguido, 2020-05-11:
yes i have a camera, but i don't record timelapse.
 i made some further testing, i use an faster travel speed  than stock firmware (for non printing movement, 250mm/s vs 180mm/s), going back to stock avoided the problem, maybe something on the acceleration settings is changed?
Since it seems to be related to custon settings i think it should not be considered a bug

leptun, 2020-05-11:
You might want to increase the stepper current in the source code if you use travel speed higher than 200mm/s. With higher current I can safely use 300mm/s speed and 5000mm/s^2 acceleration. Just keep in mind that crash detection will no longer work correctly.

bytesandbolts, 2020-05-11:
I've also been experiencing a lot of layer skipping recently on a MK2.5S since flashing 3.9.0 RC3, that has been otherwise printing without major issues for over 3 years (since starting life as a MK2). Nearly every print in the last week or so has failed in one way or another related to steps jumping/slipping. Even with running new rods, new bearings, belts recently retightened/realigned I was still experiencing the issue, realised at that point it could be F/W related, flashed the latest release last night and woke up to an 8 hour print that had been failing, same g-code now printing fine again. Maybe just a fluke, now have a 7 hour print running today that failed every time, will report the results once done.

leptun, 2020-05-11:
Please try using high power mode on the MK2.5(S) and see if the issue is fixed. I also experienced layer shifts on my MK2 a long time ago. This fixed it.

bytesandbolts, 2020-05-11:
The 7 hour print just finished running 3.8.1 and the difference is night and day to the previous attempt running 3.9.0 RC3, both using same g-code.
[image]

DRracer, 2020-05-11:
@bytesandbolts could you please verify, that your printer is running at high power? There used to be a bug in until FW 3.8.1 - the MK25/S was showing low power in the menu while running at high power. Please switch between low power an high power twice in the menu (thus the menu and the EEPROM will get in sync). And please report if it solved the problem.

@luckyguido custom settings: it might not be a bug, but it is somethign definitely worth looking into @wavexx.

3d-gussner, 2020-05-11:
@bytesandbolts can you please run `D3 Ax0fff C1` from Pronterface or Octoprint before you do what @DRracer asked you to do?

imakershop, 2020-05-11:
I started seeing the same issue after updating to 3.8.1. on long print jobs. 3 tries in a row resulted in a layer slip around Z4.8 to Z5.1 (not consistent for the 3 runs). updated to 3.0.0 -RC3 but seeing the same results.

bytesandbolts, 2020-05-12:
> @bytesandbolts can you please run `D3 Ax0fff C1` from Pronterface or Octoprint before you do what @DRracer asked you to do?

Apologies I had already cycled through the power settings multiple times before I saw your reply.

I have just flashed back to 3.9.0 RC3 running in high power mode, will reprint using the same g-code as before to see if the issue returns.

bytesandbolts, 2020-05-13:
Running back at 3.9.0 RC3 resulted in the following print, the standoffs are messed up again, I stopped the print before any layer shifting had chance to occur as it was getting late and didn't want to wake up to a failed mess. 

[image]

imakershop, 2020-05-15:
Looks like 3.8.1 has the issue. A new print is showing Y axis slip on a higher layer.
[image]
[image]
[image]

n4mb3r0n3, 2020-05-17:
I think it’s a problem with octoprint and also the known bug with the Display Layer height plugin. 

I had similar problems

3.8.1
3.9.0 Alpha 
3.9.0 RC1 / RC2 / RC3 
And final 3.9.0

Every firmware has the layshifting only while printing from octoprint. 
There is no problem printing with pronterface or printing from SD card.

3d-gussner, 2020-05-17:
To narrow down this issue. Could you try to "downgrade" the Octoprint to 1.3.12 and see if you can reproduce this shifting?
I am not saying that it is an Octoprint issue but it might help where to look for.

wavexx, 2020-05-18:
> [image]

Looking at this @bytesandbolts, I don't see a layer shift here looking at the other raised edges on the print. I do see the standoffs are messed up though. Could you share the model as well as the gcode you're using?

wavexx, 2020-05-18:
> Every firmware has the layshifting only while printing from octoprint.
> There is no problem printing with pronterface or printing from SD card.

It would be nice to have the complete serial log of a print where this happened using octoprint. Bonus points if this happens with the minimal number of plugins, or better yet with no plugins being used at all :)

There might be some combination of things that cause an unwanted delay while stepping. Looking at the first models, a shift might be more likely to happen if a short stutter hits at high speed like in a travel, which is why these isolated studs might cause it more often.

bytesandbolts, 2020-05-20:
> > [image]
> 
> Looking at this @bytesandbolts, I don't see a layer shift here looking at the other raised edges on the print. I do see the standoffs are messed up though. Could you share the model as well as the gcode you're using?

Sure thing, it's attached here. 

[Bottom_Shell.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/4655634/Bottom_Shell.gcode.txt)

wavexx, 2020-05-20:
> [Bottom_Shell.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/4655634/Bottom_Shell.gcode.txt)

Don't forget the 3mf project file :)

bytesandbolts, 2020-05-20:
> Don't forget the 3mf project file :)

I don't have a copy of that anymore - apologies, only kept the g-code output.

On a side note I have upgraded to the 3.9.0 release, couldn't see anything obvious in the release notes relating to the issues I was experiencing, so will see how it goes.

wavexx, 2020-05-20:
On Wed, May 20 2020, Dan Smith wrote:
> I don't have a copy of that anymore - apologies, only kept the g-code
> output.

Is it something you downloaded from thingiverse maybe?
The stl model could be helpful.

> On a side note I have upgraded to the 3.9.0 release, couldn't see
> anything obvious in the release notes relating to the issues I was
> experiencing, so will see how it goes.

Time was a bit tight to look into this specific issue for the release
(3.9 includes a ton of other important fixes).

angelozap, 2020-05-20:
Same Here: layer shifting at 50% of the print with latest 3.9.0 firmware. 

The printer (MK2.5S) suddenly starts to do strong jerky movements and after that it doesn’t follow the path anymore. It layer shifted 2 times in a row 
Model downloaded from Thingiverse. 

[image]
[image]

wavexx, 2020-05-20:
That's an important detail! This could link it to #2358

@luckyguido @bytesandbolts did you notice if you saw some jerky movements before the shift happens?

wavexx, 2020-05-20:
> Model downloaded from Thingiverse.

Could you link to the model and which print settings did you use? (saving the project in prusa-slicer would be the easiest for us to reproduce the issue)

luckyguido, 2020-05-21:
@wavexx it always happened long time after print start, and i wasn't watching. In a couple of print I heard some "strange" click before discovering the problem but i can't tell if it was related or just filament unrolling

angelozap, 2020-05-21:
> > Model downloaded from Thingiverse.
> 
> Could you link to the model and which print settings did you use? (saving the project in prusa-slicer would be the easiest for us to reproduce the issue)

[PushStick_layershift.3mf.zip](https://github.com/prusa3d/Prusa-Firmware/files/4661232/PushStick_layershift.3mf.zip)

Today I'll try to print it with the previous 3.8.1

bytesandbolts, 2020-05-22:
> Is it something you downloaded from thingiverse maybe?

Sure thing, here you go: https://www.thingiverse.com/thing:54528

bytesandbolts, 2020-05-22:
> @luckyguido @bytesandbolts did you notice if you saw some jerky movements before the shift happens?

I don't recall noticing any different about my printer in regards to jerky movements or strange noises etc.

angelozap, 2020-05-22:
> > @luckyguido @bytesandbolts did you notice if you saw some jerky movements before the shift happens?
> 
> I don't recall noticing any different about my printer in regards to jerky movements, but after reading luckygudio's comment about a strange click, I do remember a hearing a loud noise click/clunk followed by layers starting to shift (I repeatedly checked belt tension as I assumed it was the belt jumping teeth in the gear).

Yes is the same thing that happened here, but since I was looking at the printer  I've also noticed some strange movements as if it something was obstructing the nozzle and then layer shifting occurred.

n4mb3r0n3, 2020-05-24:
Downgrading octoprint solved the layershift so I think it’s not Firmware related. The layer shift began just after the latest upgrade from octoprint and the display layer height plugin. Deleting the plugin helped but did not solve the shifting issue. Downgrading octoprint to a previous version solved the shifting for me.

wavexx, 2020-05-24:
It is kind of weird that octoprint could result in a layer shift at all though (I would doubt). This could be verified with a complete serial log, which would also help determining a potential issue in the FW.

3d-gussner, 2020-09-21:
This issue got stale, we gonna close it.
[stale]

DRracer, 2020-09-21:
Closing, the issue is not FW related and has been stale for several months
Thank you @n4mb3r0n3 for your final report

closed 2020-09-21 by DRracer