prusa3d/Prusa-Firmware #2661: [BUG] Firmware upgrade to 3.9.0-RC3, MMU2S got confused due to unknown filament sensor type

nikkolade, 2020-05-13:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0-RC3
**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
After upgrading the firmware from 3.8.1 to 3.9.0-RC3 the MMU2S got very much confused at the beginning of any my test prints. The display was informing me that the MMU2S has issues that I have to take care of.

The printer has the old version of the filament sensor (as it is originally an early MK3 from 2017 December) and initially the “Support -> Fil. sensor”  was showing "Unknown Status" for the sensor.

After I manually did the test calibration mentioned in the RC3 release notes, the MMU2S started to co-operate again normally.

The manual calibration instructions are not valid for the MMU2S printer in my opinion, as directly inserting a piece of filament into the extruders filament sensor did not work with the "filament sensor detection" feature. The filament sensor version detection worked fine, once I just pressed slightly the filament sensor tower's lever manually for the duration of the test (Unknown status -> 0.3 or older).

**To Reproduce**
I can try again, if I can set the MK3S filament sensor status back to "Unknown status". Otherwise, it seems to be working now (first test print is currently running).
  
**Expected behavior**
The printer should be working condition after firmware update or it should point to the right direction if user input is required.

3d-gussner, 2020-05-13:
@nikkolade Is it related to https://github.com/prusa3d/Prusa-Firmware/issues/2649?

Can you run the g-code `D3 Ax0d48 C1` and report back the value?
00 = IR sensor 0.3 or older
01 = IR sensor 0.4 or newer
ff = Unknown state (after Factory rest -> All data)

Where do you see the status changing for `0.3 or older` to `unknown state` ?
In the serial output or in LCD Support -> Fil. sensor v.: ?

3d-gussner, 2020-05-13:
We provided a hotfix in https://github.com/prusa3d/Prusa-Firmware/pull/2660 to set it from the "unknown state" to "0.3 or older" during selftest in case people do the Factory rest --> All data.

3d-gussner, 2020-05-13:
> The manual calibration instructions are not valid for the MMU2S printer in my opinion, as directly inserting a piece of filament into the extruders filament sensor did not work with the "filament [...]

We are working on a better way to detect the new/old sensor with MMU2s ... And you are correct that these instructions don't make sense for the MK3s+MMU2s.

May I ask which "manual calibration" you mean?

nikkolade, 2020-05-13:
By "manual calibration" I was just referring to the given instructions in the RC3 release notes about how to manually force the detection of the filament sensor type.
Silly choice of words in this case.

nikkolade, 2020-05-13:
> @nikkolade Is it related to #2649?

No, it's not, unless the final 3.9.0 firmware release documentation specifically requires you to run the self tests again. I just did the update and tried to print afterwards ---> the MMU2S refused to work as the filament sensor type was still of type "Unknown state" in the “Support -> Fil. sensor” menu.

> 
> Can you run the g-code `D3 Ax0d48 C1` and report back the value?
> 00 = IR sensor 0.3 or older
> 01 = IR sensor 0.4 or newer
> ff = Unknown state (after Factory rest -> All data)

I still have the first test print running, but at least the “Support -> Fil. sensor” menu now says that I do have the Filament Sensor type: "0.3 or older" and the printer seems to work.

> 
> Where do you see the status changing for `0.3 or older` to `unknown state` ?
> In the serial output or in LCD Support -> Fil. sensor v.: ?

No, the status changed from the initial "unknown state" to "0.3 or older" in the “Support -> Fil. sensor” menu, once I did the manual detection process.

I just bet that there will be a lot of confused users, unless the automatic detection is able to handle the "0.3 or older" type of sensor automatically OR the release notes specifically mention to run the selftest/Filament Sensor detection procedure even if you have the old type of filament sensor.

The resulting MMU2S error messages do not really point to the missing MK3S filament sensor type value, they are way too generic in this case.

3d-gussner, 2020-05-13:
@nikkolade Thanks for the fast answer. I will have a look that.

> I just bet that there will be a lot of confused users, unless the automatic detection is able to handle the "0.3 or older" type of sensor automatically OR the release notes specifically mention to [...]

We will have to update the release notes to make it "more" clear to the user what to do.
Thanks again for the feedback.

3d-gussner, 2020-05-14:
We have a temporary fix for that https://github.com/prusa3d/Prusa-Firmware/pull/2660
At this moment:
- MK3s users updating from FW3.8.1 to FW3.9.0 need to run `Setting -> HW Setup -> Fsensor Detection` to detect which IR sensor is installed.
- MK3s+MMU2s users need to run the `Selftest` to set the the EEPROM value for the IR sensor.
  Also cleaning the gears from debris is always a good idea.

We are working on an improved version for the IR sensor, this will be in FW3.9.1.

closed 2023-10-25 by github-actions[bot]