prusa3d/Prusa-Firmware #2664: 3.9 RC3 Printing Speeds are Still Extremely Slow(video link)

tmakman, 2020-05-13:
I have tested the previous versions a few weeks ago and after noticing it was too slow, I rolled back to 3.81. Just installed this RC3, and it is still too slow. Even did a new K value test, and changed the K value from 45 to 0.2, but it is still too slow.

On the video below, machines and the files are the same. The one on the left is running 3.9 with LA1.5, and the one on the right is running the current 3.81 version. As you can see, the speed is extremely slow on the left. 

I tried both an existing gcode to let it auto convert, and created another gcode with the newer K value.

[https://youtu.be/_G1pFeUo7Q8](url)

leptun, 2020-05-14:
Are you using octoprint? If yes, try printing from the sd card. If that fixes the problem, please provide the serial log from octoprint.

Also, please provide the gcode file that was seen in the video.

Considering that PLA needs a LA1.5 value of around 0.06, 0.2 sounds a bit excessive even for PETG, but I might be wrong.

wavexx, 2020-05-14:
Speed checks out with with K0.2 and possibly low jerk.
That's too high even for PETG.

Could you share the gcode you're using?

tmakman, 2020-05-14:
Well, first of all, the auto conversion didn't work, that's why I had to manually change the K value. Most of the public would rely on the auto conversion I believe. 

Based on the K value test I did, 0.2 looked to best and that's why I went with it. At another forum, someone recommended 0.05, which I tried and as you said it was faster but still too slow. It printed the object in the video in 9hrs and 15mins, which normally prints in 7hrs and 40 mins.

The object is a proprietary object, so unfortunately I cannot share that particular one, but I quickly modeled a similar base of it and exported it the same way. The first gcode is how I would normally have it running on the other machines, which is at K45(auto conversion makes this so slow), and the second gcode is after I manually change the K to 0.2, which is still slow, but not as slow as K45.

[Test_Normal K at 45.zip](https://github.com/prusa3d/Prusa-Firmware/files/4629330/Test_Normal.K.at.45.zip)

[Test_Adjusted K at 0.2.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4629332/Test_Adjusted.K.at.0.2.gcode.zip)

wavexx, 2020-05-14:
Looking at the gcode, first thing that comes to my eyes is that E-jerk is set to 0.33, which is why the result is slow.

The FW 3.8.1 and previous versions essentially always ignored E-jerk, but now it absolutely _does_ influence the print speed. We do actually bump the E-jerk values automatically as well, but there's a sanity check for values outside a good range.

Is there a reason you have such a value? AFAIK this was never used in any official profile at least.

In PrusaSlicer, you go to "Printer Settings" -> "Machine Limits", change E-jerk to 4.5 which is the new default value (or use the original value, which was 1.5). This will bring the printer up to speed.

You can increase Max E-jerk up to 8 according to taste, depending on how judge the quality of the resulting print.

tmakman, 2020-05-14:
> Looking at the gcode, first thing that comes to my eyes is that E-jerk is set to 0.33, which is why the result is slow.
>
> The FW 3.8.1 and previous versions essentially always ignored E-jerk, but now [...]

For this particular print, I have been using Pretty PETG profile, with some minor tweaks. But the E-jerk value is based on the Pretty PETG profile, I haven't changed that one.

Would changing the E-jerk to 4.5 fix the whole speed problem, if the K value is still the same based on the old numbering system?(K45)

leptun, 2020-05-14:
Yes, as the E-Jerk only limits the speed at which the LA algorithm can act.

tmakman, 2020-05-14:
> 
> 
> Yes, as the E-Jerk only limits the speed at which the LA algorithm can act.

Alright, I was planning on rolling back to 3.81 this morning, but will give it another shot with 4.5mm/s E-jerk

wavexx, 2020-05-14:
I know Chris was the first to update the Pretty bundles to LA1.5, and he's using 4.5 (from memory) as well. Maybe this was an oversight due to E-jerk not having any effect initially.

tmakman, 2020-05-14:
> 
> 
> I know Chris was the first to update the Pretty bundles to LA1.5, and he's using 4.5 (from memory) as well. Maybe this was an oversight due to E-jerk not having any effect initially.

I am using the bottom one, I don't know if that one has been updated to LA1.5

[image]

wavexx, 2020-05-14:
Good question - not sure about this :). I know I've used the second one from the top (BMG PETG V1.5) which was done for the community 1.5 release.

tmakman, 2020-05-16:
Tested this at 4.5 E-jerk, and speeds seemed normal. In fact, I think it printed the 7H40m print a few minutes quicker as well.

wavexx, 2020-05-17:
Checked all pretty profiles, only "Pretty PETG" and "Pretty CFPETG" use unusually low e-jerk values. Given how popular the Pretty profiles are, this is indeed noteworthy.

BTW, with the 3.9 FWs you can use the "Linear Advance 1.5" Pretty profiles directly (https://prusaslicerprofiles.com/linear-advance-1-5-pretty-profiles/) which should yield better results too

DRracer, 2020-09-21:
This has been solved in FW 3.9.0, closing

closed 2020-09-21 by DRracer