prusa3d/Prusa-Firmware #2668: 3.9.0-rc2 and Pi W has Err: Part Fan err issue and hang again.

greggree, 2020-05-17:
3 MK3s
3.9.0-rc2

While printing through Octoprint on the Pi W on the Einsy, prints randomly stop with Err: Print Fan err and I can't resume. After flashing back to 3.8.1 I disconnected my x axis motor, set the fan to 255 (Really should be going from 0 to 100% in the 21st century). I moved the extruder back and forth the full motion and monitored the Support/Extruder/Fan RPM. It didn't jump around. I did 3 more prints through Octoprint and didn't see an issue. Its not my fan cabling. I can print fine with 3.8.1. Interrupts for the RPM pulses are getting blocked, or bugged, causing a false positive.
 
To repo install a Pi Zero W to the Einsy and print multiple things through the web ui. Random print hangs without resume.

jltx1, 2020-06-14:
I flashed back to 3.8.0 (I still had laying around) and the fan speeds are correctly reported again.  Appears to be 3.9.0 bug.

pomidoorq, 2020-08-17:
[image]
Same problem here with 3.9.0. After some time into any print (after random time) there is Err: Print fan error and can't resume. When enabling print fan manually the rpm values are all over the place (check attached photo). The only way to resume printing is to disable "fan check" from menu. Im not using PI, printing from SD. Issue happens only on this FW, 3.8.1 is fine. Einsy Rambo board.

DRracer, 2020-09-22:
Is this issue present in the latest FW 3.9.1? We did some improvements to ISR handling which may have helped.

JohnnyDeer, 2020-11-26:
Closing for no response as stale.

closed 2020-11-26 by JohnnyDeer