prusa3d/Prusa-Firmware #2672: Errors in "pause mode"

juanda-2225, 2020-05-21:
I have a Prusa MK3S printer, and with the las update of PrusaSlicer-2.2.0 the "pause mode" has failing, beacause, when I replay the machine, the printer detects again that that one is the "layer of pause" so the machines get in an eternal loopof pause, play, pause, play, and don't continue the printing, thanks for the peopple who can help me, and forgive my english.

JohnnyDeer, 2020-05-21:
@juanda-2225 as first thing we need to know, which version of firmware is on your printer? Updating to latest version could solve your issue.

JohnnyDeer, 2020-05-21:
@juanda-2225 I'm sorry but whole issue have to be reopened here in FW repository with filled template. I´ll close this one, can you please open a new one?

closed 2020-05-21 by JohnnyDeer