prusa3d/Prusa-Firmware #2674: [BUG]3.9.0 Final, retraction problems.

Glasofruix, 2020-05-21:
**Printer type** -MK3S
**Printer firmware version**-  3.9.0

**Describe the bug**
Very stringy prints with identical gcode used on 3.8.1 and 3.9.0 RC3 to the point of parts fusing together. No problem whatsoever before the upgrade.
  
**To Reproduce**
Just print something
  
**Expected behavior**
Little to no wispy strings
  
Here's a pic:

https://i.imgur.com/h4UY3NZ.jpg

awenelo, 2020-05-21:
Can you please attach your Gcode? I suspect this has something to do with the upgrade to LA1.5, resulting in different K values.

Related to #2543?

DRracer, 2020-05-26:
@Glasofruix looking at the photo you posted - the filament is very wet and the temperature is too high - the skirt is full of tiny bubbles and the whole print is not shiny.

Glasofruix, 2020-05-26:
> 
> 
> @Glasofruix looking at the photo you posted - the filament is very wet and the temperature is too high - the skirt is full of tiny bubbles and the whole print is not shiny.

It's not supposed to be shiny it's a matte filament, the spool is barely a month old and was properly stored with dessicant, the skirt always looked like that. My prints were fine before i upgraded to 3.9.0. I'll try to dry it though and continue testing with different filaments. It's would be funny if it's just a coincidence.

DRracer, 2020-05-26:
@Glasofruix ok, thanks for additional info. Now please post your gcode so that we can try to reproduce your issue. Also the brand and filament type would be helpful

mullcom, 2020-06-03:
Did you solve your problem. for me i get some blobs on my print with petG and  just open it. it seams that some times it push out to much filiment. try to change values. i get better result of higher speed but still blob coming.

DRracer, 2020-09-21:
All this should be fixed in FW 3.9.1 final, can we close this issue @Glasofruix ?

Prusa-Support, 2022-07-07:
Thanks for reporting.

This issue has been awaiting for response until now, moreover, it was supposedly fixed. Therefore, the issue will be closed now.

Michele Moramarco
Prusa Research