prusa3d/Prusa-Firmware #2675: error after prusa3d_fw_MK3S_3_9_0_3421 update

courtpl, 2020-05-21:
After installing the firmware prusa3d_fw_MK3S_3_9_0_3421, it is impossible to select the language "Francais", each time we return to the language selection menu until we choose "English", what to do ????

I have also an error when flash (see end of log):

avrdude-slic3r -v -p atmega2560 -c wiring -P COM7 -b 115200 -D -U flash:w:0:C:\Users\pascal\Downloads\prusa3d_fw_3_9_0_MK3S\prusa3d_fw_MK3S_3_9_0_3421.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

Using Port : COM7
Using Programmer : wiring
Overriding Baud Rate : 115200
AVR Part : ATmega2560
Chip Erase delay : 9000 us
PAGEL : PD7
BS2 : PA0
RESET disposition : dedicated
RETRY pulse : SCK
serial program mode : yes
parallel program mode : yes
Timeout : 200
StabDelay : 100
CmdexeDelay : 25
SyncLoops : 32
ByteDelay : 0
PollIndex : 3
PollValue : 0x53
Memory Detail :

Block Poll Page Polled
Memory Type Mode Delay Size Indx Paged Size Size #Pages MinW MaxW ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom 65 10 8 0 no 4096 8 0 9000 9000 0x00 0x00
flash 65 10 256 0 yes 262144 256 1024 4500 4500 0x00 0x00
lfuse 0 0 0 0 no 1 0 0 9000 9000 0x00 0x00
hfuse 0 0 0 0 no 1 0 0 9000 9000 0x00 0x00
efuse 0 0 0 0 no 1 0 0 9000 9000 0x00 0x00
lock 0 0 0 0 no 1 0 0 9000 9000 0x00 0x00
calibration 0 0 0 0 no 1 0 0 0 0 0x00 0x00
signature 0 0 0 0 no 3 0 0 0 0 0x00 0x00

Programmer Type : Wiring
Description : Wiring
Programmer Model: AVRISP
Hardware Version: 15
Firmware Version Master : 2.10
Vtarget : 0.0 V
SCK period : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\pascal\Downloads\prusa3d_fw_3_9_0_MK3S\prusa3d_fw_MK3S_3_9_0_3421.hex"
avrdude-slic3r: writing flash (248818 bytes):

Writing | ################################################## | 100% 44.91s

avrdude-slic3r: 248818 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\pascal\Downloads\prusa3d_fw_3_9_0_MK3S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: load data flash data from input file C:\Users\pascal\Downloads\prusa3d_fw_3_9_0_MK3S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: input file C:\Users\pascal\Downloads\prusa3d_fw_3_9_0_MK3S\prusa3d_fw_MK3S_3_9_0_3421.hex contains 248818 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 31.95s

avrdude-slic3r: verifying ...
avrdude-slic3r: 248818 bytes of flash verified

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done. Thank you.

avrdude-slic3r -v -p atmega2560 -c arduino -P COM7 -b 115200 -D -u -U flash:w:1:C:\Users\pascal\Downloads\prusa3d_fw_3_9_0_MK3S\prusa3d_fw_MK3S_3_9_0_3421.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

Using Port : COM7
Using Programmer : arduino
Overriding Baud Rate : 115200
avrdude-slic3r: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude-slic3r: arduino_open(): Failed to initialize MK3 external flash programming mode
avrdude-slic3r: Could not open port: COM7

avrdude-slic3r done. Thank you.

3d-gussner, 2020-05-21:
Please make sure that you don't have any other applications open that might want Co connect to com port.
Try another USB cable. Restart Windows and PrusaSlicer.

courtpl, 2020-05-21:
I try with other cable and computer, allways the same result.
The second command do not wait on the prusa finish the reboot.

leptun, 2020-05-22:
Just tried flashing the official MK3S-3.9.0 firmware with PrusaSlicer and I had no issues whatsoever.
If you connect to the printer via USB with Pronterface or Octoprint, what are the startup messages?
Also, do you get any `External SPI flash W25X20CL is not responding. Language switch unavailable.` messages on the LCD?

courtpl, 2020-05-22:
I use the official firmware, and the documented method by Prusa for flashing the printer.
And no special message on LCD.

leptun, 2020-05-22:
Can you please make a video with the LCD of the printer during the flashing process?

courtpl, 2020-05-22:
You can fine the files here:
Video : https://1drv.ms/v/s!AkhjapYZZna1hbN5IIkM0tNpLJ6GNQ
[image]

sousasov, 2020-05-22:
I have the same problem. I can't change lang to czech. Only English works.

leptun, 2020-05-22:
@courtpl in the video I saw that you have the MMU unit. Try disconnecting it and flash again.
@sousasov do you have a MMU unit as well?

sousasov, 2020-05-22:
@leptun I don't have MMU unit. My printer is  MK3S without MMU

courtpl, 2020-05-22:
I tried by disconnecting the MMU, always the same behavior.

3d-gussner, 2020-05-22:
Can you flash the "FW3.8.1" hexfile?

DRracer, 2020-05-22:
We've got a few reports about unfinished flashing last year. I've seen this only once on Windows 10 64bit, but was unable to reproduce the issue on that machine anymore.
Assuming you are using Windows 10, right?
Does FW 3.8.1 flash OK?

courtpl, 2020-05-22:
I just tried with 3.8.1, it's the same problem, the same error for the 2nd commade.
But I can choose the language, because I think this part has not been overwritten by version 3.9.0

courtpl, 2020-05-22:
My pc is powerful enough for the processor to function on average only at 10% of its capacity, and only 38% of memory.
3.25Ghz processor
Memory 32Gb
SSD drives
NVidia GPU

3d-gussner, 2020-05-22:
Can you please try an older PrusaSlicer version? Not sure if something changed with avrdude-slic3r lately.

3d-gussner, 2020-05-22:
Can you remember which version of Prusa-Firmware and PrusaSlicer successfully uploaded a hex file to your printer?

courtpl, 2020-05-22:
In the  i have installed the firmware 3.8.1 with de PrusaSlicer 2.1.0.

I just tried to install the 3.8.1 firmware again, with PrusaSlicer 2.1 from the prusa3d_win_2_2_8 and prusa3d_win_2_2_7 package.
Unable to reinstall firmware without error, always the same problem during the 2nd command.

3d-gussner, 2020-05-22:
I have no issue flashing the firmware on MK3s + MMU2s https://www.youtube.com/watch?v=p9ndWu7HgKI but that doesn't help you.

Do you have the MMU2s updated to latest firmware?
Is it disconnected?

courtpl, 2020-05-22:
I saw your video, I proceed in exactly the same way the MMU is not connected to the USB, but the 2nd update does not take place.

DRracer, 2020-05-22:
hmmm, there must be a temporal hazard - the slicer is either too fast or too slow to fit within the 500ms waiting period.
@courtpl can you try another PC (windows, linux, it doesn't matter too much)?

courtpl, 2020-05-22:
I have already tried with 2 different PCs, still under Windows 10, I have no other system available.
But in any case it is always the same thing ....

courtpl, 2020-05-22:
An odd thing I completely disconnected the MMU, impossible to boot the printer, it still remains on the Prusa screen ....

3d-gussner, 2020-05-22:
If you have custom settings run `M503` gcode and save the data.
Can you please run `D3` gcode and save the EEPROM dump data.
Perform a Factory Reset --> All Data
Try to reflash the firmware.
Can you check the COM port
[image]

As you can see we cannot reproduce your issue and trying different things.

3d-gussner, 2020-05-22:
Download and reinstall newest DRIVERS & APPS 2.3.0 
version https://www.prusa3d.com/drivers/

courtpl, 2020-05-22:
On pronterface D3 -> *** Unknown syntax: D3
How can use it ?

3d-gussner, 2020-05-22:
Putty works. Pronterface doesn't "know" D-codes

courtpl, 2020-05-22:
I have a good new,
I finally flash the printer successfully,
With that and the PrusaSlicer-2.2.0-armhf.AppImage distribution:
[image]

Is it possible to have a problem with the Windows version with the last Windows 10 update ?

courtpl, 2020-05-22:
Amazing isn't it ??

DRracer, 2020-05-22:
@courtpl nice 4x4 computer ;) . There is something fishy with Windows 10, however we are still unsure what's the real cause. It might just be the fact, that opening a serial port takes more? time than before due to security reasons - the slicer/avrdude doesn't catch the moment of resetting the printer for uploading the languages into the SPI flash.

3d-gussner, 2020-05-23:
My Windows 10 is version 1909 OS build 18363.836 and it works. So still wondering what the issue is.
Do you have any applications, virus scanner that may cause the issue?

courtpl, 2020-05-23:
I use Norton for anti-virus.
October last year, i have updated sucessfully with the same computer, the FW 3.8.1

3d-gussner, 2020-05-23:
As you can see we are fishing... These kind of issues are hard to solve.
Glad that it is not an issue with
- your printer
- the firmware

3d-gussner, 2020-05-23:
I have the same Windows version and build as you and I can flash it with any issues. So there must be something else.

courtpl, 2020-05-23:
For now I know that I can flash the firmware with my 4x4 rover :-)
Ok, it's a Raspberry P4 b +
IF you continue to support this kind of OS it’s ok for me.

3d-gussner, 2020-05-23:
If you find some time to investigate the issue with YOUR Windows PC and find something please let us know.

Can you please close this issue?
You can always reopen it and post new results or things you consider as an issue or worth being published.
It always helps others if they search the issues (open and closed) section.

Again thanks for being so fast in response and willing to try out different things.

courtpl, 2020-05-23:
You're welcome
It is with pleasure to help you.

GreenShadowMaker, 2020-05-25:
I experienced and resolved what appears to be matching behavior. Adding my experience.

### First Flash attempt:
Started with MK3s first 

```
avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex"
avrdude-slic3r: writing flash (248818 bytes):

Writing | ################################################## | 100% 44.80s

avrdude-slic3r: 248818 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: load data flash data from input file C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: input file C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex contains 248818 bytes
avrdude-slic3r: reading on-chip flash data:

[... 163 lines cut ...]
                SCK period      : 3.3 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: reading input file "C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex"
avrdude-slic3r: writing flash (64488 bytes):

Writing | ################################################## | 100% 7.67s

avrdude-slic3r: 64488 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: load data flash data from input file C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: input file C:\Users\mpask\Downloads\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex contains 64488 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 7.24s

avrdude-slic3r: verifying ...
avrdude-slic3r: 64488 bytes of flash verified

avrdude-slic3r done.  Thank you.
```

leptun, 2020-05-29:
I'm working on a fix that might influence this issue. Can anyone who still has the issue or can generate it in some way test this firmware and confirm if the upload is complete or it still fails at the second upload stage?

[FW390-Build3421 + optiboot fixes.zip](https://github.com/prusa3d/Prusa-Firmware/files/4702736/FW390-Build3421.%2B.optiboot.fixes.zip)

DRracer, 2020-05-29:
@GreenShadowMaker interesting finding with the mouse - this could be one of the possible causes. Something is fiddling with the serial port while trying to upload the language pack.
The problem is, that the COM port is closed after flashing the FW and the same COM port is being opened shortly after for the language pack.

A11lessandro2, 2020-06-02:
for leptun, I tested your firmware and I confirm that it installs without errors on windows 10, in the end you can choose the language in the right way.

leptun, 2020-06-02:
@A11lessandro2 thank you for testing. Just for the record, did you have issues flashing before my firmware?

A11lessandro2, 2020-06-02:
Before trying your Firmware, I tried to install the official version 3.90 downloaded from the prusa site using windows 10 and also windows 7, but an error always came out on prusaslice and even if the firmware was installed then it was not possible to change the language , I had solved it in part by installing the firmware with linux. see the discussion here : https://github.com/prusa3d/Prusa-Firmware/issues/2702

leptun, 2020-06-03:
@A11lessandro2 If it is not a problem, can you please try flashing again and post the logs? The thing is that now that you had a successful flash with Linux, even if it keeps failing in Windows the firmware will work since the language section is already flashed.

A11lessandro2, 2020-06-03:
I did other firmware update tests today.
I tried to update the firmware with windows 10 and it worked without errors, and I could change the language to Italian. While a few days ago it always gave me an error in the firmware and you could not change the language. Probably the installation of the firmware via linux fixed something, which also allowed the subsequent installation via windows 10.
I attach the logs of the installation of your firmware:
avrdude-slic3r -v -p atmega2560 -c wiring -P COM4 -b 115200 -D -U flash:w:0:C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3S-EINSy10a-E3Dv6full.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM4
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3S-EINSy10a-E3Dv6full.hex"
avrdude-slic3r: writing flash (249036 bytes):

Writing | ################################################## | 100% 43.61s

avrdude-slic3r: 249036 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3S-EINSy10a-E3Dv6full.hex:
avrdude-slic3r: load data flash data from input file C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3S-EINSy10a-E3Dv6full.hex:
avrdude-slic3r: input file C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3S-EINSy10a-E3Dv6full.hex contains 249036 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 31.89s

avrdude-slic3r: verifying ...
[... 197 lines cut ...]
                Oscillator      : 28.800 kHz
                SCK period      : 3.3 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: reading input file "C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3-EINSy10a-E3Dv6full.hex"
avrdude-slic3r: writing flash (64488 bytes):

Writing | ################################################## | 100% 7.69s

avrdude-slic3r: 64488 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3-EINSy10a-E3Dv6full.hex:
avrdude-slic3r: load data flash data from input file C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3-EINSy10a-E3Dv6full.hex:
avrdude-slic3r: input file C:\Users\Dell_1\Desktop\FW390-Build3421.+.optiboot.fixes\FW390-Build3421-1_75mm_MK3-EINSy10a-E3Dv6full.hex contains 64488 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 7.23s

avrdude-slic3r: verifying ...
avrdude-slic3r: 64488 bytes of flash verified

avrdude-slic3r done.  Thank you.

closed 2023-10-25 by github-actions[bot]