prusa3d/Prusa-Firmware #2676: [FEATURE REQUEST] MMU - Eject All

Mabritoj, 2020-05-21:
Couldn't find anything posted on this, so apologies if this already has been asked.

With the MMU/MMU2S, it would be nice to have a "Eject All" button to eject all the filament slots in a sequential pattern. This is available for filament load and wondering if this could be enabled for unload or whats the technical reason its not? 

After completing a MMU print or just ejecting multiple filaments in general, at least from my perspective...its repetitive to go thru the menu, click Eject Filament, choose the slot number, let the filament eject, click the knob so the selector moves back to the now ejected slot, and repeat by starting to click thru the menu again. Not to mention, each time this action is performed, it tries to to eject filament from the extruder and after the first eject, not sure why this is needed? 

Ideally it would be nice to choose "Eject All" and then be able to click the knob button to move to the next slot and eject the filament in a sequential pattern on each knob click until slots 1 thru 5 are gone thru. Also, stop trying to eject from the extruder after the first slot.

vintagepc, 2020-05-27:
Sounds like you're confusing the intent of that function - it's purely to unload from the extruder back to the MMU. 

You don't really need to eject any other filaments, from past experience so long as the idler is not engaged on a particular filament it's not too difficult to pull it out the back of the MMU manually, no eject call required.

Mabritoj, 2020-05-28:
Hmmm not sure I am following then. In the menu, there are two options:
- Unload Filament
- Eject Filament

Unload Filament will heat up the extruder if not at temp and unload filament from the extruder back to the MMU. 

Eject Filament does the Unload Filament sequence and then also moves the selector to the side and dis-engages the idler, and slightly pushes the filament out the top. Since the idler is dis-engaged, the filament is easily pulled out the back. 

At least in my experience with the MMU I have, I simply cannot pull the filament out of the back to completely remove it, or less I perform the "Eject Filament" menu option and choose each slot one by one. After re-reading my initial query, I think I left out this detail. This request/scenario is for removing the filament completely from the printer/MMU. When I complete a MMU print, I usually unload all the rolls and pack them back up.

vintagepc, 2020-05-28:
OK, I apologize, I was thinking of the wrong option. (I'lll admit I've never used it. Maybe it's easier for me to just pull filament out the rear because I'm not using the stock buffer)

JKinch03, 2020-05-31:
This probably wouldn't work the best because, as it ejects each filament, it has to move the selector body out of the way so the filament doesn't run into it. This means it would have to move all the way to the right side of the MMU2 unit, and looking at it, the 5th filament can't be pushed out without colliding with the selector body. 
Also, the idler/rotating selector would have to push all 5 filaments out, meaning it would have to rotate to all 5 positions before you could trim off the tips of the filament. 
If you can find a way to avoid these, this would definitely work!

Mabritoj, 2020-05-31:
The idea I am proposing is a little different. Currently today, if you want to eject more then 1 filament completely from the MMU, the operation is as follows:

1. Go into the menu and select "Eject Filament"
2. Once you pick the filament type and the extruder is at temp, you pick what slot to eject. 
3. The extruder ejects the filament 
4. The selector then moves out of the way and the filament is pushed out. You cut the tip if needed.
5. You can then pull the filament out of the back of the MMU
6. You press the knob and the selector moves back to the slot the filament was removed from
7. You start the process all over again by going to number 1 and back into the menu to find "Eject Filament"

What I am proposing is more of a sequence to remove some repetitive steps (admittedly in laziness and convenience :) )
1. Go into the menu and select "Eject Filament"
2. When the extruder is at temp, instead of choosing a individual slot, a new option for "Eject All" is listed as well (just like "Load All" for loading filament)
3. The extruder ejects the filament starting at slot 1
4. The selector then moves out of the way and the filament is pushed out of slot 1
5. You can then pull the filament out of the back of the MMU
6. You press the knob and the selector moves to be out of the way for slot 2. The extruder is NOT engaged as it already was for slot 1...no need to keep trying to eject filament from it. Now its just merely moving the selector in a sequence for each additional slot when clicking the knob button. 
7. Continue this until all 5 slots are completed

At least this is how I envision it. It would save some repetitive tasks, less menu clicks, etc.

JKinch03, 2020-05-31:
Ah, okay! I see now. That makes sense

ulab, 2022-12-05:
I am not sure you guys are understanding what "eject" is for.

- `Unload` will retract the filament from the extruder back into the MMU. If you want to remove it then, you can just pull it out.
- `Eject` feeds the filament forward, as though you're loading it to the extruder, but it doesn't move the selector to that position. 

If a spool ran out of filament, you can't just `unload` and pull the filament out of the back to remove it, because the filament is too short. It will just be in the tube in the back of the MMU, but you can't grab it.

In that case you `eject` the filament out of the front of the MMU.

Prusa-Support, 2022-12-28:
Exactly. Thanks for explaining.
 
The feature is mainly (/only) designed to work together with the SpoolJoin feature: the remaining part of the depleted spools could be stuck inside the MMU2S module (and tubing), so you can "eject" the filament in the position of the depleted specific spool, and pull the remaining filament to load new filament.
This is described in chapter _6.4 Ejecting filament_ of the [MMU2S Handbook](https://www.prusa3d.com/downloads/manual/prusa3d_manual_mmu2s_en.pdf).

The application of "eject filament" to cut the end of the filament is surely interesting but such a feature is not indeed for that use. Indeed, ejecting non-depleted spools may not be necessary, as an irregular filament tip could be cut by the blade behind the selector. The best practice though is to use quality filaments and reliable settings to possibly prevent an irregular filament tip.

I hope you don't mind me closing this issue as a non-applicable request.

Michele Moramarco
Prusa Research

closed 2022-12-28 by Prusa-Support