prusa3d/Prusa-Firmware #2677: [QUESTION] Belt status after upgrade FW3.9.0

Forfelet, 2020-05-22:
Hello,

I had the FW3.8.1. Before upgrade to FW3.9.0, I did a Self-test.
Results from Belt-status : X: 249 and Y: 256

After upgrade to FW3.9.0, I did a Belt test with new belt test feature.
Checking X axis:
X: 249 -> 186
Checking Y axis:
Y: 256 -> 228
Values are very different !

According to Prusa documentation ([https://help.prusa3d.com/en/article/adjusting-belt-tension-mk3-s_112380]), the number should be between 240 and 300 (275~).

But, what values are correct? Values from FW3.8.1 or FW3.9.0?

Have a nice day.
Cyril

cskozlowski, 2020-05-25:
My readings are 'interesting' as well.  I have 3 PRUSA MK3 printers:

X/Y:  228/243  (Purple Mk3s)
X/Y:  235/245  (Red Mk3 Bear)
X/Y:  174/260   (Orange Mk3s)

When I do the crude deflection test, the x-axis all 'feel' about the same but the results for my Orange Mk3s are considerably different than my other two printers.  None of my belts for any of these printers 'feels' particularly tight.

Barnabasek, 2020-05-26:
I confirm the same issue on my MK3S.
The label for this issue should be changed from **Question** to **Bug**, IMHO. 

On FW 3.8.1 X/Y was 285/286. After upgrade to 3.9.0 and Belt test (or Self test) it shows X/Y 235/277 (+/- 3 after multiple tests).
After downgrading FW again to 3.8.1 X/Y is 279/288
After another upgrade to 3.9.0 X/Y is 235/271

vintagepc, 2020-05-26:
FWIW I think this happened because the belt/self-test was updated to use TMC2130 homing mode for better sensitivity. 
This alters some of the driver's configuration (tmc2130_home_enter()) which changes the value that comes back from the stallguard register.

adam-wood, 2020-05-29:
Confirming that something is weird with the belt tension test. I was ~ 270/260 (X/Y) prior to upgrade and am now on 175/260 (XY) on FW3.9
When completely loosening the X belt by unscrewing the stepper motor, it is still reporting X 191

goncalossilva, 2020-06-18:
I have the exact same discrepancy with the new version. The X axis belt is practically loose and reporting 250. When it has a similar tension to the Y axis belt, it drops to around 170~190.

3d-gussner, 2020-06-19:
Please checkout https://github.com/prusa3d/Prusa-Firmware/issues/2677 and report there back your results. Thanks in advance for your help.

mamurluk, 2020-10-31:
I have a self-assembled mk3s bear printer. when self-testing gives an error "selftest error please check axis length x" how to solve this problem?

closed 2023-10-26 by github-actions[bot]