prusa3d/Prusa-Firmware #2681: [QUESTION]<Enter comprehensive title>

landrunner22002, 2020-05-23:
Hi, In Firmware 3.9.0 : I didn't find a G-code / T-code command to launch Cutting Sequence of MMU2S

Thanks

leptun, 2020-05-23:
Cutting is only used when loading filament. If it fails 2 times in a row, the third loading attempt also cuts the filament tip automatically.

landrunner22002, 2020-05-23:
> Cutting is only used when loading filament. If it fails 2 times in a row, the third loading attempt also cuts the filament tip automatically.

Yes I know and I request if G-Code exist to launch systematically a cutting filament at the end of printing (for nylon and flex filaments which are subject of stringing effect after unload)

leptun, 2020-05-23:
There isn’t a gcode command for that.

landrunner22002, 2020-05-23:
So to be implemented by the team ;)

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

See https://reprap.org/wiki/G-code#M706:_Cut_filament

closed 2023-07-31 by gudnimg