prusa3d/Prusa-Firmware #2684: [FEATURE REQUEST]<Show and manipulate Z-Offset instead of Print Speed for the first layer(s)>

mike-wi, 2020-05-24:
MK3S Info-Screen

Show Z-Offset instead of Print Speed in the Info-Screen while printing the first layer(s) and the Z-Offset is adaptable. 

Turning the LCD-Knob would manipulate the Z-value directly without the need to change menu. 

After finishing these adaptable layers Info-Screen returns to show Print Speed again.

awenelo, 2020-05-24:
During a print, if in an adaptable layer, a long press on the knob will open the Z-offset menu directly (if I remember correctly). I believe that it would be better to leave the change print speed function as is, the one extra button press is not that large of an inconvenience.

mike-wi, 2020-05-24:
During the first 2 layers the information printing on 100% print speed is negligible - on the other hand the 1st layer adaption is critical. Ergonomics - you know.

leptun, 2020-05-24:
I agree with @awenelo. Doing such a change will impact the usual workflow for many people, especially if you try to speed up the print from the start.

closed 2023-10-25 by github-actions[bot]