prusa3d/Prusa-Firmware #2693: decrease in print quality with FW 3.9.0

rcaslis, 2020-05-27:
Printer type: MK3
Printer firmware version: 3.9.0 vs 3.8.1
SD card
Upgrades: BMG extruder variant, Vesconite bearings, cable chains

I've seen a significant decrease in quality after moving to 3.9.0.

Attached is a picture of the same piece, printed on the same printer, with the same filament, with the same gcode about 12 hours apart.  The top piece is with 3.9.0 while the bottom piece is 3.8.1.  I did stop the print for the 3.9.0 version before the top layers since the problem was already evident.

It must be the conversion to LA15 that causes the problem, but I'm uncertain of how to proceed past that.  For reference, the gcode uses a K factor of 28.

I've also attached the gcode.

What can be done to improve the quality back to 3.8.1 behavior?

[image]
[crusher.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/4691489/crusher.gcode.txt)

awenelo, 2020-05-27:
Linear advance can be calibrated following instructions [here](https://marlinfw.org/docs/features/lin_advance.html). Can you calibrate it then try again? You can set the K value in pursaslicer in `Filament settings -> Custom Gcode`(you need to be in expert mode)

rcaslis, 2020-05-27:
> Edit: Prusaslicer has updated the filament profiles, can you reslice the model using a `System Presets` filament? (no need to calibrate anymore) Also, what filament are you using?

When my printer isn't busy anymore I'll try reslicing with the system presets filament.

It's a shame I have to though... I have a library of gcode for my current project and I don't look forward to reslicing and re-printing them to test.  Many are 3-10 hour prints that I've printed a half-dozen times tweaking the slicer settings each time to get it just right.  Suddenly having a known quantity become unknown again is frustrating.

I'm using PLA from https://coexllc.com/.  Decent filament, but it prints hot (~220c).

> Linear advance can be calibrated following instructions [here](https://marlinfw.org/docs/features/lin_advance.html). Can you calibrate it then try again? You can set the K value in pursaslicer in `Filament settings -> Custom Gcode`(you need to be in expert mode)

I looked into that last night.  I generated the k-factor calibration pattern using the defaults and honestly all of them looked poor, but a K factor of 0 looked best.  Using that, my print was complete trash.

awenelo, 2020-05-27:
A K factor of 0 disables LA. Prusaslicer sets the K factor to .06, try something around that.

If you have Visual Studio Code, you can replace the K value (the only really important change) by:
`Ctrl + K` `Ctrl + O` and selecting the directory
`Ctrl + Shift + F`
Clicking the arrow to the left of "search"
In "search" typing `M900 K28`
In "replace" typing `M900 K0.06`
`Ctrl + Alt + Enter`

rcaslis, 2020-05-27:
I can update the gcode easy enough (even using notepad works), but isn't 3.9.0 supposed to automatically convert old LA10 values to something sensible for LA15?

awenelo, 2020-05-27:
I believe that to convert, it needs to know what firmware version the Gcode is for. For automatic conversion, I believe that there needs to be `M115 U3.8.1;` in the Gcode.

rcaslis, 2020-05-27:
> I believe that to convert, it needs to know what firmware version the Gcode is for. For automatic conversion, I believe that there needs to be `M115 U3.8.1;` in the Gcode.

That's super interesting.  I'll try that out once my printer is done with it's current print.

mw2matze, 2020-05-28:
For me, after upgrading the Firmware the Quality decreased dramaticly...
As you can see in the picture. (top is 3.8.1, bot is 3.9.0 i printed the same file, with the new config in prusaslicer)
[image]

awenelo, 2020-05-28:
What profiles were those printed with and when were they sliced?

mw2matze, 2020-05-28:
I sliced them yesterday with these Settings : 
# generated by PrusaSlicer 2.2.0+win64 on 2020-05-28 at 18:25:22 UTC
avoid_crossing_perimeters = 0
bed_custom_model = 
bed_custom_texture = 
bed_shape = 0x0,250x0,250x210,0x210
bed_temperature = 90
before_layer_gcode = ;BEFORE_LAYER_CHANGE\nG92 E0.0\n;[layer_z]\n\n
between_objects_gcode = 
bottom_fill_pattern = rectilinear
bottom_solid_layers = 4
bottom_solid_min_thickness = 0.5
bridge_acceleration = 1200
bridge_angle = 0
bridge_fan_speed = 50
bridge_flow_ratio = 0.9
bridge_speed = 60
brim_width = 0
clip_multipart_objects = 1
colorprint_heights = 
compatible_printers_condition_cummulative = "printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.4";"nozzle_diameter[0]!=0.6 and printer_model!=\"MK2SMM\" and printer_model!=\"MINI\" and ! (printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/ and single_extruder_multi_material)"
complete_objects = 0
cooling = 1
cooling_tube_length = 20
cooling_tube_retraction = 40
default_acceleration = 2400
default_filament_profile = "Prusament PLA"
default_print_profile = 0.15mm QUALITY @MK3
deretract_speed = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
draft_shield = 0
duplicate_distance = 6
elefant_foot_compensation = 0.2
end_filament_gcode = "; Filament-specific end gcode"
end_gcode = G1 X0 Y210 F7200\nG1 E2 F5000\nG1 E2 F5500\nG1 E2 F6000\nG1 E-15.0000 F5800\nG1 E-20.0000 F5500\nG1 E10.0000 F3000\nG1 E-10.0000 F3100\nG1 E10.0000 F3150\nG1 E-10.0000 F3250\nG1 E10.0000 F3300\n\nM702 C\n\nG4 ; wait\nM221 S100 ; reset flow\nM900 K0 ; reset LA\nM104 S0 ; turn off temperature\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\n; Lift print head a bit\n{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up\nG1 X0 Y200 F3000 ; home X axis\nM84 ; disable motors
ensure_vertical_shell_thickness = 1
external_perimeter_extrusion_width = 0.55
external_perimeter_speed = 120
external_perimeters_first = 0
extra_loading_move = -25
extra_perimeters = 0
extruder_clearance_height = 20
extruder_clearance_radius = 45
extruder_colour = ""
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 0.99
extrusion_width = 0.55
fan_always_on = 1
fan_below_layer_time = 20
filament_colour = #FF8000
filament_cooling_final_speed = 3.4
filament_cooling_initial_speed = 2.2
filament_cooling_moves = 4
filament_cost = 28.69
filament_density = 1.23
filament_deretract_speed = nil
filament_diameter = 1.75
filament_load_time = 0
filament_loading_speed = 28
filament_loading_speed_start = 3
filament_max_volumetric_speed = 25
filament_minimal_purge_on_wipe_tower = 15
filament_notes = ""
filament_ramming_parameters = "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
filament_retract_before_travel = nil
filament_retract_before_wipe = nil
filament_retract_layer_change = nil
filament_retract_length = nil
[... 148 lines cut ...]
toolchange_gcode = 
top_fill_pattern = rectilinear
top_infill_extrusion_width = 0.5
top_solid_infill_speed = 100
top_solid_layers = 5
top_solid_min_thickness = 0.6
travel_speed = 250
use_firmware_retraction = 0
use_relative_e_distances = 1
use_volumetric_e = 0
variable_layer_height = 1
wipe = 1
wipe_into_infill = 0
wipe_into_objects = 0
wipe_tower = 1
wipe_tower_bridging = 10
wipe_tower_no_sparse_layers = 0
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 170
wipe_tower_y = 125
wiping_volumes_extruders = 70,70
wiping_volumes_matrix = 0
xy_size_compensation = 0
z_offset = 0

mw2matze, 2020-05-29:
> 
> 
> What profiles were those printed with and when were they sliced?

:/ i don't know, what to do.

awenelo, 2020-05-29:
Looking at the start G-codes, it appears as if you are giving the printer a LA1.5 K value, but telling it that the G-code was sliced for 3.8.1. I believe that this causes the printer to believe that it was sliced for LA1.0, so it converts the values (that it thinks are LA1.0) to LA1.5, even though they are already LA1.5, causing different K values than ideal.

In the printer start G-code, can you change the line `M115 U3.8.1 ; tell printer latest fw version` to `M115 U3.9.0 ; tell printer latest fw version` and re-slice your file?

leptun, 2020-05-29:
The M115 command doesn't influence Linear Advance

awenelo, 2020-05-29:
How does it decide to auto-convert?

leptun, 2020-05-29:
If you provide two M900 commands, the new firmware accepts the first one and older firmware accepts the last one. This is what you might encounter in a start gcode now:
```
M900 K0.06 ; will be accepted by FW >=3.9.0
M900 K30 ; will be accepted by FW <3.9.0 or by >=3.9.0 if the LA1.5 one is missing
```

mw2matze, 2020-05-29:
Ok, so we still don't know, why this happens?
Here is my cutdown Gcode i used for both prints:

mw2matze, 2020-05-29:
; generated by PrusaSlicer 2.2.0+win64 on 2020-05-27 at 08:06:16 UTC

; 

; external perimeters extrusion width = 0.55mm
; perimeters extrusion width = 0.55mm
; infill extrusion width = 0.55mm
; solid infill extrusion width = 0.55mm
; top infill extrusion width = 0.50mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.42mm

M73 P0 R263
M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X5.00 Y5.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M862.3 P "MK3SMMU2S" ; printer model check
M862.1 P0.4 ; nozzle diameter check
M115 U3.8.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S265 ; set extruder temp
M140 S85 ; set bed temp
Tx
M190 S85 ; wait for bed temp
M109 S265 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
Tc
; purge line
G1 X55.0 F2000.0
G1 Z0.3 F1000.0
G92 E0.0
G1 X240.0 E25.0 F2200.0
M73 P0 R262
G1 Y-2.0 F1000.0
G1 X55.0 E25 F1400.0
G1 Z0.20 F1000.0
G1 X5.0 E4.0 F1000.0

M221 S95
G92 E0.0
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K0.12 ; Filament gcode LA 1.5
M900 K45 ; Filament gcode LA 1.0
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-2.00000 F3300.00000
G1 Z0.600 F15000.000
;AFTER_LAYER_CHANGE
;0.2
G1 X246.524 Y148.968
G1 Z0.200
G1 E2.00000 F3300.00000
M204 S1200
G1 F3600.000
G1 Z20.600 F15000.000
G1 X61.907 Y112.954
[... 248 lines cut ...]
; toolchange_gcode = 
; top_fill_pattern = rectilinear
; top_infill_extrusion_width = 0.5
; top_solid_infill_speed = 100
; top_solid_layers = 5
; top_solid_min_thickness = 0.6
; travel_speed = 250
; use_firmware_retraction = 0
; use_relative_e_distances = 1
; use_volumetric_e = 0
; variable_layer_height = 1
; wipe = 1
; wipe_into_infill = 0
; wipe_into_objects = 0
; wipe_tower = 1
; wipe_tower_bridging = 10
; wipe_tower_no_sparse_layers = 0
; wipe_tower_rotation_angle = 0
; wipe_tower_width = 60
; wipe_tower_x = 170
; wipe_tower_y = 125
; wiping_volumes_extruders = 70,70
; wiping_volumes_matrix = 0
; xy_size_compensation = 0
; z_offset = 0

rtyr, 2020-05-30:
@mw2matze Can you please attach your full gcode? (you might need it to archive it to zip first or just change file extension)

DRracer, 2020-06-01:
@rcaslis thanks for your report, the new FW shall definitely convert existing G-codes seamlessly (i.e. no need to reslice everything). Looking at your photo this looks like something else than LA kicking in. You can disable LA completely just by setting M900 K0. The corners might not be sharp, but you can see if it makes any difference on the straight perimeter lines.

FalconMuc, 2020-06-01:
Just wanted to state, that I have the same/similar issue.
I updated my I3 MK3S to 3.9.0., printed one of my standards objects that I already printed many times before, with the exact same filament and the exact same gcode and the quality decrease immediately!!!
I then DOWNGRADED to 3.8.1 again because I had to print in a good quality.
After downgrading, I printed the object again and the quality increased immedately, BUT I can't get rid of the feeling that the print quality (surface finish,...) is not the same as before. I compared several of the prints and the quality is slightly decreased. I presented the different prints to several TEST persons and they all declared the 3.9.0 as the worst, followed by the downgraded 3.8.1 print and the best was the original 3.8.1. print. All 5 TEST persons came to the same result.
Is it possible that the firmware 3.9.0 kind of changes some parameters in the printer and that the DOWNGRADING didn't change the parameters back?
I was looking forward to 3.9.0. and at the moment I'm really disappointed...

wavexx, 2020-06-01:
@FalconMuc pics and model would greatly help debugging the issue.

There's no setting which is explicitly modified in 3.9.0. Switching back to 3.8.1 should restore identical results to before.

ptashek, 2020-06-04:
I also have this issue with my MK3 MMU2S after moving from 3.8.x (last version before 3.9) to the initial 3.9 release. The quality drop is quite terrifying to be frank. Filament used was a dry stored roll of silver Prusament PLA (roll ID: 1b675bc0).

All relevant files are in this Dropbox folder: 
https://www.dropbox.com/sh/5067p5agudhh7o3/AADQVEJjEJE3ZrozDfsZDPvma?dl=0

* image of the part perfectly showing the issue(s) (also below)
* STL of the model
* gCode as generated by PrusaSlicer 2.2.0 (latest available for download)
* 3MF project file saved from same PrusaSlicer 

I've never seen anything like this post-update :(

[image]

mullcom, 2020-06-04:
I have a tip for all that have issues. I did have Other issue but when I
did reset of printer power it of for a while and recalibrating it my issue
was away. So if you not test this it can be a good way to do.

send from mobile

rtyr, 2020-06-04:
@ptashek I tried your gcode and it printed fine on my MK3SMMU2S with 3.9.0. Same filament.

[image]

Kachidoki2807, 2020-06-05:
Hi,
I had the same issue here. I printed a bunch of ears savers on both of my MK3s and MK3s MMU2s before the upgrade to 3.9.0. After the upgrade, the same G-Code decreased in quality. So I resliced the project, keeping my print settings but resetting the filament's and the printer's settings to take the new LA parameters. I had the exact same issue.

So I made a test by printing the same G-Code once with the normal K factor and once with K0. Below are the results on both printers. Please be so kind to not blame me or my printers with the ugly print quality, I push the machines to the extrusion limits (probably even a bit over) to get the highest yield as I print 200 of these ears savers a day.

The defect is not as visible as when I print a full plate, but still visible. It can miss up to 2 or 3mm of plastic and thus the part is not more robust

Global view:
[image]

Near view with LA:
[image]

Near view without LA:
[image]

Another sample with more pronounced defect:
[image]

Generally speaking, except this defect, the quality is better with the LA enabled, especially at high speed. But the perimeter loops are not properly closed in this situation with the LA enabled. No issue when disabling it. The under/over-extrusion on straight lines is due to the extreme settings, do not take care of it.

rtyr, 2020-06-05:
@Kachidoki2807 Have tried to calibrate K value to reflect speeds/temperature used for these prints?

Kachidoki2807, 2020-06-05:
Nope, I didn't looked for a better K factor since I never had issues with the default value. The point is that there is a visible change between those two versions.
I could maybe try to find a better K factor just to see if it helps.

Will try that.

wavexx, 2020-06-05:
On Fri, Jun 05 2020, Kachidoki wrote:
> I could maybe try to find a better K factor just to see if it helps.
>
> Will try that.

It would be great, thanks. Doing so will help us see if there's a
mis-calculation of the new conversion factor.

Kachidoki2807, 2020-06-05:
Ok so I am not sure of the procedure to apply, but here is what I did. First test at "normal" speed, and second test at "insane" speed. I set the other values (temp, accel, extrusion multilier and so on...) to the same I used for the print.

First:
[image]

Second:
[image]

As far as I can read the result, by looking either the acceleration or the deceleration, the best value in both cases is between 0.04 and 0.06 with a nominal at 0.05.

I have no idea on how to trap this issue, I will think about it this night.

ptashek, 2020-06-05:
@rtyr That print looks like what I have come to expect from my MK3S over the years I had it. However, I can't get that print to any reasonable quality on 3.9. It fails in a myriad of random but similar ways to the pic I have posted - missed perimeters, extreme underextrusion etc.

Meanwhile, I have downgraded to 3.8.1, cycled power and printed *the same* gCode - no issues whatsoever.

From where I stand, this is clearly a firmware problem as that's the only variable in my case.

enzyme00, 2020-06-06:
> Generally speaking, except this defect, the quality is better with the LA enabled, especially at high speed. But the perimeter loops are not properly closed in this situation with the LA enabled. No [...]

I'm having the same issue with 3.9.0 using a .4mm volcano.  Perimeters are not connecting correctly.  Sigh.

Kachidoki2807, 2020-06-06:
@ptashek here is the result of your G-Code printed on my MMU2S:

[image]
[image]

Maybe you should look elsewhere, it looks like an adhesion issue to me.

DanielSan3318, 2020-06-06:
> @ptashek hier ist das Ergebnis Ihres G-Codes, der auf meiner MMU2S gedruckt ist:
> 
> [image]
> [image]
> 
> Vielleicht solltest du woanders suchen, es sieht für mich nach einem Haftungsproblem aus.

Sorry, I can not agree.
The fact is, for whatever reason, some people have problems with FW 3.9, I am one of them, others who have the same printer, even from the 1st generation, it works. I don't know why that is, but I hope that the problem will be recognized in the near future, but with FW 3.8 the MK3S again delivers clean results.

DanielSan3318, 2020-06-06:
OK, can be a liability problem, but the FW 3.9 is faulty for many and have tested different FW 3.9.

DanielSan3318, 2020-06-06:
[image]
Printed here with Hilbert curve (FW 3.8.1)
[image]
Printed here with Hilbert curve (FW 3.9)
Exactly the same G code, tested with both LA 1.0 and LA 1.5.

DanielSan3318, 2020-06-06:
[image]
Printed here with Hilbert curve (FW 3.9)
Exactly the same G code, tested with both LA 1.0 and LA 1.5.

mullcom, 2020-06-06:
I try to find a way to test the difference between LA1.0 and 1.5

I looked on website and forums but no explaining how to change it in the
software. Please send me a picture of the setting.

send from mobile

DanielSan3318, 2020-06-06:
It's very simple, just change the K value in the G code.
M900 K45 (PETG) LA 1.0
or
M900 K0.12 (PETG) LA1.5

PLA
M900 K30 (PETG) LA 1.0
or
M900 K0.06 (PETG) LA1.5

ABS / ASA
M900 K20 (PETG) LA 1.0
or
M900 K0.04 (PETG) LA1.5

DanielSan3318, 2020-06-06:
With the standard setting in the slicer, both are output, just solve an M900 K ?? you don't need that

Here is an example with PETG
M73 P0 R22
M73 Q0 S22
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.40 E4.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M862.3 P "MK3S" ; printer model check
M862.1 P0.4 ; nozzle diameter check
M115 U3.9.0 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M140 S80 ; set bed temp
M104 S170 ; set extruder temp
M190 S80 ; wait for bed temp
G28 W ; home all without mesh bed level
M104 S240 ; set extruder temp
G80 ; mesh bed leveling
M109 S240 ; wait for extruder temp
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
M73 Q0 S22
M73 P0 R22
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0
M221 S95
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K0.12  ; Filament gcode LA 1.5
M221 S95 ; Flow
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

mullcom, 2020-06-06:
Ahaa. Thx. Now I can make some test and know which I print with.

send from mobile

Kachidoki2807, 2020-06-06:
> Sorry, I can not agree.
> The fact is, for whatever reason, some people have problems with FW 3.9, I am one of them, others who have the same printer, even from the 1st generation, it works. I don't [...]

Indeed, I was misleaded on the purpose of the topic, I was too focus on the LA which should have a massive effect on acceleration / deceleration but not on straight lines. My apologies.

Maybe here it is a more global extrusion issue and maybe it is not related to the LA at all. In all cases, there is a quality regression, exactly as the topic is qualified.

When I look all the pictures here, it seems to be two separate behaviors. The first (@rcaslis and @ptashek and maybe @DanielSan3318 but not sure) is like over-extrusion or rather a "dotted" extrusion. The second (@mw2matze and mine) is like under-extrusion related to the LA.
That is only an assumption based on an analysis of the pictures.

DanielSan3318, 2020-06-06:
My parts are printed with PETG, so light threads, Hilbert curve is nicer with PLA. :-)

But exactly when there was an extrusion.
Perhaps to mention that I have a Bondtech extruder (E852 steps), but as I said, I have tested various FW 3.9 and also checked the values ​​with M503, just like with FW 3.8.1

So I ask the people who have a problem whether they use the original extruder (E280.00 steps with original) or another one?

ptashek, 2020-06-06:
@Kachidoki2807 I agree that this is a wider issue than just LA, as in my case the same gCode printed with LA1.0 and LA1.5 fails to print correctly on FW3.9.

@DanielSan3318 My MK3S MMU2S is stock, however it started life as an MK3 (Sept 2018) and was later (Feb 2019) upgraded using the official parts kit.

Maybe there's a clue to the issue in specific hardware revisions of the control board?

Kachidoki2807, 2020-06-07:
I made some more tests. Firstly, I prepared a very quick test (1min print) able to reproduce my defect.
I printed several combinaisons of the LA parameter.

Freshly sliced with my insane settings (assume it must use LA1.5):
`M900 K0.06 ; Filament gcode LA 1.5`
`M900 K30 ; Filament gcode LA 1.0`
[image]

Very same G-Code but with M900 K0 instead of both:
`M900 K0`
[image]

Very same G-Code but with only LA1.0:
`M900 K30 ; Filament gcode LA 1.0`
[image]

Freshly sliced with stock 0.20mm QUALITY settings (excepted thin wall):
[image]

Next step for me is to find out exactly which setting has such an impact, before to downgrade the firmware and repeat these tests.

DRracer, 2020-06-08:
@Kachidoki2807 @DanielSan3318 thank you for detailed reports, from your photos it looks like we have generally two issues (FYI @wavexx ):
- with LA enabled the start of the path is missing filament (sometimes less, sometimes more) - best seen on @Kachidoki2807 's star
- short infill lines (roughly up to 5mm of length) are overextruded in some cases - probably due to some miscalculation in slowdown/acceleration. Clearly visible on @DanielSan3318 's cube.

@Kachidoki2807 could you please post your G-code for the star?

Also, please note there is only one LA algorithm in the FW - the LA1.5. If you input old LA1.0 K-factor into the G-code, they will be converted into LA1.5 K-factor, but the algorithm stays the same.

Kachidoki2807, 2020-06-08:
> please note there is only one LA algorithm in the FW - the LA1.5. If you input old LA1.0 K-factor into the G-code, they will be converted into LA1.5 K-factor, but the algorithm stays the same.

Indeed, but we're never sure of the side effects on rounding/approximation while using floats values, especially when using so small values and equality tests. At least, the test proved that there is no apparent issue here.

> @Kachidoki2807 could you please post your G-code for the star?

Sure, here you go.
[Body1_0.25mm_PLA_MK3S_1m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4744384/Body1_0.25mm_PLA_MK3S_1m.gcode.zip)

I had enough time to do only one more test, where I slowed down the accelerations to the default values. It was better but still present, about half less missing filament.

3d-gussner, 2020-06-08:
@DanielSan3318 Which firmware do you have?
> My parts are printed with PETG, so light threads, Hilbert curve is nicer with PLA. :-)
> 
> Perhaps to mention that I have a Bondtech extruder (E852 steps), but as I said, I have tested various FW 3.9 and also checked the values ​​with M503, just like with FW 3.8.1
> 
> So I ask the people who have a problem whether they use the original extruder (E280.00 steps with original) or another one?

With E832 you are running with 32 microsteps.

This isn't the supported configuration from Bondetech, see:
https://github.com/BondtechAB/Bondtech-Prusa-Firmware/blob/Bondtech-MK3-FW390/Firmware/variants/Bondtech-MK3S-16-EINSy10a-E3Dv6full.h#L49
https://github.com/BondtechAB/Bondtech-Prusa-Firmware/blob/Bondtech-MK3-FW390/Firmware/variants/Bondtech-MK3S-16-EINSy10a-E3Dv6full.h#L216

There are know issues with E-ustep 32 and high speeds.

I know that these values work for you with FW3.8.1 but we might hit similar know issues with FW3.9.0 and Bondtech extruders.

DanielSan3318, 2020-06-08:
Have tried and tested different FW, from original to Bondtech, but actually need the original FW, because I have no problems in height (Z), because my Z_top is adjusted accordingly.
Now I have original FW 3.8.1 on it again.

3d-gussner, 2020-06-08:
@DanielSan3318 Okay. Please keep in mind that even with FW3.8.1 and Bondtech Extruder upgrade people reported issues with E ustep 32 and E830.
Do you compile your own firmware? If so what changes do you apply to your custom firmware to be able to run the Bondtech Extruder?

3d-gussner, 2020-06-08:
@rcaslis Which firmware do you use? Can you please share the `M503`? Also see my comments above.

3d-gussner, 2020-06-08:
@Kachidoki2807 and everyone else interested:
Can you try to print https://github.com/prusa3d/Prusa3D-Test-Objects/issues/12 and report back the results?
@Kachidoki2807 can you share your star as 3mf file? I guess you have seams aligned or back, please changes these to nearest or random to see if you get better results with the different seams settings.

DanielSan3318, 2020-06-08:
No, I have no problem with Bondtech. Uthe official and untouched Prusa firmware 3.8.1 and the customized Z_top to pass the calibration.

Z_Top for Bondtech (Original MK3 / MK3S)
https://www.thingiverse.com/thing:3913567

Z_Top for Bondtech (Full Bear MK3 / MK3S)
https://www.thingiverse.com/thing:3667401

Z_Top for Bondtech (Full Bear Linear Mod MK3 / MK3S)
https://www.thingiverse.com/thing:3849454

So you can use the original FW without making any changes.

And I came to the E852 where I calibrated the extruder and not as usual the E830 at Bondtech.

Secondly, I am currently unable to use my printer because I am installing my power supply, another fan and therefore I am unable to provide any details (M503) at the moment. :-)

ezu00261, 2020-06-08:
It seems that the defect coincide with retractions.
[image]

DanielSan3318, 2020-06-08:
> @rcaslis Welche Firmware verwenden Sie? Kannst du bitte das teilen `M503`? Siehe auch meine Kommentare oben.

No, I have no problem with Bondtech. Uthe official and untouched Prusa firmware 3.8.1 and the customized Z_top to pass the calibration.

Z_Top for Bondtech (Original MK3 / MK3S)
https://www.thingiverse.com/thing:3913567

Z_Top for Bondtech (Full Bear MK3 / MK3S)
https://www.thingiverse.com/thing:3667401

Z_Top for Bondtech (Full Bear Linear Mod MK3 / MK3S)
https://www.thingiverse.com/thing:3849454

So you can use the original FW without making any changes.

And I came to the E852 where I calibrated the extruder and not as usual the E830 at Bondtech.

Secondly, I am currently unable to use my printer because I am installing my power supply, another fan and therefore I am unable to provide any details (M503) at the moment. :-)

3d-gussner, 2020-06-08:
> No, I have no problem with Bondtech. Uthe official and untouched Prusa firmware 3.8.1 and the customized Z_top to pass the calibration.
>
> Z_Top for Bondtech (Original MK3 / MK3S) [...]

Now I am confused are you both?

DanielSan3318, 2020-06-08:
no

3d-gussner, 2020-06-08:
> And I came to the E852 where I calibrated the extruder and not as usual the E830 at Bondtech.

It is always good to calibrate the extruder. Please keep in mind that Bondtech lowered the E usteps for a reason from 32 to 16 set E steps to 415.

Wish you good luck with your upgrade.

DanielSan3318, 2020-06-08:
My mistake, I didn't know they were different person.

DanielSan3318, 2020-06-08:
I just don't feel like it right now because I have to solder the plug. The original fan of the LRS-350-24 is so loud, I'm curious to see what the Noctua NF-A8 FLX runs.

3d-gussner, 2020-06-08:
> Thanks a lot
> I don't feel like it right now because I have to solder the plug. The original fan of the LRS-350-24 is so loud that I'm curious to see how the Noctua NF-A8 FLX runs.
@DanielSan3318 
Be careful that you don't destroy/overheat your PSU ... most times the Noctua fans have lower air flow and pressure compared to the stock one.

DanielSan3318, 2020-06-08:
> > Vielen Dank, 
> > ich habe momentan keine Lust dazu, weil ich den Stecker löten muss. Der ursprüngliche Lüfter des LRS-350-24 ist so laut, dass ich gespannt bin, wie der Noctua NF-A8 FLX läuft.
> 
> @ DanielSan3318 Achten
> Sie darauf, dass Sie Ihr Netzteil nicht zerstören / überhitzen ... Meistens haben die Noctua-Lüfter einen geringeren Luftstrom und Druck als die serienmäßigen.

Hence the 80mm instead of the 60mm fan.
I recently got a printer case where I reach 40 crad and don't know if the original power supply (black) lasts long and the MeanWell is specified up to 70 crad.

[45 more replies]

closed 2020-09-22 by DRracer