prusa3d/Prusa-Firmware #2694: [BUG] MMU2 stuck after unload with firmwares 3.9.0 + 1.0.6

myrecek73, 2020-05-28:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
After firmware update (3.8.1+1.06 -> 3.9.0+1.06), MMU sometimes behaves strange:
On filament unload, extruder stepper seems to stay enabled and MMU is not able to pull the filament out. When I press the MMU2 middle button to try to fix the issue, MMU2 steppers do not move. They do not even move after reset (the X button). They start working again after switching the printer off and on.
Tried to flash the firmware again, also tried MMU2 factory reset. The problem persisted.
I downgraded the printer firmware back to 3.8.1 (kept MMU 1.0.6 as it had already been there before). The problem did not occur again
  
**To Reproduce**
  Happens randomly at filament unload during any multi-material print, e.g. Prusa MMU2 Sample Gear and Pliers.
  
**Expected behavior**
  MMU2 does not stop reacting
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
[gear_0.2mm_PLA_MK3SMMU2S_3h43m.zip](https://github.com/prusa3d/Prusa-Firmware/files/4696588/gear_0.2mm_PLA_MK3SMMU2S_3h43m.zip)

Mirarkitty, 2020-06-06:
This is not a software problem according to Prusa, but a hardware issue. They sent me a new board, which works fine. Contact support, they should know about this issue.

Typical issue is that the MMU2 hangs until power cycle and can't be reset with the reset button on the MMU2 or the reset button on the printer.

cbrinker, 2020-09-15:
I have precisely the same problem. Trying to get in touch with support now. Strange too as the machine had been rock steady reliable fo a few months until this started happening, now it has been a reliability nightmare.

cbrinker, 2020-09-15:
Contacted support, they said that this might be due to a bug with 3.9.0. I backrev'd down to 3.8.1 (with 1.0.6 on mmu2) and my prints are working reliably again. Support said that ver 3.9.1 (currently in RC) will likely solve the underlying bug.

Also, if you are willing to manually feed the filament into the hotend, and retract it at the end of the print, you can still get successful prints with this bug, it's just not convenient :)

closed 2023-10-17 by github-actions[bot]