prusa3d/Prusa-Firmware #2695: [QUESTION]<question mark fixed. How can I verify?>

mullcom, 2020-05-28:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**What is your question?**

I have read all statement of fixes and I wondering if this down below..

Unknown Z coordinate displayed
In a situation when the printer doesn't know the exact Z-axis position (e.g. after the power on) a question mark is displayed on the screen next to the Z coordinate.

There was also an issue with displaying the “?” instead of the correct Z-axis coordinate caused by the behavior of the black Delta PSU. This is now fixed.

How do I know this is fixed. I have this still after a calibration home ..

See some video and IMG below

https://photos.app.goo.gl/C4CWM2q66MkQ5Joz9[Link here video and img.](https://photos.app.goo.gl/C4CWM2q66MkQ5Joz9)

leptun, 2020-05-29:
@mullcom The “?” Means that the printer is not sure whether it’s Z coordinate is showing is accurate. As a result, every time you reset the firmware the printer looses track of it’s actual Z coordinate. The firmware simply doesn’t know if it is near the bed or 100mm high at that point.

The issue we had with the black psu delta was that the “?” would return for no reason even though the position was known. This would happen after a print or when the steppers were disabled.

mullcom, 2020-05-29:
> @mullcom The “?” Means that the printer is not sure whether it’s Z coordinate is showing is accurate. As a result, every time you reset the firmware the printer looses track of it’s actual Z [...]

Yes I understand the first part. Of course it don't know position after disable power for printer.

But I want to understand what it dose the same when pressing exit button. Also why it not find its position when you do auto home calibration?

leptun, 2020-05-29:
I'm not sure I understand he question. Are you asking why it shows "0.15" even after auto-homing?

awenelo, 2020-05-29:
I believe that by the "exit" button, you mean the X button beside the encoder. That button resets the printer, not exits the menu you are in. A reset is just like removing power from the printer, and thus the printer no longer knows it's location.

mullcom, 2020-05-29:
> I'm not sure I understand he question. Are you asking why it shows "0.15" even after auto-homing?

Sorry i take up the time. but now i know that X button are as power off. 

when i take a look my self again on the video i didn't actually see the ? mark. 

But it was the same 0.15. So in other words auto home calibration seams working to get read of that ? mark.  i suppose that 0.15 is that value that it should be.

Thx for all.

closed 2023-10-25 by github-actions[bot]