prusa3d/Prusa-Firmware #2696: Printing crashed on 3.9.0, worked fine on 3.8.1

KrzychuM, 2020-05-28:
I'm printing adapters for the facemasks against COVID. Gcode created using PrusaSlicer 2.2.0 So before using the firmware 3.9.0 I printed 15 times this model without any single issue.
After update problems started occurring. Printing is crashing, but not every time. During printing attached gcode on new version of the firmware 3.9.0 it happened every 3-4 printing.  I had this issue 4 timesso far.
There are two pairs of crashed printed models in two different moments (first pair ~9-10 mm printing height, and 2nd ~35mm). Looks it likes to crash in specific heights: ~10mm or 35mm .
See pictures od these models.

Let me explain more details.
First 3 times it happened when I was using OctoPrint. I noticed info from the terminal (attached at the end). But then I decided to use direct print from memory card. And again the issue occurred. But this time I have no idea what happend, As I just saw that printing of this model stopped. The extruder always during this issue was on the position when it stopped (even using OctoPrint or just direct from memory card). In this prining from the memory I can say that the display shows normal screen like when the printer just turned on. When I checked errors in the menu from the last print I had only zeros. No errors.

So I decided to check if this issue can occur on the previous firmware again. I downgraded to 3.8.1 and printed this model already more than 10 times without any issue like that.

You can find this gcode and pictures here [LINK](https://drive.google.com/drive/folders/1i3AwRPSESRBM4snXasPdBd-Xf2ekEfP2?usp=sharing)

OctoPrint code:
```
Send: N137525 G1 X128.986 Y110.301 E0.02691*87
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Recv: \x00start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Send: N0 M110 N0*125
Changing monitoring state from "Printing" to "Cancelling"
Recv: echo: 3.9.0-3421
Recv: echo: Last Updated: May 18 2020 17:11:15 | Author: (none, default config)
Recv: Compiled: May 18 2020
Recv: echo: Free Memory: 1675  PlannerBufferBytes: 1792
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect ENABLED!
Recv: FSensor ENABLED (sensor board revision: 0.3 or older)
Recv: Sending 0xFF
Recv: echo:SD card ok
Recv: echo:busy: processing
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Send: N0 M110 N0*125
Changing monitoring state from "Cancelling" to "Error: Printer requested line 1 but no sufficient history is available, can't resend"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Send: N3 M140 S0*102
Changing monitoring state from "Error: Printer requested line 1 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 1 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
Closing down send loop
```

drelich, 2020-10-24:
Just had the same issue. One of the many as of late, actually. The first time it only ruined a 53 hour print!!! Firmware ver. 3.9.1.

I thought it's OctoPrint messing up but found out in some reddit posts from 2018 that it indeed was a FW problem and that it was fixed. Obviously, the issue is back...

InfotecDehmer, 2020-11-02:
Printer type - MK3S
Printer firmware version- 3.9.1

MMU Upgrade - No

SD card or USB/Octoprint
****SD card and Octoprint 1.4.2

Similiar issue on my end. After upgrading to 3.9.1 printer stops midprint when printing from octoprint without an error message. The extruder stays over the print and doesnt move into a parking position . In Octoprint it shows timeouts in the terminal view.

closed 2023-10-25 by github-actions[bot]