prusa3d/Prusa-Firmware #2697: [QUESTION] Electromagnetic field on the Heatbed?

alomex21, 2020-05-29:
Does this affect the MK42 Heatbeds? Some people keep telling me that there are no magnets, which I doubt they read the changelog correctly. My primary question is how did you discover this issue and what result gave so far before this method of turning off the heated?

Cheers.

floridaservices, 2020-05-29:
Part of the reason that Prusa developed their own board definition for Arduino is because they created their own timer for the heat bed.  I don't remember all the details of it(was a tech blog posting so it was lengthy!) but apparently the old way of doing things made the heatbed "noisy" , like audibly apparently.  I also run a MK42 heatbed, and suffer no ill effects from running 3.9.0 other than having to adjust some mesh_bed_calibration.cpp parameters.

alomex21, 2020-05-29:
@floridaservices Now I have two questions: How did you get the 3.9.0 with the MK42(or MK2s) and is there any link to that tech blog?

leptun, 2020-05-29:
For getting a MK2S build with FW 3.9.0, you must compile the firmware on your own.
Link to the blog post he is talking about: https://blog.prusaprinters.org/dev-diary-2-how-we-made-the-heatbed-silent_30946/

alomex21, 2020-05-29:
Thanks for the link @leptun! I've read the blog(quite cool) but I did not found anything related to the electromagnetic field which my question is still on point.

leptun, 2020-05-29:
The PINDA v2 had issues during Mesh Bed leveling where if the bed heating PWM was active while probing, then the measurements could be off and the procedure would fail. I'm not sure if magnets have any play in this, but rather the fact that high current is flowing through the bed traces.

mullcom, 2020-05-29:
I think this soved my issue.. before on 3.8 it sometimes needed to calculate same spot twice. 

Now on 3.9 it do all spots fine as it should be. I also get the value more perfect. But that can be Frome I move the printer to other places.

leptun, 2020-05-29:
@mullcom It's not just the bed PWM that is affecting the MBL repeatability. During testing me and @3d-gussner discovered that having a spool mounted on top of the frame greatly reduced the wobble in the frame, hence why it failed less often. Unfortunately, my printer runs exclusively on 2.2Kg spools, so I can't mount them on the frame for dampening.

mullcom, 2020-05-29:
hmm. i still have some trubbel that calibration for mech is stooping mitt in the calibration. and it doing calibration of Z axel. I going to try to get it on camera

mullcom, 2020-05-29:
Here is a video of the loops it doing after failing the mech bed calibration.

https://photos.app.goo.gl/mHYwsTuggaU3vnFeA

mullcom, 2020-05-31:
I was need to reset printer from beginning and do Wizard calibration. now i can do mech bed calibration

closed 2023-10-25 by github-actions[bot]