prusa3d/Prusa-Firmware #2699: [BUG] FRENCH LANGUAGE DON'T WORK WITH FW 3.9.0

bendisco51, 2020-05-29:
Printer type : MK3S with MMU2S
Printer firmware version : 3.9.0

MMU Upgrade : MMU2S
MMU upgrade firmware version : 1.0.6

SD card or USB/Octoprint : N/A

Describe the bug :
After firmware flash (3.9.0), I select French language and validate, the printer return to language selection. English is OK.

leptun, 2020-05-29:
Please post the PrusaSlicer logs after the firmware upload. Uploading a firmware with languages via Octoprint is not supported.

bendisco51, 2020-05-29:
avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\benoit\Desktop\prusa3d_fw_3_8_1_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_8_1_2869.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\benoit\Desktop\prusa3d_fw_3_8_1_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_8_1_2869.hex"
avrdude-slic3r: writing flash (241630 bytes):

Writing | ################################################## | 100% 42.29s

avrdude-slic3r: 241630 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\benoit\Desktop\prusa3d_fw_3_8_1_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_8_1_2869.hex:
avrdude-slic3r: load data flash data from input file C:\Users\benoit\Desktop\prusa3d_fw_3_8_1_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_8_1_2869.hex:
avrdude-slic3r: input file C:\Users\benoit\Desktop\prusa3d_fw_3_8_1_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_8_1_2869.hex contains 241630 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 30.94s

avrdude-slic3r: verifying ...
avrdude-slic3r: 241630 bytes of flash verified

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

leptun, 2020-05-29:
This is all?? From what I can see in these logs, PrusaSlicer failed to detect the fact that the file provided included languages. That's why the firmware fails switching languages.
Try reinstalling PrusaSlicer or try downloading the file again on another computer and try again

bendisco51, 2020-05-29:
I revert firmware to 3.8.1 and reflash 3.9.0, French language work now.

bendisco51, 2020-05-29:
I reflash 3.8.1, french language is ok.
Try to flash 3.9.0 after redownload firmware and reinstall prusa slicer. Same problem.
Prusa Slicer Log :

avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\benoit\Desktop\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\benoit\Desktop\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex"
avrdude-slic3r: writing flash (248818 bytes):

Writing | ################################################## | 100% 44.80s

avrdude-slic3r: 248818 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\benoit\Desktop\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: load data flash data from input file C:\Users\benoit\Desktop\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:
avrdude-slic3r: input file C:\Users\benoit\Desktop\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex contains 248818 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 31.85s

avrdude-slic3r: verifying ...
avrdude-slic3r: 248818 bytes of flash verified

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

avrdude-slic3r -v -p atmega2560 -c arduino -P COM3 -b 115200 -D -u -U flash:w:1:C:\Users\benoit\Desktop\prusa3d_fw_3_9_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK3S_3_9_0_3421.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Mar 21 2020 at 12:45:57
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM3
                Using Programmer              : arduino
                Overriding Baud Rate          : 115200
avrdude-slic3r: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude-slic3r: arduino_open(): Failed to initialize MK3 external flash programming mode
avrdude-slic3r: Could not open port: COM3

avrdude-slic3r done.  Thank you.

leptun, 2020-05-29:
This is a duplicate of #2675

3d-gussner, 2020-05-29:
@bendisco51 
Can you please post the PrusaSlicer firmware upload with FW3.8.1 as I don't see the 2nd writing to the external flash compared to the FW3.9.0
With OS, version, build nr do you use to flash the firmware?

bendisco51, 2020-05-29:
I try with 2 computer with Windows 10 1909 x64 : same problem. When flashing 3.8.1, Prusa Slicer log is the same (error for the 2nd part but printer and language is ok).

I flash firmware 3.9.0 with a VM on Linux Mint 19.3 (same computer), the flash is OK, French language is OK.

The problem is surely Windows 10.

3d-gussner, 2020-05-29:
Strange thing is that I have a Windows 10 Pro 1909 x64 build 18363.836 and it works.
You are now the 2nd person with that issue.
Can you please try an other PrusaSlicer version like https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.1.1 ?
I don't think something in the avrdude-slic3r changed with PS 2.1.1 -> 2.2.0 but you never know.

Do you have any applications that might try to connect/monitor the COM ports?
If you have a virus scanner can you temporary disable it?

closed 2023-10-25 by github-actions[bot]