prusa3d/Prusa-Firmware #2701: Anyone had problems with their Flashair since 3.9 upgrade?

MR2C280, 2020-05-29:
I've been using a FlashAir card (W03) for years.  Shortly after installing 3.9 it seems to have slowed or been unreliable and now, it won't connect with my router.

I've gone through the usual reboots and reverting to 3.8, changing the setting of the SD card on the printer (MK2.5S), reinitialising the FlashAir card, etc. but now it just won't connect.

I've tried the card in a camera and the WiFi works fine.

The only thing changed in 3.9 that is remotely related to the LCD panel is the new ability dim the brightness on recent MK3s.  From reading other posts, I understand that, setting the SD card to FlashAir on the printer simply boosts the power as the WiFi on the FlashAir needs a bit more.

Any ideas?

leptun, 2020-05-29:
The flashair option on the lcd does almost nothing in the current firmware. A long time ago it used to show the ip address in the support menu. That isn’t working now, but a PR is scheduled for 3.10. No power boosting happens :)
I have a flashair W04 and didn’t notice any issue. It sounds a lot like a hardware issue with the card.

Hakadek, 2020-05-29:
Yes, mine has stopped connecting right after the 3.9 update. Only just started to do troubleshooting, so not definitive yet. I'm on W04 BTW.

MR2C280, 2020-05-29:
Hakadek, 

I know what an effort it is to troubleshoot this sort of problem.  What I tried was:
- Reinitialise the card using the FlashAir software and recheck all the settings (many times).
- Reverted to 3.8 and to 3.7
- Set the SD Card type to Normal (and back again after restart)
- Unplugged and reinserted the ribbon cables
- Tried the card in a camera (where it worked as it was able to connect to my router).

I've now ordered a new W04 FlashAir card to see it that solves the problem.  If you hadn't responded as you did, my next step was to order a new LCD screen from Prusa (but that's starting to get expensive).  I assume Prusa monitors this site and will see this issue.  If more people respond with the same problem, I'll raise this as a bug.

Hakadek, 2020-05-29:
I'm just waiting for a long print to finish then I'll start work on the problem. The odd thing is that like you my initial feeling was that the connection quality was degrading but still working and then it stopped connecting to the network completely. 

I will be able to verify from file timestamps once I finish the current print, so more later...

Hakadek, 2020-05-30:
Update: FlashAir is now working again. 

I've checked timestamps and yes, I can confirm that following the update to 3.9 it was still working but file transfers were very slow, getting worse and then the connection was lost completely. 

Multiple power-cycles of the Prusa made no difference, the FlashAir was not pingable and the mapped drive on my PC would not connect to the card.

Put card in camera and it was immediately pingable.

Checked config file on card, saw no problems so didn't change anything.

Put FlashAir back in Prusa and was able to ping card again.

Checked PC and mapped drive was also working again, but could not copy files from card to PC, getting error 'an unexpected error is keeping you from copying the file'. This was fixed by adding the IP address for the card to 'Internet Options'. 'Security', 'Local','Sites'.

This had not previously been needed. 

All now working normally through a couple of power cycles.

I've checked for Windows Updates as well, but nothing was installed recently other than the usual daily Windows Defender Updates. 

No further checking I can perform at this point as all is working as expected. 

In summary, no configuration changes made to Prusa Mk3S, no changes made to FlashAir. All working as expected. One security change required to Windows 10, but only after the card was connecting, but not allowing copies from the card.

leptun, 2020-05-30:
I’ve also had weird issues occur to me with the W04 flashair that had nothing to do with the printer firmware (connected at my computer with an SD reader). Most of these issues are fixed by a full reformat of the card in windows.

The first weird issue I had was not being able to open the web interface where it shows all files with a “too many redirrects” error. Oddly enough, if I accessed the card through a proxy it worked correctly.

Another issue I had was that I couldn’t upload files to it anymore and windows couldn’t access it anymore. Turns out it was because of a filename including % or - (not sure). Apparently that messed up the proxy.

And probably the only issue I was not able to fix was that I can’t upload large files (60MB+). It is a limitation of the flashair.

Hakadek, 2020-05-30:
I have multiple files well over 60MB that have been uploaded to the FlashAir W04 64GB card...

The largest so far is 91.3MB.

There is a default Web Client limit around 47MB, on copying files from the card to a Windows PC, but that is easily fixable with a Registry edit (HKEY_LOCAL_MACHINE -> SYSTEM -> CurrentControlSet -> Services -> WebClient -> Parameters called FileSizeLimitInBytes)

hergtoler, 2020-06-08:
After working fine for over a year, my FlashAir W04 also stopped working a few months ago.  After fighting with it a few days, I finally got it working by moving one of my access points to the same room as the FlashAir.  I has worked well since then.
I wish there was a way to have the printer connected over CAT6.  I have Octoprint (wired), but I don't like using it to control prints, only for monitoring.  Using it to transfer to the SD card is an exercise in patience.

Hakadek, 2020-06-08:
> I finally got it working by moving one of my access points to the same room as the FlashAir. I has worked well since then.

One thing you might want to do is make sure you have this command in the vendor section of the config file:

WLANAPMODE=0x82

That will ensure that 802.11ng mode is activated, otherwise it bg by default and most of us will want n these days.

rhounsell, 2020-06-23:
> One thing you might want to do is make sure you have this command in the vendor section of the config file:
> 
> WLANAPMODE=0x82
> 

My card stopped being visible on my home network after I upgraded it to Google Nest Mesh Wifi. After adding this, and also adding STA_RETRY_CT=0, my card suddenly was connected to my network again. Thanks for the tip.

gogglespisano, 2020-06-28:
> The first weird issue I had was not being able to open the web interface where it shows all files with a “too many redirrects” error. Oddly enough, if I accessed the card through a proxy it worked correctly.

Add REDIRECT=0 to the FlashAir CONFIG file.

closed 2023-10-25 by github-actions[bot]