prusa3d/Prusa-Firmware #2703: [ENHANCEMENT]Automatically recover when SD card is temporarily removed during a print

esb7, 2020-05-29:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - all
**MMU Upgrade** - all - N/A

**Is your enhancement related to a problem? Please describe.**
Many users have issues with a flaky SD card slot.  This can cause the card to disconnect during a print which then causes the print to stop with no means of recovery.
  
**Describe the solution you'd like**
It would be very useful to pause the print when this occurs and allow the card to be re-inserted.  Once re-inserted, the print should be able to be resumed (assuming no changes to the file being printed).

rdragonrydr, 2020-06-09:
I agree that this may be useful. I suggest that this not be integrated into the power-panic code, however, at least as far as the save to EEPROM goes. It may be useful to reuse that resume feature, but there would be no point in saving the current line in persistent memory since the printer presumably remains on during this procedure.

polymetric1, 2020-06-17:
I've been having this issue occasionally where my printer just stops in the middle of a print, and says "Prusa MK3S OK." on the LCD.

What's happening is that the pins in the SD card reader are extremely sensitive to tiny movements - simply pressing the encoder or tapping the side of the LCD unit can make the card connect and/or disconnect. I could replace the LCD/reader unit, but the pins may just become loose again. This started happening mere months after receiving the printer and has only gotten worse over time (more likely to disconnect with smaller bumps and jostles).

With all the failsafe measures in the MK3, I'm surprised there isn't one for the SD reader failing yet. It's happened to me dozens of times in the two years that I've owned this printer, and I've lost tons of time and filament from it. It seems like it would be trivial to implement this feature.

Obviously, the correct solution would be to replace the reader/LCD unit, but that's a pain to do. Why should I pay $30-50 for a new reader when the firmware could easily automatically pick up from where it left off? I don't really understand why it even cancels the print when the SD is removed - it should just pause the print when the SD card disconnects. Seems like that could be done with like, 3 lines of code. I'm only whining about this because I have no idea how to do it, but someone more familiar with this codebase would probably be able to do it in like 5 minutes.

The point is, it's not just a convenience thing. It's literally a huge problem that I've been having that seems like it could be fixed trivially.

Someone let me know if I should start a separate issue for this.

esb7, 2020-06-17:
That's the specific intent of this enhancement request.  If the SD card is removed during a print, pause the print and wait for it to be re-inserted.  As long as the file is still present (and perhaps has the same timestamp/size), automatically re-start the print.  This seems like a simpler problem than power panic and potentially will have a bigger impact as my SD card disconnects much more often than I lose power.

closed 2023-10-25 by github-actions[bot]