prusa3d/Prusa-Firmware #2710: [BUG]Extruder motor is getting very hot, after pausing a print

m50008, 2020-06-03:
Printer Type: MK3S
Printer firmware version:  3.9.0
Print via SD card

When pausing a print and letting it sit for a while, the extruder motor is getting so hot, that it can not be touched anymore (feels hotter than the 60 degree warm heatbed). After pausing the print, the extuder motor made some strange noises. 
The hotend was switched of and the temperature of the hotend was at 26 degree.

leptun, 2020-06-04:
The Extruder stepper is run at a very high current. Oddly enough, the faster you move, the less heat is made, so fast prints should be better than very slow ones. When the motor isn't moving it has to pass through one of the two coils all the current continuously.
I'd say what you wrote here is a duplicate of this issue #1757. Can you confirm and close the issue if that's the case?

mkbel, 2020-06-17:
It doesn't look like duplicate. This issue is more specific than #1757. In #1757 E motor is said to be too hot in all modes of operation. In this issue only pause print mode is mentioned. Root cause of the issue may be but doesn't need to be the same. I could imagine, that when X axis moves, E motor is better cooled because of air movement around motor.

m50008, 2020-06-19:
Yes exactly. The question is whether it is necessary to leave the motor switched on during a paused print? The hot end is switched of, so should the E motor as well?

3d-gussner, 2020-06-19:
@m50008 Switching off the extruder motor could change the pressure in the hotend, as the filament isn't hold in position.
This could cause:
- extra oozing
- or missing filament when it starts again. 

A pause should be always as sort as possible. So if you have a pause in your gcode add some sound code to it so you get kind of notified that your attention is required.

m50008, 2020-06-19:
@3d-gussner Thanks for the response, but imagine you print several hours and want to pause the print, because the printer is to loud and you have to do something important in the same room as the printer. 
Here it would be very nice to pause and disable the E motor and when starting again, we extrude a little filament and ask the user, if anything went out. If we accept, we can resume the print.

Prusa-Support, 2022-05-21:
Thank you for sharing your ideas.

The high temperature of the stepper motors should not be concerning for the motors themselves, which can survive at very high temperatures - Prusa stepper motors would probably work just fine even above 100°C.
Also, the Prusa design rather thick PETG printed parts of the extruder-block have proven themselves to withstand exceptionally well to the relatively high temperature of the Extruder stepper motor - if anyone experiences otherwise, please let us know.

Probably the concern here would move to low-temperature filaments (like PLA) which may probably get softer and slightly deformed after long exposure to the Extruder motor temperature during long pauses.
On the other hand, some heat would be dissipated through the motor shaft and the extruder gear. Surely a warm ambient and an over-tightened idler-tension-bolt can have an impact on this too.
Please let us know if you have experience with deformed filament after a long pause.

Another side effect of keeping the motor engaged during a long pause is the slight but not totally negligible waste of energy.

An idea of feature request could be a separate option to disengage the motors if the user plans to pause the print for a "long" time or a safety timer for the motor to be automatically disengaged. The printer would probably require an XY-homing and a purge before resuming the print.

The issue could be closed as stale, but I'd keep it open for now.
I've removed the "bug" label (probably improper) and added a "feature request" label.

Michele Moramarco
Prusa Research

closed 2023-10-25 by github-actions[bot]