prusa3d/Prusa-Firmware #2714: [FEATURE REQUEST]<Toggle between Mk3S MMU2s amd MK3s>

tedukes, 2020-06-04:
If possible, it would be nice to be able to toggle the MMU2S ON or OFF from the LCD. Doing calibrations that require printing the same file with the same filament numerous times, I see no need to be loading and unloading the filament each time.

Also, a lot of profiles and projects are not for the MMU2S.

Thanks!!

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

**Is your feature request related to a problem? Please describe.**
  No
  
**Describe the solution you'd like**
  A simple menu item to toggle the MMU2S ON or Off

JKinch03, 2020-06-15:
Yeah, I did the same hannoHI.

@tedukes, what you can do is simply disconnect the power (2 Red & 2 Black cable bundle), and the data (Blue, White, Green & Brown cable bundle) from the back of the MMU using a small hex wrench. It's pretty simple.

3d-gussner, 2020-06-19:
It is possible to get the toggle menu, BUT as the MMU2S has the IR sensor tower it makes things already complicated:
- If we disable the filament sensor then users will complain if they run out of fillament
- If we let the sensor enabled users will complain that the filament cannot be pulled out as they already passed the gears.

tedukes, 2020-06-19:
> 
> 
> It is possible to get the toggle menu, BUT as the MMU2S has the IR sensor tower it makes things already complicated:
> 
>     * If we disable the filament sensor then users will complain if they run out of filament
> 
>     * If we let the sensor enabled users will complain that the filament cannot be pulled out as the filament already passed the gears.

Thanks,
Is the ball and magnet sensor for the MK3S handled differently in the firmware? I don't understand the second point you made.

tedukes, 2020-06-19:
What if the start and end gcodes for a non-MMU2S was used? Don't mind manually loading and unloading filament. 

It just makes no sense when doing calibration files or printing multiple prints of the same file and same filament, to go through the load and unload process.

3d-gussner, 2020-06-22:
> Thanks,
> Is the ball and magnet sensor for the MK3S handled differently in the firmware? I don't understand the second point you made.

Yes it is:
- MK3s sense the filament with  the IR sensor before the bondtecg gears
- MK3s + MMU2s senses the filament via the FINDA on the MMU2s and again with the IR sensor on the tower **AT** the bondtech gears.

leptun, 2020-06-22:
The thing with the MMU2S extruder design is that it is only useful for knowing when the filament was grabbed. As a result, a filament runout will be detected too late as it is detected as soon as it is past the gears. This is the theory.
In practice, I’ve had 100% success rate even with the sensor at the gears, so you could argue it’s still usable (although not ideal)

alecperkins, 2020-09-02:
Duplicate of #1698 and #1876. Looks like it's slated for the 3.10 release per [this comment](https://github.com/prusa3d/Prusa-Firmware/issues/1876#issuecomment-591898496) though I would very much like it sooner as well 😄

JKinch03, 2020-09-02:
Wahooo!😄 I hope they come out with it soon🤞

JohnnyDeer, 2021-08-09:
I´m closing this issue as duplicate.

closed 2021-08-09 by JohnnyDeer