prusa3d/Prusa-Firmware #2717: [ENHANCEMENT]<Turn OFF print fan while changing the filament during a print>

majedation, 2020-06-05:
**MK3S**
**Firmware 3.9.0**
**PrusaSlcier 2.2.0**

When the filament runs out during a print, and while changing filament during a print, the print fan stays ON at full speed (for PLA). That pushes the new extruded/purged filament towards the printed part.

**Suggestion**
My suggestion is to turn OFF the print fan during the filament change, then turn it ON again when print resumes.

flyingmarvin, 2020-06-08:
I would like this feature as well, sometimes I even had my print being removed from the build plate because a full blowing fan at the edge of the print for a long time caused too much warping.

gudnimg, 2022-12-10:
This was fixed recently here https://github.com/prusa3d/Prusa-Firmware/pull/3746 (will be included in 3.13)

Prusa-Support, 2023-01-03:
Thank you for your collaboration.

This issue will be closed in the same way as issue https://github.com/prusa3d/Prusa-Firmware/issues/3670 that led to such Pull Request.

Michele Moramarco
Prusa Research

closed 2023-01-03 by Prusa-Support