prusa3d/Prusa-Firmware #2718: [BUG]M112 Error

rmilyard, 2020-06-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**- 3.9.0

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**SD card or USB/Octoprint**
Either

**Describe the bug**
At random times since updated firmware getting M112 errors and print stops.
  
**To Reproduce**
 Print again
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

leptun, 2020-06-05:
Can you please give the firmware I posted here a try? https://github.com/prusa3d/Prusa-Firmware/issues/2675#issuecomment-636038008
The optiboot fix should eliminate most of the unintended M112 causes.

rmilyard, 2020-06-05:
> Can you please give the firmware I posted here a try? [#2675 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/2675#issuecomment-636038008)
> The optiboot fix should eliminate most of the unintended M112 causes.

Right now I am using the Bondtech version for my Extruder:
https://github.com/BondtechAB/Bondtech_Prusa_i3/blob/master/Firmware/3.9.0/FW390-Build3922-Bondtech-MK3S-16-EINSy10a-SliceE_HT-Mosquito_Magnum.zip

cwbullet, 2020-08-27:
What hotend and themristor are you using?

closed 2023-10-18 by github-actions[bot]