prusa3d/Prusa-Firmware #2719: Portuguese Language translation from Portugal

isulpt, 2020-06-05:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK3S
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

**Is your feature request related to a problem? Please describe.**
The portuguese speakers don't have any language.
  
**Describe the solution you'd like**
i made a translation from the english po file to Portuguese. If you want i can send to you to include in the firmware.

3d-gussner, 2020-06-08:
@isulpt Can you provide the po files?

isulpt, 2020-06-08:
Sorry the delay. It's here. Translated by me in Portuguese - Portugal (pt-pt)

[Firmware_pt.zip](https://github.com/prusa3d/Prusa-Firmware/files/4746604/Firmware_pt.zip)

isulpt, 2020-06-08:
Anything i can help, please tell me :)

3d-gussner, 2020-06-08:
Thanks for the translation, I will try to add this in to my github.
Let's hope Prusa will allow in the future to provide community translations.

Have you read https://github.com/prusa3d/Prusa-Firmware/blob/MK3/lang/translations.md ?

I can see that you use í é and other Western European characters, BUT as we are have a LCD that has only Japanese ROM it is not possible to use these.

I am not sure if it is allowed to write without ´ ` and other things.
Also you can see in most messages that these are limited in chars and rows by:
c=xx r=yy
xx = max allowed chars
yy = max allowed rows

So if you can please review your Firmware_pt.po file.

isulpt, 2020-06-08:
Didn't see that. Will review the translation and sent again to you.

danielalexis, 2020-06-15:
Isn't it better to make a PR?

3d-gussner, 2020-06-15:
Sure, but preparing a language PR isn't easy if you don't know how to do it.

closed 2023-10-25 by github-actions[bot]