prusa3d/Prusa-Firmware #2722: [BUG]<Heaters stay on after "Crash Detected">

martinglass, 2020-06-06:
**Printer type** - MK3S
**Printer firmware version**-  3.8.12869
**SD card or USB/Octoprint** - Printing via USB/Octoprint

**Describe the bug**
The bug is that the Heaters stay on after a Crash is detected.

Last night, I was printing with PETG, using supports. The printed object warped and the Printer had a "Crash Detected" error. The crash occurred because of the warping - one side of the print raised itself up off the bed, and the printer detected a crash when it got to that area which was higher. I was using Octoprint to start the print job. 

The Octoprint records showed that the crash error occurred about 1am, while I was alseep. Thereafter, Octoprint was waiting for User Input, while the MK3/S printer was waiting for User input to clear the Crash Detection flag.

However, when I woke up I noticed the printer bed and hotend were at full temperature, still 7 hours later! That was very unexpected and could have been dangerous, no?  

It is very alarming that this could happen. If the printer detects a crash and stops, but keeps the heaters on, that's an unsafe condition. No?

**To Reproduce**
  To recreate the issue, the printer will need to raise a "Crash Detection" flag. Then, check whether the bed heater and the hotend are still on, or if they have been turned off without usr input.
  
**Expected behavior**
  I would have expected the heaters to go off, at the same time as (or shortly after) the head is retracted home.

DFliyerz, 2020-06-11:
I believe the heated bed is supposed to stay on so your prints don't detach, but the nozzle it supposed to turn off. I've had multiple overnight crashes but only the bed has stayed on, the nozzle has cooled down.

martinglass, 2020-06-11:
Yeah - you could be right. I can be sure the heated bed remained on, but I didn't check the nozzle and I haven't tried to recreate the problem either. I think you're right the nozzle gets turned off, to prevent a dangerous runaway, but the heated bed isn't considered a safety hazard (just a cost issue!!). I guess this is not a valid bug report then, so I'll close it.

closed 2020-06-11 by martinglass