prusa3d/Prusa-Firmware #2723: Unload Filament Stops Working the First time

Doc1455, 2020-06-06:
Every time since the 3.9.0 update when ever i go to the Menu to Unload Filament I choose PLA  or Petg and it starts to heat up and after 10 seconds it goes to the main menu screen and stops working.
If i do it again it works fine.

Prusa-Support, 2022-06-15:
Hello! Sorry for intervening so late on this thread. Is the problem still there at this time, even with the firmware versions released in the meantime? In case the issue is still there, can you please send me a video of it? Any errors shown on the screen, and does the printer stop heating in that case? 

Alessandro Pantaleo
Prusa Research

gudnimg, 2023-07-03:
@Prusa-Support should we close this ticket as it seems to be stale. I don’t think the described issue is present in the firmwares anymore.

Prusa-Support, 2023-09-20:
I think we haven't heard of similar reports so probably it was the case of a temporary glitch or a minor mechanical problem.
Also, the firmware was substantially improved over the years.

It seems safe to close this issue as obsolete now.

Michele Moramarco
Prusa Research

closed 2023-09-20 by Prusa-Support