prusa3d/Prusa-Firmware #2730: [BUG]MMU2 "Needs user attention" is unrecoverable

readmodifywrite, 2020-06-08:
MK3S + MMU2S
3.9.0/1.0.6
SD Card

Anytime I get "Needs user attention" on the MMU (usually it's jammed or something), I correct the issue, ensure the filament is correctly loaded, hit the button, and.... nothing happens.

I've also seen it select the wrong filament on the drum when trying to recover, so that obviously it will never work because it cannot feed the filament into the sensor with the incorrect drum position.

There is no way to get past this screen, even to check sensor status.  All I can do is keep hitting the MMU button hoping something will happen.  Usually nothing does, so I have to hit reset and restart the entire print.

readmodifywrite, 2020-06-09:
Another note - sometimes the "press knob to preheat" message, alternating with "Needs user attention" is actually a lock up condition: pressing the knob does nothing.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

Error screens have been re-implemented to be more helpful

closed 2023-07-31 by gudnimg