prusa3d/Prusa-Firmware #2733: [ENHANCEMENT] Faster changing of temperatures etc

flyingmarvin, 2020-06-08:
Hi, I'm using a MK3S. 
The firmware is great, that's one of the things I love most about this printer, but when I want to set the temperature or fan speed to a given value, this is usually a bit slow because I have to scroll a lot until I reach the desired value. This is especially noticeable when I want to change my nozzle and therefore just want the nozzle to heat up to 285° or when I want to turn the print cooling fan on or off and then have to scroll 255 numbers.

Therefore, my idea for a future firmware update is a different scrolling behavior related to the spinning speed. This means that values change much faster when you are scrolling fast, but when you turn the control knob slower, the values change exactly in the way they are doing right now so you are still able to precisely set your values.
So for example, if you turn the control knob fast, every new physical "click" changes the value about 5 degrees, but if you turn it slowly every new click means 1 degree like right now.
This can then be applied for both temperatures, fan speed and maybe even printing speed and flow.

I would love to see this feature implemented because it would make using the printer even more intuitive and easy.

Have a nice day and stay safe,
Marvin

fiveangle, 2020-12-31:
Came to the Issue list to suggest a similar enhancement to improve setting temperature via LCD then found this.  When printer is off, selecting "Settings:Temperature:Nozzle|Bed:" the temperature starts at "0" and requires a lot of rotary scrolling in order to get to any meaningfully usable temperature, especially for the Nozzle.  While investigating the feasibility of an accelerating rotary temperature adjustment, please consider making the enhancement jump to full acceleration state if the starting value is exactly "0".

Alternatively, if accelerating rotary temperature adjustment is deemed not desired by PR (although why on Earth one would 🤣), consider making the initial value of the "Settings:Temperature:Nozzle|Bed:" an appropriate non-zero value such as "50" for Bed and "160" for Nozzle?

Thank you for considering these time-saving and frustration-reducing UX improvement!

wavexx, 2023-08-02:
@3d-gussner I think this can also be closed as completed via #3494

3d-gussner, 2023-08-02:
@wavexx Yes even not exactly what was proposed as we don't jump in 5°C steps when turning fast BUT as it jumps to 200 it will be faster for the user to reach the target temp without endless knob turning.

3d-gussner, 2023-08-02:
@flyingmarvin Please have a look at #3494 if that solves your request. Please consider the issue (always nicer when the owner of a ticket closes it instead we are doing that).

flyingmarvin, 2023-08-02:
This is completely fine, I'm glad to see that implemented! :)

closed 2023-08-02 by flyingmarvin