prusa3d/Prusa-Firmware #2734: [BUG] Resume menu after pressing reset to cancel

n4mb3r0n3, 2020-06-08:
Printer wants to resume a Print after pressing the reset button to cancel a print. Only power off and on will bring back normal function. 

**Printer type** - [MK3S]
**Printer firmware version**-  [3.9.1]

**MMU Upgrade** - No MMU

**SD card or USB/Octoprint**
  Printing from SD

**Describe the bug**
  Printer wants to resume a print that was canceled with the reset button. 
  
**To Reproduce**
  Start a Print Job and press reset to cancel it. 
 Resume menu appears and can’t be canceled.

eeclark, 2020-06-12:
I did not think FW 3.9.1 was out.  Are you sure this is 3.9.1 and not 3.9.0?

n4mb3r0n3, 2020-06-12:
It is 3.9.1. I am testing it.

3d-gussner, 2020-06-12:
It is more a self compiled version of the MK3 branch where Fw3.9.1 labled PRs have been merged since fw3.9.0 release.

Panayiotis-git, 2020-06-18:
@n4mb3r0n3 I tried to reproduce your issue, but after the reset, the printer waits to the status screen as normal.

closed 2023-10-25 by github-actions[bot]