prusa3d/Prusa-Firmware #2741: [ENHANCEMENT] Adjustable Crash Detection Sensitivity

DFliyerz, 2020-06-11:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3S, MK3

**Is your enhancement related to a problem? Please describe.**
  I often get seemingly "false" crash detections on prints, especially in the y-axis. I usually just turn off crash detection for the rest of the print and don't run into any further issues, but it would be nice if I could still have it on for major crashes.
  
**Describe the solution you'd like**
  Add an option to adjust the crash detection sensitivity either in the printer UI, or through gcode that doesn't require you to flash firmware with //#define TMC2130_SERVICE_CODES_M910_M918 uncommented.

ManuGithubSteam, 2020-06-22:
Yes false crashes are the reason why i have turned that off.

kalleknall, 2020-09-21:
That would be nice. My Linear Rail mods causes false x-crash from time to time (i think).

closed 2023-10-25 by github-actions[bot]