prusa3d/Prusa-Firmware #2744: Extruder Z End Position after abort of current print.

boostfreak, 2020-06-14:
**Current situation:**

Once an print is started and then stoped / aborted by the user, the nozzle stops very close to the printbed.

Mostlikly the print was aborted due to some problem and usually rather early in the prints stage.

That having said, it might be a good idea to leave the nozzle in the air so that there is immidiatly room to:

1. diagnose the printing system
2. clean the nozzle
3. clean the print bed

**Future situation:**

Aussuring an minimal distance to the print bed might be a good compromise of time needed to bring the extruder back in printing position as well as ability to clean / diagnose the system without navigating manually through the menu to drive z-axis in upper position.

This feature "might" also come in handy when a print is paused if during printing there has been some accumulation of material around the nozzle.

boostfreak, 2020-06-16:
[image]

Printer in end position after aborting the current print.

neophyl, 2020-06-18:
Cant you just use the press and hold the control knob and then use the Move Z function ?  No navigation needed around the LCD

closed 2023-10-25 by github-actions[bot]