prusa3d/Prusa-Firmware #2745: [BUG]run time statistic

Etom808, 2020-06-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3S]
**Printer firmware version**-  [3.90]

SD card 

under statistics , it says 54d 10h 30m
and I only owned the printer for 20 day. I did check the statistics during the first week and its said 3day.

3d-gussner, 2020-06-19:
Thanks for the bug report.
Can you connect your printer with Octoprint, Putty or another terminal software. Pronterface doesn't work here. Run a report back the following gcodes commands 
1. `D3 Ax0ff1 C4` 
2. `D3 Ax0fed C4`

The 1st one reads the eeprom value for used filament and 2nd the total time.

3d-gussner, 2020-06-22:
@Etom808 any update on this?

Etom808, 2020-06-22:
Hello,

I'm waiting for my pi zero board to come in the mail.  Can I run these
commands using putty and my computer connected to the printer?

Thanks,
Elton

Etom808, 2020-06-24:
0ff1 98 56 02 00
0fed 71 59 01 00

Etom808, 2020-06-24:
my printer statistics show:
filament: 1532.40
61d 09h 53m

RyanEwen, 2023-01-11:
I have the same issue. My printer shows approx 3800m filament used, which seems about right. However it says 615 days and I've only owned it for about 6 months and don't use it daily (and I use Octoprint).

Prusa-Support, 2023-02-07:
Thanks for reaching us.

We may be looking at memory corruption.
Possibly quickly check all the menus and report if there is any missing menu, broken options, and especially lost values/profiles. You can use the following article as a guide while checking the menus.
https://help.prusa3d.com/article/lcd-menu-original-prusa-i3_142322

Next, a memory dump could be of help.

.
Dumps are automatically saved after a firmware crash (FW crash error, SPI error...).

In case there are no error messages, the dump can be created on the printer's memory via "dump memory" command in the menu, as per this article.
https://help.prusa3d.com/article/dump-memory-mk3-s-and-mk2-5-s_245237
Then the dump can be downloaded as per this article.
https://help.prusa3d.com/article/crash-dump_364959

In order to avoid sharing it publicly, you can send the dump via email to my attention.
_info@prusa3d.com
Subject: Att. Michele Moramarco - Prusa-Firmware/issues/2745_

Michele Moramarco
Prusa Research

3d-gussner, 2023-11-14:
We have found an issue with the pause time calculation and we will fix the stats with https://github.com/prusa3d/Prusa-Firmware/pull/4493

Please consider to close this issue.

closed 2023-12-11 by 3d-gussner