prusa3d/Prusa-Firmware #2751: [BUG] Updated to 3.9.0 Firmware (Mk3s), and caused bad bridging and seam issue?

john5634, 2020-06-20:
Tried to run old gcode for previous firmware, but it just makes the prusa restart?

So i re sliced the files using the 3mf that originally generated the gcode, and the print now looks terrible.

Any suggestions? The issue is in two places (edge going up seems to be a seam setting issue, even though I didn't change the 3mf settings, and the bridging is now terrible in the top center hole)

The one of the left is after the new firmware.

MK3S
3.9.0

No Upgrades or changes

SD card

[older-GCODE-that-makes-the-printer-restart_0.25mm_PLA_MK3S_3h41m.txt](https://github.com/prusa3d/Prusa-Firmware/files/4807329/older-GCODE-that-makes-the-printer-restart_0.25mm_PLA_MK3S_3h41m.txt)

Link to 3mf that was used for perfect prints on old firmware, that no longer makes perfect prints

https://drive.google.com/file/d/1SJ_oDQvqbeupTQVIlz9YH-Ol__7fLIA7/view?usp=sharing

IMAGE SHOWING THE ISSUES:
[image]

3d-gussner, 2020-06-20:
Thanks for detailed bug report. I think it is kind of duplicate of #2693

Can you check out https://github.com/prusa3d/Prusa3D-Test-Objects/issues/13
And report back your results?
Would be great if you also could have a look at https://github.com/prusa3d/Prusa3D-Test-Objects/issues/14 as the included firmware has both potential fixes.

DRracer, 2020-09-21:
All this should be fixed in FW 3.9.1 final, can we close this issue @john5634 ?

john5634, 2020-09-21:
Yeah! Of course! Thank you so much.

closed 2020-09-22 by DRracer