prusa3d/Prusa-Firmware #2753: [BUG] M300 command not working properly when fw is built from source

TilTisback, 2020-06-21:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0

**MMU Upgrade** - no MMU
**MMU upgrade firmware version** - No MMU

**SD card or USB/Octoprint**  - SD card and USB

**Description of the bug**
  M300 S1000 P1000 does not give a 1khz tone for 1 second when firmware is built from source.
I have compiled the firmware from source because i have a different z-height on my printer. After this  the m300 command does not work properly.
 If i send m300 s1000 p1000 it will beep for a very short amount of time. 
If i send m300 s1000 p65535 (0xffff) however, it will beep roughly for a second. 
To me this looks like the input number is not scaled properly to milliseconds.
Maybe this has to do with the recent change of the beeping function to a different timer.

**HOWEVER:**
If i flash the prebuild .hex file from the releases page, the m300 command works as expected.
https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0

**To Reproduce**
  Build the firmware from source and upload it to the printer. issue a m300 command via SD card or pronterface.
  
**Expected behavior**
  M300 S1000 P1000 should give a 1khz tone for 1 second

leptun, 2020-06-21:
Make sure you are using the correct board definition. Use Einsy rambo for both rambo boards.
https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json

TilTisback, 2020-06-21:
Hello, i have double checked that i use the correct board and have the latest arduino ide version.
still the m300 command does not work properly.

[image]

[image]

[image]

leptun, 2020-06-21:
Please try to use Arduino 1.8.5

TilTisback, 2020-06-21:
I compiled it with Arduino 1.8.5 now but the bug is still there.

3d-gussner, 2020-06-21:
Please use PF-build.sh to compile the firmware. You get a known and stable build environment.

3d-gussner, 2020-06-22:
If you still want to use your own Arduino IDE1.8.5
- Please make it portable to have a dedicated Prusa Firmware build environment
- Use the correct Boards definition
- Update Arduino IDE 1.8.5 portable with https://github.com/prusa3d/PF-build-env/tree/1.0.6-WinLin

TilTisback, 2020-06-25:
Hello 3d-gussner, i have compiled the firmware now with a portable arduino 1.8.5 version and added the file from your link into it. Still the m300 command does not work properly :(

Do you have some more information about that PF-build.sh and how to use it?

ChipCE, 2020-07-01:
I have my firmware build on linux with the <code>build.sh</code> file. M300 works fine. if there is diffirent when building with arduino and cli?

closed 2023-10-31 by github-actions[bot]