prusa3d/Prusa-Firmware #2755: [BUG] MK3S - FW 3.9 with Igus Drylin bearings - Selftest fails, no Zero Autohome random

ManuGithubSteam, 2020-06-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.9.0

**MMU Upgrade** - None
**MMU upgrade firmware version NA

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
 After flashing 3.9.0 from 3.8.1 the selftest can not be completed because the supplied feedrate/current is not strong enhough to move the extruder to the left side of the x axis with drylin bearings. I suspect it will work with ball bearings but it used to work fine with dryline before in 3.8.1
Falsh 3.9.0 after installing drylin bearings on the x and y axis and try the selftest.
  
**Expected behavior**
  Selfttest would finish and the extruder move.
  
**G-code**
  Selftest routine in ultralcd.cpp

**Video**
Video to big to upload.....

https://drive.google.com/file/d/1vOxznVe0ZvxwbyEtlbL5bx-mci3QuLGA/view?usp=sharing

Be aware that i changed the Stall Guard value to 10 on the Xaxis to even get the extruder head so far. With the default 4 it would just move like 3 mm or so. Also it sometimes needs a push....

Barnabasek, 2020-06-23:
See if the 3.9.1 Beta fixes your issue - https://github.com/prusa3d/Prusa3D-Test-Objects/issues/14

JohnnyDeer, 2020-11-26:
Hi @ManuGithubSteam is this issue still actual on latest FW version? Thank you.

closed 2023-10-31 by github-actions[bot]