prusa3d/Prusa-Firmware #2758: [BUG]Extruder motor remains very hot after crash

kawaguchi20, 2020-06-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

Printer Type: MK3S
Printer firmware version: 3.9.0
Print via SD card

**Describe the bug**
After a "crash" is detected, the printer stops printing. The bed and print head turns off and cools down however the extruder motor remains powered and is very very hot to the touch.  (Fans were still running also.)
Basically this is the same issue as #2710 however I'm opening a new issue because I arrived to the issue in a different way and even if one could argue that you want to keep this behavior during a pause, I don't believe it makes sense to do so after a crash.  Or at least it should time out and completely shut down after some time. 
  
**To Reproduce**
In my case, I entered a "crash" detected situation after the filament was not freely feeding (it was stuck in the roll winding and completely bound). As this was my first time trying to print overnight, it was about 5 hours before the time it crashed and I checked it and saw the "crash" notice on the screen.
  
**Expected behavior**
  I would expect it to be completely powered down.  If not right away, after some timeout period.  It should not stay hot indefinitely. At least with a deliberately triggered pause the operator knows the printer's state. With a crash, the paused state may not be detected for hours.

wavexx, 2020-09-22:
I think it's relatively safe to disable "E" if we're hitting the safety timer. I'd love to merge it with #2710 if possible.

It's true that the extruder can lose the position when powered off. However, when the printer has been let idle for more than 10 minutes or so, *even* if the filament has been retracted changes are high that the tip will be deformed. In this case to have a "good" result when unpausing or resuming the print, ideally we would need to prime the extruder again.

I don't think that unpowering the motor after the safety timer hits changes the quality of the resume enough to be noticeable (mostly because I expect the quality not to be that great anyway).

wavexx, 2020-09-22:
Just to make it clear, I'd still keep E powered during a pause, just allow to timeout E on idle.
If you're doing something with the printer while it's paused (like cleaning the print or moving stuff around), keeping the extruder position is important.

Prusa-Support, 2022-05-23:
Thank you for sharing your ideas.

The high temperature of the stepper motors should not really be problematic as explained in [this comment](https://github.com/prusa3d/Prusa-Firmware/issues/2710#issuecomment-1133641005) in issue #2710, but, as said in the same comment, probably it makes little sense to keep the Extruder motor engaged for a long period.

So the matter discussed here is very similar (or equal) to the one discussed in issue #2710, and probably both cases would be resolved if the Extruder stepper motor (or the stepper motors in general) was disengaged by a safety timer.
After that, homing the axes X and Y will be required before resuming of course.

A paused print after multiple crashes is a rather rare scenario. A dedicated fix for this very specific scenario could less likely be implemented anytime soon because of reduced free memory on the motherboard.
For this reason, I suggest closing the issue and possibly sharing your ideas on issue #2710.

Also, to clarify, this is not an unintended behavior (bug) but a request for a new feature (eg motors safety timer), or to improve an already existing feature (either the crash detection, or the already existing temperature safety timer to disengage the motors too...).

The issue has been silent for nearly two years and could be closed as stale, but I'll update the labels and keep it open for now.

Michele Moramarco
Prusa Research

closed 2023-10-31 by github-actions[bot]