prusa3d/Prusa-Firmware #2760: [BUG]External perimmiters with 0.6mm nozzle

2Kasi2, 2020-06-25:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0-3421

**SD card or USB/Octoprint**
  I used SD card

**Describe the bug**
  Well there are 2 issues, but they are both with external perimiters on my prints. 
1st issue is when i use .3mm layer height with .6mm nozzle the "seam" is terrible. It looks like the nozzle starts to extrude a bit too late and therefore there is a gap(see picture of skull). Also the seam function doesnt properly put the seam where it is supposed to. I set it to Rear, but as you can see some of it was in the front!
Second issue is ugly external perimiter on 1 side of the print, and they might be related because it kinda widens as the Zaxis rises. I found that each new layer on top of previous one was a bit worse, both with the bad seam and the ugly finish.
Also with the secong print you can see that the cilinder which is in one corner kinda sticks out of the print(see pics)

**To Reproduce**
 The bad seam I used both .2 and .3mm layer heights with .6mm nozzle. It happend both while using Prusa PLA white, and PETG prusament Gold. The PLA was much better as the PETG. 
The second issue which might 
  
**G-code**
First Gcode is of the Skull, printed in PETG, it had the wide and wrongly placed seam.
Second is a pot and the seam here seems to look better, as the "seams" (or whatever you call them) are on the corner. But one side of the print has this ugly finish. 
(cpuldnt upload Gcode directly so i converted it to .txt)

** Actual gcodes**
[second_issue.txt](https://github.com/prusa3d/Prusa-Firmware/files/4832419/second_issue.txt)
[issue_one.txt](https://github.com/prusa3d/Prusa-Firmware/files/4832427/issue_one.txt)

**pictures of the Skull (issue one)**
[image]
[image]
[image]
[image]

**Pictures of the pot (issue 2)**
[image]
[image]
[image]
[image]

**Video**

wavexx, 2020-06-30:
Sounds like #2468. We have a fix in the works, which can be tested [here](https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.9.1-BETA-3459)

paalmiso, 2020-07-27:
@2Kasi2 did you test it with beta firmware?

wavexx, 2020-09-22:
@2Kasi2 I think this should be fixed in the new 3.9.1 FW just released. Could you give us some feedback?

2Kasi2, 2020-09-28:
> @2Kasi2 did you test it with beta firmware?

no

2Kasi2, 2020-09-28:
> @2Kasi2 I think this should be fixed in the new 3.9.1 FW just released. Could you give us some feedback?

will as soon as i try it out, but idk whet that willl be

Kachidoki2807, 2020-09-29:
Is there any way to test 'objectively' the print quality with the 0.6mm nozzle? Maybe by printing the same object with a 0.4mm nozzle?
I have some doubts, the print quality improved since the LA15 fix in 3.9.1 with the 0.6mm nozzle (associated to the new K-Factor from the default Prusament PETG profile). I still had some gaps on tests models but I also have a very bad perimeter quality on fast print. I printed a motherboard box for a 3D printer, in Prusament PETG, 0.4mm layer height, the walls are so uneven.

Here the straight wall is OK:
[image]

But here it is not really properly 'aligned' (don't take care of the small drop of PETG inside the wall):
[image]

I suspect those features to be the source of this issue:
[image]

The inside view:
[image]

wavexx, 2020-09-29:
@Kachidoki2807 can you share the 3mf for that one?

The banding on the outer perimeters almost looks specular between top and left side. It looks to more more like a difference in extrusion direction (going clockwise vs clockwise), which *may* be triggered by the feature. Such a difference is always visible depending on the extrusion path, but becomes larger with either higher layer heights or larger nozzles.

But could also be a dirty or ovalized nozzle BTW. Not claiming it's the case here, but I had a similar pattern on another printer and when I had to swap the nozzle due to wear the problem vanished.

Kachidoki2807, 2020-09-29:
Thanks for the answer.
Regarding the nozzle, it is a plated copper nozzle from Trianglelab that passed less of a kilo of PLA, PETG, ABS, no abrasive materials. I don't think it is already worn out, and I haven't spare yet as it was only to give a try.

I don't have the 3mf, I am not yet used to save them, but I still have the G-Code file. So I rebuilt the 3mf by importing the STL and the config from the G-Code (very useful feature BTW, saved me several times). I zipped the resulting 3mf and the original G-Code.
[GTR box-v2.zip](https://github.com/prusa3d/Prusa-Firmware/files/5298053/GTR.box-v2.zip)

I think I told something wrong, the Prusament profile wasn't up-to-date at this time (I upgraded the firmware, but not the filament profiles when I printed this specific part). The K-Factor was 0.05 instead of 0.04, not sure it can result in a such behaviour.
Some settings are irrrelevant (like supports), since it was the same PrusaSlicer project that I used for several parts.

I don't exclude a mechanical issue with the linear bearings, they can be worn out since the massive Face Shield printing period. I did notice a very small play and forseen to replace them at the next service time.

Kachidoki2807, 2020-10-04:
Hello,
I have a new case.
Below a picture of a print in ASA (quad white) with very pronounced banding:
[image]

Next to it (quad purple), the very same G-Code in PLA (at 5mm height, I had to manually decrease the temperatures and increase the fan).
In backgroung (single purple), the same model on the same printer but with 0.30mm layer height instead of 0.35mm.
I am not sure it is firmware related so far, but it seems to occurs when layer height is >0.30mm (the previous box was in 0.4mm layer height).

Attached the .3mf of the bad print:
[Profil.zip](https://github.com/prusa3d/Prusa-Firmware/files/5323084/Profil.zip)

Below, the same ASA print based on the 0.30mm profile:
[image]

Attached its .3mf:
[Profil_0.3mm.3mf.zip](https://github.com/prusa3d/Prusa-Firmware/files/5323098/Profil_0.3mm.3mf.zip)

So it's definitely not a mechanical issue, but I can't tell if it comes from the slicer or the firmware.

wavexx, 2020-10-05:
Massive! I don't have a .6 nozzle at hand though. You don't happen to have seen this effect with a .4 nozzle as well?

 I happened to print a small ASA box, showing odd banding on one corner which is similar to the first box that you were showing. In my case the outer and inner perimeter are featureless, and the print direction is the same through-out:

[image]

I'm looking into this, and I think it can be related.
However I see I clear pattern in this one, which is also repeating "nicely".

On your towers it looks more random. Is the hotend temperature stable while you're printing at .35 layers? The higher flow will suck more heat, and maybe we're seeing a difference in viscosity (just guessing here)

Kachidoki2807, 2020-10-06:
In all showed cases, I speed up the print to be stuck at the max volumetric speed (AFAIK 15mm³/s for PLA, 8mm³/s for PETG and 11mm³/s for ASA). So the .35 layers print at lower speed but the flow is the same as the .30 layers. I have no benefit in print time by printing at .35, but it allow the printer to go at slower speed and thus to do less movements to preserve the mechanics or (in that specific case of 18cm towers) to introduce less wobble on tall object. But I'll keep an eye on the temperature during the next prints, just to be sure.

I have to do more tests at lower speed to see if it's not an interlayer adhesion or cooling issue, even though I don't believe it. I'll give a try also on the second printer with the .4 nozzle in order to try to reproduce it, but I didn't observe that effect so far yet. Unfortunately I'll go on holidays up to this weekend, so it will take a bit time before the next report.

Your box has featureless perimeters, but maybe the pattern reflects some infill spacing of the thick wall?

**Note:** Certainly not related at all, but I noticed sometimes that the same g-code, printed twice in a row, does not generate the same 'noise'. I suspect something with the babystepping or the MBL, that sometime gives a regular diagonal pattern or moiré on flat vertical surface.

wavexx, 2020-10-06:
> I have to do more tests at lower speed to see if it's not an interlayer adhesion or cooling issue, even though I don't believe it. I'll give a try also on the second printer with the .4 nozzle in [...]

I noticed both ABS and ASA are quite picky about flow in general. A sudden increase/decrease of flow will create a very visible rebounding effect during the extrusion that looks like a ripple. Calibrating the LA factor for ASA can be quite tricky with the normal test, and I often just look at a print instead.

You can often see that on the purge line:

[image]

I only *very* rarely see this with PETG, and never with PLA. The purge line has higher flow than what it should be, but I've seen this effect in several other places, including starting during crossings in grid infill.

Kachidoki2807, 2020-11-09:
New case today:
[image]
[image]

Prusament Galaxy Silver.
Same banding as before PLUS some gaps. I am using PS 2.3.0-alpha3 with almost stock 0.30mm QUALITY profile, at least nothing about perimeters settings can explain that.

3MF project:
[power_supply_fan_silencer_timqui.zip](https://github.com/prusa3d/Prusa-Firmware/files/5511022/power_supply_fan_silencer_timqui.zip)

What is really weird is that I regularly print on this machine, and the prints are mostly good with perfect walls, and sometimes this happen.

Prusa-Support, 2022-07-08:
Thank you all for your feedback.

This issue looks stale, but I'd like to possibly hear from you before closing.

The following may have improved this issue.
- In November 2020 - after the opening of this issue - the MK3S received a design upgrade: the printer model MK3S+ was released. The MK3S+ printer has slightly shorter filament path inside the extruder block, also X and Y linear bearing mounting was also slightly improved to reduce chances of overtightening and deforming the bearings.
- The firmware received multiple updates too.
- This issue may have been related to Prusa Slicer and its configuration. Both were continuously updated over time.
Speaking of which, a major update of Prusa Slicer is going to be released soon. It is already available as alpha testing version ([2.5.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha2)). The perimeters and seam generation has been improved, which may be especially helpful with issues similar to the ones discussed here. Also the new experimental _Pressure Equalizer_  may turn out handy.

The state of the hardware may play a role too. Bearings, smooth rods, belts, nozzle, PTFE tube, extruder gears, etc... These are examples of components that could be consumed by fatigue and show signs of wearing. Their lifespan surely depends on assembly, use and maintenance. A visual inspection could be enough but please consider a replacement to rule out a concern about consumed components.

Does the issue persist with design, firmware, Prusa Slicer, and Prusa Slicer configuration, all genuine Prusa and up-to-date?

Michele Moramarco
Prusa Research

rasssta, 2023-01-29:
@2Kasi2 : Did you ever fix this problem? I recently swapped to 0.6 on my Mini and I have the same issue as you -> "It looks like the nozzle starts to extrude a bit too late and therefore there is a gap".

2Kasi2, 2023-01-29:
Sorry, no I have not solved the issue.

Prusa-Support, 2023-02-09:
At this point, I wonder if we are looking at artifacts related to print settings and specific print geometries.
Those should be posted in [prusa3d/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) repository.

Arachne perimeter generator, pressure equalizer, seam-specific improvements (and more...) could significantly reduce a few known artifacts typical of FDM printing.

.

A "delayed" extrusion after a retraction on a MINI printer could be caused by a shaking hotend PTFE tube or specific settings and print scenarios.
Even that could be relevant for other repositories like [prusa3d/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer), or [prusa3d/Prusa-Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy). or [prusa3d/Original-Prusa-MINI](https://github.com/prusa3d/Original-Prusa-MINI).

.

This may not be the most suitable repository to discuss these problems and still I think this issue could be closed as stale.
However, if you think your problem is still relevant for this issue (and repository), I wouldn't mind checking a few pictures/videos, and related gcodes.

Michele Moramarco
Prusa Research

rasssta, 2023-02-10:
> At this point, I wonder if we are looking at artifacts related to print settings and specific print geometries. Those should be posted in [...]

Hi,

I don't think this is a issue with PrusaSlicer as I get the same type of scars with PrusaSlicer/SuperSlicer/Cura. Seems to be mainly on the right hand side of the build plate. I've swapped all PTFE tubes with no improvement. I've also tested with and without filament sensor, but the issue persists.

However, as my problem is for the Buddy firmware (Prusa Mini, bowden) it might be an idea to move it to that repo, but as the issue seems to exist in the MK firmware (MK, direct drive) as well it might be of interest to keep it here? But honestly I'm quite sick of this issue now and will revert to 0.4 mm nozzle so I won't be able to test anything.

Have you tested 0.6mm nozzles at the factory for MK/Mini or do you only guarantee the functionality for 0.4mm nozzles?

Another interesting note, it seems to be mainly related to extrusions that are made after the print head is moved from the left hand side to the right hand side of the sheet.

In the print below I placed 4 towers, one in each corner of the sheet. The tower on the right was on the bottom right, the top tower was on the top right, the left one on the top left and the lower one on the bottom left. But if I move the right hand towers close to the middle I still get the same issue with them, whereas if I move the left hand towers to the middle they print just fine.

[image]

rasssta, 2023-02-10:
Adding 2 gcodes for PLA, Prusa Mini, 0.6 mm nozzle.

Cura:
[OPM_rettest_2.txt](https://github.com/prusa3d/Prusa-Firmware/files/10706479/OPM_rettest_2.txt)

PrusaSlicer 2.5:
[rettest_0.6n_0.2mm_PLA_MINI_15m.txt](https://github.com/prusa3d/Prusa-Firmware/files/10706486/rettest_0.6n_0.2mm_PLA_MINI_15m.txt)

Prusa-Support, 2023-02-17:
Sorry, I didn't mean to call it a Prusa Slicer problem, but something that can be improved there (starting with placing the objects closer).

I find it hard to understand if the perimeters were printer clockwise or counterclockwise. I belive they should be printer counterclockwise but they may look like printed clockwise.
Other than this, the settings that I see should not be specifically problematic.
I may be missing something about the gcode, but I can confirm this problem is not relevant for this issue.

I think I've seen this problem before though.
**I think it is specific to distant objects printed on the MINI.**

Also, I'm not sure if this is hardware or software related, probably hardware related but they may come out with improvements in these regards on the software level.

It can be posted in [prusa3d/Prusa-Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy) and we will consider adding a "HW issue" label.
You may want to post it twice, on [prusa3d/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) too, as software improvements may also come from the print settings side, but please mention the issue posted in Prusa-Firmware-Buddy there.

.
Back to the issue in the main post here: Arachne perimeter generator and pressure equalizer could help.
This issue could be closed because stale too.

Michele Moramarco
Prusa Research

rasssta, 2023-02-18:
> I find it hard to understand if the perimeters were printer clockwise or counterclockwise. I belive they should be printer counterclockwise but they may look like printed clockwise. Other than this, [...]

IIIRC they were printed counterclockwise. Is it possible to change this in PrusaSlicer, just to test?

> I think I've seen this problem before though. **I think it is specific to distant objects printed on the MINI.**

Do you mean that you've seen it internally or publicly? I tried to search for it but didn't find anything.
  
> It can be posted in [prusa3d/Prusa-Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy) and we will consider adding a "HW issue" label. You may want to post it twice, on [...]

I've added the issue to the Buddy repo now.

> . Back to the issue in the main post here: Arachne perimeter generator and pressure equalizer could help. This issue could be closed because stale too.

I've tried with and without Arachne, same issue. Pressure equalizer in the slicer could be a workaround though. I haven't played around with that part in Cura, and it doesn't exist in PS afaik.

I have done K-factor calibration though, it helps...but the problem persists.

Prusa-Support, 2023-03-03:
Thanks for reporting in the relevant repositories.

On a side note, Pressure Equalizer would help where there is a transition between fast and slow features. It won't make a big difference if there is no fast infill.
By the way, clockwise or counterclockwise is not relevant for the prints. It caught my attention because it didn't match your prints' direction. Probably there is no correspondence between the picture and G-code that I analyzed (different .gcode file).

We can continue this on a separate thread, where hopefully other users will be available to test and reproduce the issue.

Michele Moramarco
Prusa Research

closed 2023-10-31 by github-actions[bot]

georgiys, 2024-01-09:
Has anyone solved this problem? Or at least could minimize it? I still faced with this issue(

Given:
1. 3 Minis(6 mo - 18 mo old) with new brass nozzle, new PTFE, cleaned heatbreak, cleaned extruder gears. 
2. 1 totally new Mini(Semi-assembled), just from the box
3. PLA(couple of manufacturers including Prusament) dried for 12 hrs
4. PrusaSlicer 2.7.1
5. Firmware 4.4.1 first, then 5.1.2(with and without InputShaper)

Result:
[image]
[image]

Small test file. For 0.6 it's enough to use common profile without changes(e.g. 0.2 SPEED or 0.25 STRUCTURAL)
[seams.zip](https://github.com/prusa3d/Prusa-Firmware/files/13880596/seams.zip)

wavexx, 2024-01-09:
@georgiys This issue is for the MK2/MK2.5/MK3 firmware. The Mini/MK4 uses a different firmware with a different implementation (either firmware with IS is now using pressure advance instead).

Please report this to https://github.com/prusa3d/Prusa-Firmware-Buddy/. The gap though is pretty big, and makes me think some issue with the profile.

Prusa-Support, 2024-02-16:
It could actually be the case of a shaking/deformed Hotend PTFE tube or worn Extruder-to-Hotend PTF tube.
Please check or replace these parts before posting such an issue in the Prusa MINI's repository.
Our Customer Support can help you check this.

Michele Moramarco
Prusa Research

georgiys, 2024-02-16:
> It could actually be the case of a shaking/deformed Hotend PTFE tube or worn Extruder-to-Hotend PTF tube. Please check or replace these parts before posting such an issue in the Prusa MINI's repository. Our Customer Support can help you check this.
> 
> Michele Moramarco Prusa Research

Already did. My story is here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997#issuecomment-1927590095

Prusa-Support, 2024-04-06:
@georgiys This looks much worse than issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2997's artifacts and very likely related to a problem of filament, hardware, or print conditions... I can't tell for sure so please continue with the [Customer Support](https://help.prusa3d.com/article/customer-support_2287) in the official support channels.
Ultimately, I'd like to remind you this is the wrong repository - https://github.com/prusa3d/Prusa-Firmware/issues/2760#issuecomment-1883942452.

Michele Moramarco
Prusa Research

georgiys, 2024-04-06:
@Prusa-Support Please read all the story one more time. 
This artifact is the same on totally new semi-assembled Prusa Mini(while working in closed box), printed with Prusament PLA, dried for 12 hrs. If it's still filament/hardware problem, Prusa Company did this, not me!

Unfortunately, customer support is totally useless in my issue. I proposed them to try print my test project. They refused it. Too complicated?

Ultimately, I'd like to remind you whether it's right repository or not @Prusa-Support keeps to do nothing to help their customers.

Prusa-Support, 2024-06-15:
@georgiys I'm sorry we can't assist you in the **MK2/MK3** firmware repository.
Unfortunately, **we don't talk about Prusa MINI here** and _your issue doesn't seem to be related to the firmware either_.
I'm very sorry to say that we should interrupt this conversation here, and I hope you can find answers in the last few comments.

Michele Moramarco
Prusa Research